RIOT-OS/RIOT #9402: What is the best hardware for multi-hop WSN ?

mmaxus35, 2018-06-22:
I have studied Contiki-OS in TelosB mote and so far i have seen that Riot is more promising and opened. I would like to shift my project to Riot-OS. However, for a quick adaptatiton, is there anyone who tried to create a multi-hop newtork using TelosB or Z1. Maybe another hardware which is better ?

Thanks.

mmaxus35, 2018-06-23:
Sadly i still use 16-bit platform because of lack of sources in the University. Yet we are preparing for upcoming season and would like to shift to newest 32-bit platforms and operating system. I would like to know more about Samr21-xpro and investigated a bit. . In the light of your suggestions i will start searching on my thesis. However, I couldn't find any information about Hamilton board ?

Thanks

PeterKietzmann, 2018-06-25:
@mmaxus35 are you aware of the [FIT IoT-LAB](https://www.iot-lab.info/) testbed? The [M3](https://www.iot-lab.info/hardware/m3/) nodes are well supported by RIOT and on the Grenoble (and Lille?) site you can build multi-hop topologies.
As for a local setup I'd recommend:

[Atmel ATSAM21-XPRO](http://www.microchip.com/developmenttools/productdetails.aspx?partno=atsamr21-xpro&utm_source=MicroSolutions&utm_medium=Link&utm_term=FY18Q1&utm_content=DevTools&utm_campaign=Article):
- on-board 802.15.4
- used by many RIOTers
- no on-board sensors

[STM32Nucleo](http://www.st.com/en/evaluation-tools/nucleo-f411re.html) + external [Openlabs](http://openlabs.co/store/Raspberry-Pi-802.15.4-radio) radio:
- requires [adapter PCB](https://github.com/MichelRottleuthner/rpi-mrf-radio-shield) or jumper cables
- no on-board sensors
- cheap and easily available
- allows using any of the Nucleo boards. Further, the adapter fits any
Arduino R3 connector. The radio itself was designed for RasPis, so it's
obviously applicable there as well

Can we close this issue again?

mmaxus35, 2018-06-25:
Sure thanks!

closed 2018-06-25 by PeterKietzmann