RIOT-OS/RIOT #9405: avr: C standard library is incomplete

ZetaR60, 2018-06-24:
#### Description
avr-libc 1.8.x does not implement the full C standard library. It is missing at least code related to ```time.h```. Thus things related to RTC work on the ATmegas cannot be linked (they can be compiled due to ```cpu/atmega_common/avr-libc-extra``` providing ```time.h```, but an implementation is not provided). avr-libc 2.0.x is required for an implementation for time. Unfortunately, the Docker image and build server use the old version (Debian does not provide 2.0.x).

#### Steps to reproduce the issue
Attempt to compile code using functions from ```time.h```, such as in #8842

#### Expected results
Function calls using ```time.h``` link correctly.

#### Actual results
From build server output:
```
--- build output of app tests/periph_rtc for board mega-xplained (raw, runtime=0.8s):
-- running on worker haw-icc thread 1, build number 84590.
make: Entering directory '/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/tests/periph_rtc'
Building application "tests_periph_rtc" for "mega-xplained" with MCU "atmega1284p".

/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/build/atmega_common_periph.a(rtc.o): In function `rtc_set_time':
rtc.c:(.text.rtc_set_time+0x20): undefined reference to `mk_gmtime'
/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/build/atmega_common_periph.a(rtc.o): In function `rtc_get_time':
rtc.c:(.text.rtc_get_time+0x64): undefined reference to `gmtime_r'
/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/build/atmega_common_periph.a(rtc.o): In function `rtc_set_alarm':
rtc.c:(.text.rtc_set_alarm+0x2a): undefined reference to `mk_gmtime'
/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/build/atmega_common_periph.a(rtc.o): In function `rtc_get_alarm':
rtc.c:(.text.rtc_get_alarm+0x6): undefined reference to `gmtime_r'
collect2: error: ld returned 1 exit status
/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/tests/periph_rtc/../../Makefile.include:356: recipe for target '/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/build/tests_periph_rtc.elf' failed
make: *** [/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/build/tests_periph_rtc.elf] Error 1
make: Leaving directory '/tmp/dwq.0.9010834105496806/82cd9ea90eaac3dd14c6d7f70c0c870c/tests/periph_rtc'
-- build directory size: 692K
```

#### Versions
From build server output:
```
	Installed compiler toolchains 
	-----------------------------
	             native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
	      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
	                avr-gcc: avr-gcc (GCC) 4.9.2
	       mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
	             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
	   riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bits) 7.2.0
	                  clang: clang version 3.8.0-2ubuntu4 (tags/RELEASE_380/final)
	
	Installed compiler libs
	-----------------------
	   arm-none-eabi-newlib: "2.5.0"
	    mips-mti-elf-newlib: "2.1.0"
	riscv-none-embed-newlib: "2.5.0"
	               avr-libc: "1.8.0svn" ("20111229")
	
	Installed development tools
	---------------------------
	                  cmake: cmake version 3.10.0
	               cppcheck: Cppcheck 1.72
	                doxygen: 1.8.11
	                 flake8: missing
	                    git: git version 2.7.4
	             coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

#### References:
#8842 
#8857

ZetaR60, 2018-06-29:
@kYc0o No, #8857 solved a separate issue of time.h being outdated, which caused compiled code to call the time functions with the wrong variable types. It does not add the library implementation. The library time implementation requires an updated library version on Docker / build server.

ZetaR60, 2018-06-30:
After looking into this issue, I think that upgrading the Docker image from Ubuntu Xenial to Ubuntu Bionic would solve the problem. [https://packages.ubuntu.com/bionic/avr-libc](url) lists the version as 2.0.0 and Bionic is also a long term support version (like Xenial). I assume that the build server is using the Docker image, so it would fix both.

@gebart would it be possible to update the Docker image from Ubuntu Xenial to Ubuntu Bionic?

ZetaR60, 2018-11-27:
The Dockerfile changes have been merged. Once the changes filter through to Murdock, we can close this.

smlng, 2018-11-29:
all Murdock-Slaves should be up to date now!

ZetaR60, 2018-11-29:
Thank you! Closing this issue.

closed 2018-11-29 by ZetaR60