RIOT-OS/RIOT #9412: Unable to build for AVR on Ubuntu : undefined reference to 'main'

Test6778678, 2018-06-26:
#### Description

Cannot build for any off the AVR - based boards.
#### Steps to reproduce the issue

Install the required packages : 
sudo apt-get install gcc-avr binutils-avr gdb-avr avr-libc avrdude
Clone the RIOT git repository, create the Makefile (Used the one presented here : http://doc.riot-os.org/getting-started.html)
Try to build it, for example make BOARD=arduino-uno
#### Expected results

RIOT builds without issues.
#### Actual results

user@erver:~/UNO$ make BOARD=arduino-uno
Building application "Test" for "arduino-uno" with MCU "atmega328p".

"make" -C /home/user/RIOT/boards/arduino-uno
"make" -C /home/user/RIOT/boards/common/arduino-atmega
"make" -C /home/user/RIOT/boards/common/atmega
"make" -C /home/user/RIOT/core
"make" -C /home/user/RIOT/cpu/atmega328p
"make" -C /home/user/RIOT/cpu/atmega_common
"make" -C /home/user/RIOT/cpu/atmega_common/periph
"make" -C /home/user/RIOT/drivers
"make" -C /home/user/RIOT/drivers/periph_common
"make" -C /home/user/RIOT/sys
"make" -C /home/user/RIOT/sys/auto_init
"make" -C /home/user/RIOT/sys/isrpipe
"make" -C /home/user/RIOT/sys/tsrb
"make" -C /home/user/RIOT/sys/uart_stdio
/home/user/UNO/bin/arduino-uno/core.a(kernel_init.o): In function `main_trampoline':
/home/user/RIOT/core/kernel_init.c:59: undefined reference to `main'
/usr/lib/gcc/avr/5.4.0/../../../avr/lib/avr5/crtatmega328p.o:(.init9+0x0): undefined reference to `main'
collect2: error: ld returned 1 exit status
/home/user/UNO/../RIOT/Makefile.include:356: recipe for target '/home/user/UNO/bin/arduino-uno/Test.elf' failed
make: *** [/home/user/UNO/bin/arduino-uno/Test.elf] Error 1
user@erver:~/UNO$
#### Versions

Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
      arm-none-eabi-gcc: missing
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: missing
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: missing
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.1
             coccinelle: missing

bergzand, 2018-06-26:
Hi @Test6778678 

Are you able to compile the arduino_hello-world example? With:
```
make -C examples/arduino_hello-world all BOARD=arduino-uno
```
From your RIOT base dir.

From your issue it looks like you're missing a file with the `main` function for your project. Try to have a look at the examples from your clone for a few pointers on how to start.

Test6778678, 2018-06-26:
Hi @bergzand 
I am able to build it without issues.

make: Entering directory '/usr/home/(user)/Desktop/RIOT/examples/arduino_hello-world'
Building application "arduino_hello-world" for "arduino-uno" with MCU "atmega328p".

"make" -C /usr/home/(user)/Desktop/RIOT/boards/arduino-uno
"make" -C /usr/home/(user)/Desktop/RIOT/boards/common/arduino-atmega
"make" -C /usr/home/(user)/Desktop/RIOT/boards/common/atmega
"make" -C /usr/home/(user)/Desktop/RIOT/core
"make" -C /usr/home/(user)/Desktop/RIOT/cpu/atmega328p
"make" -C /usr/home/(user)/Desktop/RIOT/cpu/atmega_common
"make" -C /usr/home/(user)/Desktop/RIOT/cpu/atmega_common/periph
"make" -C /usr/home/(user)/Desktop/RIOT/drivers
"make" -C /usr/home/(user)/Desktop/RIOT/drivers/periph_common
"make" -C /usr/home/(user)/Desktop/RIOT/sys
"make" -C /usr/home/(user)/Desktop/RIOT/sys/arduino
"make" -C /usr/home/(user)/Desktop/RIOT/sys/auto_init
"make" -C /usr/home/(user)/Desktop/RIOT/sys/div
"make" -C /usr/home/(user)/Desktop/RIOT/sys/isrpipe
"make" -C /usr/home/(user)/Desktop/RIOT/sys/tsrb
"make" -C /usr/home/(user)/Desktop/RIOT/sys/uart_stdio
"make" -C /usr/home/(user)/Desktop/RIOT/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
   9606	    302	   1001	  10909	   2a9d	/usr/home/(user)/Desktop/RIOT/examples/arduino_hello-world/bin/arduino-uno/arduino_hello-world.elf
make: Leaving directory '/usr/home/(user)/Desktop/RIOT/examples/arduino_hello-world'

I have already checked if any file was missing, but they are not. All example compile perfectly, but I'm unable to compile the OS itself.

bergzand, 2018-06-26:
Hi @Test6778678 

Thank you for providing the build output. From what I see there is no issues with your installation. From the error of your original post, it seems that the linker is unable to find a function called `main`. This function should be provided by your custom project and is not provided by RIOT-os.

Please let me know if you have more questions.

Test6778678, 2018-06-26:
Thanks. The problem was that I had misunderstood something. I was trying to build a project without my own projects files... That why its failed. After putting them in the right place, everything's fine.
Thanks for your support ! I'll close this now.

closed 2018-06-26 by Test6778678