RIOT-OS/RIOT #9414: periph/i2c: [TRACKING] i2c bugs list

MrKevinWeiss, 2018-06-26:
This will be a compilation of the issues and corner case bugs found during the new i2c if embargo.  The tracking of progress is found in #6577

- [ ] i2c_read_bytes() with I2C_NOSTOP set in sam_common0 cpu will cause lockup.  _[Suggested Fix]_ if I2C_NOSTOP then don't wait until the bus is free.  Since the SCL line stays low due to the master at this point the bus should not be finished.
- [ ] i2c_read_bytes() with I2C_NOSTOP followed by a repeated start locks up cpu if the first bit of slave data is 0, this occurs on multiple cpus.  _[Suggested Fix]_ check if the SCL line is low before sending a start, if it isn't return an error.
- [ ] The stm32 cpu, under i2c_1.c it appears that sending to addresses that aren't there does not catch the address NACK, as a result data gets entered into the DR and a following operation will have the data offset by one.  _[Suggested Fix]_ Add a sufficient time delay before checking the status register for the NACKed bit to be set. May be related to #7846
- [ ] The stm32 cpu, under the i2c_2.c has some funny issues with what is on the line during a I2C_NOSTOP read.  The data seems find however extra bytes are clocked out due to the ACK being set by the master (which it should be since we are expecting more data).  _[Suggested Fix]_ Don't support I2C_NOSTOP read since users can then mess up a proper frame, there seems to be no practical case for which it is useful.
- [ ] Basic error messages in stm i2c implementation.  Everything seems like it is a timeout and there is more information available.  See #8461  for some fixes
- [ ] Bus initialization related to #2528.  This may be solved by moving the init to the CPU and not making it a part of the API, I need to look into it further.

aabadie, 2018-06-26:
If you don't mind, I'd prefer to keep track of it in #6577, this will avoid potential confusion.

MrKevinWeiss, 2018-06-26:
Agreed, moving info to #6577

closed 2018-06-26 by MrKevinWeiss