RIOT-OS/RIOT #9449: github: rename "Ready for CI build" label to "CI: ready for build"

miri64, 2018-06-28:
I'd like to rename the "Ready for CI build" label to "CI: ready for build" to be more in line with the "CI: run tests" label. However, I don't know how hard coded it is into Murdock that it has to be *that* label. So I rather ask ;-).

kaspar030, 2018-06-28:
> I'd like to rename the "Ready for CI build" label to "CI: ready for build" to be more in line with the "CI: run tests" label.

+1

> how hard coded it is into Murdock that it has to be that label.

I checked, it is not hard-coded in current master (minus one place that I just commited a fix). As this is rather cosmetic, I'd rather not restart murdock. How about I keep this in mind and next time we do maintanance on Murdock I rename the label?

miri64, 2018-06-28:
> How about I keep this in mind and next time we do maintanance on Murdock I rename the label?

If it doesn't *slip* your mind :+1: ;-)

miri64, 2018-07-09:
Ping? When will that be?

kaspar030, 2018-07-09:
> next time we do maintanance on Murdock I rename the label?

... still applies. 

(Problem is, Murdock is almost maintanance free, last restart was in Oktober. )

smlng, 2018-07-09:
@kaspar030 the label is defined in `.murdock` file shouldn't be as simple as:
- defining the new label but  leaving the old intact
- create PR modifying the label in `.murdock`
- remove old label as soon as above is merged

No restart or update of workers needed?

smlng, 2018-07-09:
see https://github.com/RIOT-OS/RIOT/blob/master/.murdock#L22

smlng, 2018-07-09:
ah damn, sorry mixed things up

miri64, 2018-07-09:
>> next time we do maintanance on Murdock I rename the label?
> 
> ... still applies. 
>
> (Problem is, Murdock is almost maintanance free, last restart was in Oktober. )

Ok, but there is something like a maintenance schedule, right?

kaspar030, 2018-07-09:
> Ok, but there is something like a maintenance schedule, right?

Nope, not really. We schedule maintanance as needed.

miri64, 2018-09-01:
@smlng @kaspar030 is this included into the current maintainance run?

miri64, 2018-09-03:
Label is still called the old name, so apparently not :(

kaspar030, 2018-09-09:
Ok, current live code can do this now, just needs to be configured. CI team is on it.

miri64, 2018-10-03:
This is done :-)

closed 2018-10-03 by miri64