RIOT-OS/RIOT #9452: make: build for Cortex fails due to '-W missining-include-dirs`

haukepetersen, 2018-06-28:
#### Description
Sadly, this is the second time in a week I have to open this issue. I can, again, not build anything for Cortex-M boards due to the `-Werror=missing-include-dirs`. This time though the compiler complains that some actually existing include path(s) can not be found. 

Example building `hello-world` for the `samr21-xpro`:
```
Building application "hello-world" for "samr21-xpro" with MCU "samd21".

cc1: error: -I/home/hauke/dev/riot/RIOT/core/include: No such file or directory [-Werror=missing-include-dirs]
cc1: all warnings being treated as errors
/home/hauke/dev/riot/RIOT/Makefile.base:83: recipe for target '/home/hauke/dev/riot/RIOT/examples/hello-world/bin/samr21-xpro/application_hello-world/main.o' failed
make[1]: *** [/home/hauke/dev/riot/RIOT/examples/hello-world/bin/samr21-xpro/application_hello-world/main.o] Error 1
/home/hauke/dev/riot/RIOT/examples/hello-world/../../Makefile.include:359: recipe for target '/home/hauke/dev/riot/RIOT/examples/hello-world/bin/samr21-xpro/application_hello-world.a' failed
make: *** [/home/hauke/dev/riot/RIOT/examples/hello-world/bin/samr21-xpro/application_hello-world.a] Error 2
```

The location of ` -I/home/hauke/dev/riot/RIOT/core/include` is actually correct, so I don't quite get the why the compiler is angry here...

Anyone with similar issues or an idea how to fix this?

cladmi, 2018-06-28:
I think it must be something like `-I $(an_empty_variable) -I/path/to/RIOT/core/include`.
I got this before when working on https://github.com/RIOT-OS/RIOT/pull/9243 and adding `-I` without a path after.

Can you paste the output of
`make -C examples/hello-world/ BOARD=samr21-xpro info-debug-variable-INCLUDES`

cladmi, 2018-06-28:
When I overwrite `NEWLIB_NANO_INCLUDE_DIR` to an empty variable I get the same error:
```
$ make -C examples/hello-world/ BOARD=samr21-xpro clean all NEWLIB_NCLUDE_DIR=
make: Entering directory '/home/harter/work/git/RIOT/examples/hello-world'
Building application "hello-world" for "samr21-xpro" with MCU "samd21".

cc1: error: -I/home/harter/work/git/RIOT/core/include: No such file or directory [-Werror=missing-include-dirs]
cc1: all warnings being treated as errors
/home/harter/work/git/RIOT/Makefile.base:83: recipe for target '/home/harter/work/git/RIOT/examples/hello-world/bin/samr21-xpro/application_hello-world/main.o' failed
```

If it is the case, it means your NEWLIB_INCLUDE_DIR location set from `makefiles/libc/newlib.mk` is still not correctly detected. So the error is the same as before but with a worst error message.

The new error message is a consequence of https://github.com/RIOT-OS/RIOT/pull/9394

Can you try building by setting the variable `NEWLIB_INCLUDE_DIR` to the directory where the `nano` directory is:

```
make -C examples/hello-world/ BOARD=samr21-xpro clean all NEWLIB_INCLUDE_DIR=/path/to/your/gcc/directory/with/the/directory/nano
```

cladmi, 2018-06-28:
This is the reason I created https://github.com/RIOT-OS/RIOT/pull/9384 a week ago to at least give an informative warning message.

haukepetersen, 2018-06-28:
so, finally spend some time looking into this. It is indeed an empty variable as in `-isystem -ISOME_PATH`, so this is what causes the trouble.

I did fix this last week by simply setting `NEWLIB_INCLUDE_DIR` for my environment directly, but db6cb1ac65e1fde042c511e68ef13c42bf4b477e broke it again. Turns out, that the changes introduced in db6cb1ac65e1fde042c511e68ef13c42bf4b477e simply expected a trailing `/` for my exported path, so adding this to my `NEWLIB_INCLUDE_DIR` env var solved the issue. 

This is the kind of trouble that happens if we start to make assumptions on toolchain locations (and similar) in our Makesystem -> let's try to keep this to a minimum, its getting harder and harder to use RIOT :-(

Anyway, issue solved (for now), so thanks for the help.

closed 2018-06-28 by haukepetersen

cladmi, 2018-06-29:
> simply expected a trailing / for my exported path

That should not be the case, I will do a PR for fixing it.

> This is the kind of trouble that happens if we start to make assumptions on toolchain locations (and similar) in our Makesystem 

Agreed, it is also the goal of https://github.com/RIOT-OS/RIOT/pull/9415 and related PR to not assume the toolchain location by using `nano.specs`.
Remaining problem is with llvm where it is actually required to find the newlib path from another compiler than the one currently used… but limiting it for this case would help.

toonst, 2018-07-02:
#9415 fixes this, it adds a check to only use NEWLIB_NANO_INCLUDE_DIR if it is not empty.

pyropeter, 2018-07-15:
Why is this closed? This issue still persists.