RIOT-OS/RIOT #9460: docker: make commands for docker require insecure setup.

jcarrano, 2018-06-29:
#### Description

Communicating withe the docker daemon requires elevated privileges. This can be accomplished by either using `sudo` or adding a user to the `docker` group. The latter option is advocated by the [RIOT wiki](https://github.com/RIOT-OS/RIOT/wiki/Use-Docker-to-build-RIOT), and is expected by the build system:

> Depending on your operating system, you might need to add yourself to the docker group:

Using the `docker` group, however, is not as secure as it seems, and in effectively the same as granting the user root permissions (without password). See:

* Docker manual: [Manage Docker as a non-root user](https://docs.docker.com/install/linux/linux-postinstall/), [Docker daemon attack surface](https://docs.docker.com/engine/security/security/#docker-daemon-attack-surface)
* This [blog post](https://www.projectatomic.io/blog/2015/08/why-we-dont-let-non-root-users-run-docker-in-centos-fedora-or-rhel/)
* This [other one](https://fosterelli.co/privilege-escalation-via-docker.html)

#### Steps to reproduce the issue

Try to follow the instructions at https://github.com/RIOT-OS/RIOT/wiki/Use-Docker-to-build-RIOT with a user that is not in the 'docker' group, or on a system where there is no such group.

#### Expected results

`make BUILD_IN_DOCKER=1` should work anyways.

#### Actual results

`make BUILD_IN_DOCKER=1` fails with insufficient permissions.

cladmi, 2018-08-13:
As you raised the issue in the first place, do you mind updating the documentation now that https://github.com/RIOT-OS/RIOT/pull/9635 is merged ?

jcarrano, 2019-05-31:
Fixed, closing.

closed 2019-05-31 by jcarrano