RIOT-OS/RIOT #9518: periph/i2c: tracking bugs and untested acks

MrKevinWeiss, 2018-07-09:
This is a list of bugs found and untested acks given (due to lack of hardware) discoved during the I2C embargo.  After the new_i2c_of branch is merged to master each point can be checked off given appropite time to fix the bug or acquire the hardware.  Untested acks just require a hardware test.  The bugs will require a proper fix.

BUGS:
- [ ] sam_common0 - i2c_read_bytes with NO_STOP causes lockup
- [ ] many cpus - i2c_read_bytes with repeated start causes lockup if first data bit is 0
- [ ] many cpus - improve error messages for I2C

UNTESTED ACKS:
- [ ] bmp180
- [ ] ds1307

TODO: Add more bugs/untested acks

aabadie, 2018-10-19:
@MrKevinWeiss, do you have updates on this one ?

MrKevinWeiss, 2018-10-19:
I was just going through it.  I haven't worked on it since the I2C embargo finished.  IT appears that most problems still exist.  It will be more apparent when the tests get added to the HiL CI.

I can also try to get some of the sensors in house during the next order.

MrKevinWeiss, 2018-11-05:
@jia200x untested ack on the ds1307...  interesting.

gebart, 2018-11-05:
>many cpus - i2c_read_bytes with repeated start causes lockup if first data bit is 0

This sounds like a serious issue. Could you give at least one example of a CPU where this happens, so that others can try to reproduce the bug?

MrKevinWeiss, 2018-11-05:
IMO it is an issue with the protocol (I don't think NOSTOP reads should be supported).  I will try reproducing on all HiL boards (it just takes some time).

MrKevinWeiss, 2018-11-05:
### How to reproduce
I2C_NOSTOP flag is 0x04.
```
i2c_acquire 0
i2c_read_bytes 0 <valid address> 2 4
i2c_read_bytes 0 <valid address> 2 0
```
When reading a 0x00 first or a 0xFF

Board | Result | Comments
-| - | - 
samr21-xpro| lockup | reset recovers
nrf52dk| good | 
nucleo-l073rz| good | 
nucleo-f411re| lockup |  reset does not recover
remote-revb| error | it can read the nostop bytes but everything after has EAGAIN error message
arduino-mega2560| good | 

I believe this issue relates to #10127 and (to quote @gschorcht) 
"3.1.16 Bus clear
...
If the data line (SDA) is stuck LOW, the master should send nine clock pulses. The device
that held the bus LOW should release it sometime within those nine clocks. If not, then
use the HW reset or cycle power to clear the bus.
"
That may be the required solution, add a check then try to clockout.

aabadie, 2019-07-07:
The HIL tests are green on the mentioned boards. @MrKevinWeiss, do you think this issue could be closed ?

MrKevinWeiss, 2019-07-08:
Some of the things in here are very old.  I don't think we put the nostop flag test in the HiL yet.  I was unsure if that could be expected behaviour or not.  Let me first add the test into the HiL so it is exposed.

TODO:
- Add HiL I2C test for read with nostop flag
- Add repeated start read_bytes test with 0 and with 0xFF

MrKevinWeiss, 2020-10-09:
I think this issue is pretty obsolete now.

closed 2020-10-09 by MrKevinWeiss