RIOT-OS/RIOT #9526: Semtech-loramac confirmable TX broken

aabadie, 2018-07-09:
While working on #9524, I realized that sending confirmable messages was broken on the current master branch (master at 15b0fe4368fe473193de2e0aab10be8068da5138).

### How to reproduce

1. Build and flash `tests/pkg_semtech-loramac` application
```
$ make -C tests/pkg_semtech-loramac flash term
```
2. Configure the device with LoRaWAN parameters (deveui, appeui, appkey):
```
> loramac set deveui <DEV_EUI>
> loramac set appeui <APP_EUI>
> loramac set appkey <APP_KEY>
```
3. Activate the device, using OTAA, and ensure activate succeeds
```
> loramac join otaa
Join procedure succeeded!
```
4. Send a confirmable message (default behavior):
```
> loramac tx test
2018-07-09 17:23:53,925 - INFO # sys/random/random.c:69 => 0x8000d3d
2018-07-09 17:23:53,926 - INFO # *** RIOT kernel panic:
2018-07-09 17:23:53,928 - INFO # FAILED ASSERTION.
2018-07-09 17:23:53,928 - INFO # 
2018-07-09 17:23:53,929 - INFO # *** halted.
```

Note that it works with an unconfirmable message:
> loramac tx test uncnf
TX complete, no data received

### Expected behaviour:

The message is sent and no failed assertion happens

### Actual behavior:

The message is sent but the node falls into a failed assertion.

### Useful information:

The failed assertion comes from [this line](https://github.com/RIOT-OS/RIOT/blob/master/sys/random/random.c#L69) and this change was introduced by #9410.

Assigning @PeterKietzmann, since he contributed #9410.

(And the issue template is gone by the way, certainly a consequence of #9504 ?)

jia200x, 2018-07-09:
@aabadie @PeterKietzmann LoRaWAN is sending negative values in the first operand `a`. When casted to uint32_t, the number will be bigger than `b`. That's the reason of the assertion.

I'm checking why LoRaWAN gives negative values

jia200x, 2018-07-09:
@aabadie I think LoRaWAN needs a random_int32_range function instead of the current map to uint32_t random (as seen [here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/semtech-loramac/patches/0003-adapt-utilities-functions.patch#L36))

aabadie, 2018-07-09:
Thanks @jia200x for the good analysis. I opened #9528 to provide a fix. Tested and now it works as before.

PR #9528 (2018-07-10): pkg/semtech-loramac: use random_uint32() to generate random integer
Then the patched srand1 function used by loramac is closer to the original code and won't crash because of negative values

### Contribution description

This PR is an attempt to fix #9526 issue. As @jia200x pointed out, there's a problem with negative values used by loramac and the `random_uint32_range` function that only expects unsigned value as input parameter.

The actual fix is just to keep the same code that is used in the original semtech code and just replace `rand1()` with `random_uin32()`.

### Issues/PRs references [... cut]
closed 2018-07-10 by jia200x