RIOT-OS/RIOT #9534: linkerscripts for stm32 incorrect for MCUs with non-contiguous SRAM1 & SRAM2

MichelRottleuthner, 2018-07-10:
The common linker script for stm32 based platforms doesn't take into account that on some platforms the available RAM is split up in different RAM sections e.g. SRAM1 & SRAM2. 
On some MCUs there is a hardware feature to remap the address space of SRAM2 consecutively after SRAM1, there it doesn't actually impose a problem. This feature is not available on all MCUs though.
On stm32l476rg for example SRAM1 is at 0x20000000 while SRAM2 resides at 0x10000000. After SRAM1 unmapped address space is located (see [RM0351 reference manual page 73](https://www.st.com/content/ccc/resource/technical/document/reference_manual/02/35/09/0c/4f/f7/40/03/DM00083560.pdf/files/DM00083560.pdf/jcr:content/translations/en.DM00083560.pdf#page=73)).

I stumbled across this when testing low power RAM retention which is only available on SRAM2 on the named MCU. 

As a minimum fix the RAM_LEN value in [stm32_mem_lengths.mk](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/stm32_mem_lengths.mk) for affected platforms should be reduced to the minimum consecutively addressable amount (96K in the above case).

A better fix would need to add another ram section to [cortexm.ld](https://github.com/RIOT-OS/RIOT/blob/master/cpu/cortexm_common/ldscripts/cortexm.ld) and [cortexm_base.ld](https://github.com/RIOT-OS/RIOT/blob/master/cpu/cortexm_common/ldscripts/cortexm_base.ld) and respectively additional parameters to [stm32_mem_lengths.mk](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/stm32_mem_lengths.mk). 

I'd like to use this Issue to collect which other MCUs are affected by this (if any) and discussion on how to properly integrate the split RAM to the linker scripts before opening up a PR. Feel free to help, of course ;)

MichelRottleuthner, 2018-07-11:
Checked this for L0, F3, F4, F7 and found some more problems regarding the RAM sizes:
```
<mcu>       (              <Datasheet specs>                | <linkerscript specs>          )
stm32f303vc (40K SRAM @0x20000000; 8K CCM SRAM @0x10000000  | RAM_LEN = 48K, CCMRAM_LEN = 8K)
stm32f303re (64K SRAM @0x20000000; 16K CCM SRAM @0x10000000 | RAM_LEN = 80K)
stm32f303ze (64K SRAM @0x20000000; 16K CCM SRAM @0x10000000 | RAM_LEN = 80K)
stm32f303k8 (12K SRAM @0x20000000; 4K CCM SRAM @0x10000000  | RAM_LEN = 16K, CCMRAM_LEN = 4K)
stm32f334r8 (12K SRAM @0x20000000; 4K CCM SRAM @0x10000000  | RAM_LEN = 16K, CCMRAM_LEN = 4K)
```
Another question that arises: the the CCMRAM_LEN is used to declare a memory region in [stm32_common.ld](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/ldscripts/stm32_common.ld) but there is no section defined in the other linker scripts that makes use of this region. Shouldn't there at least be a "dummy"-section to allow placing stuff there with the help of `__section__` attribute? 
It would be even nicer if we could tell the linker that it should use both memory regions automatically for .bss / .stack or whatever. I think the linker is not able to do that though.

cladmi, 2018-07-18:
Handling two ram regions may be interesting to also into when doing https://github.com/RIOT-OS/RIOT/pull/9104

closed 2018-07-18 by kYc0o