RIOT-OS/RIOT #9546: dht: driver for dht11 sensor sometimes stuck in dht_read() on Atmel SAM R21

stegomon, 2018-07-11:
#### Description
Using the DHT11 temperature and air humidity sensor on the Atmel SAM R21 Xplained Pro, sometimes the function `dht_read` to read the value of the DHT11 sensor gets stuck.

#### Steps to reproduce the issue
Connect DHT11 sensor to a pin of the Atmel Board e.g. to Pin A07. 
Initialize the DHT11 sensor with the following code:

```
dht_t DHT_SENSOR;
DHT_SENSOR.type = (DHT11);
dht_params_t dht_params;
dht_params.pin = GPIO_PIN(0, 7);
dht_params.type = DHT11;
dht_params.in_mode = DHT_PARAM_PULL;
dht_init(&DHT_SENSOR, &dht_params);
```
Then, read the DHT11 sensor with the following loop:

```
while(1) {
    dht_read(&DHT_SENSOR, &air_temp, &air_hum);
    // output values to stdout
    // wait x seconds
}
```

#### Expected results
The values are read from the DHT11 sensor and output to stdout consistently.

#### Actual results
Exepcted results are observed for a indefinite time, varying from a few seconds to a few minutes.
Then, the `dht_read` function gets stuck and the thread stalls.

#### Versions

Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 5.5.0-12ubuntu1~16.04) 5.5.0 20171010
      arm-none-eabi-gcc: arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.2.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: missing
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.7.4
             coccinelle: missing

MichelRottleuthner, 2018-07-12:
Thanks for opening the issue! 
I think this happens because the data pin of the sensor is [polled](https://github.com/RIOT-OS/RIOT/blob/master/drivers/dht/dht.c#L54). If then, while polling, either a "long"-running ISR or another thread is scheduled (because of an interrupt), the driver will miss a flank and wait forever for a pin change that will not happen anymore. 
Two possible solutions: timeouts or interrupt-based decoding of the signals.

MichelRottleuthner, 2018-07-12:
@stegomon you can take a look at [this branch](https://github.com/MichelRottleuthner/RIOT/tree/dht_fix_lockup) for a possible solution. This served me well as a quick fix, but on some platforms the timing requirements of the sensor signaling may be a bit too tight to work in all cases. Thus, as stated above there should probably still be some kind of timeouts added just to be sure.

stegomon, 2018-07-21:
@MichelRottleuthner Thank you for your help.. however, using that branch reading the sensor is not possible at all. The problem causing this is the [mutex lock in read()](https://github.com/MichelRottleuthner/RIOT/blob/dht_fix_lockup/drivers/dht/dht.c#L77).

stegomon, 2018-07-23:
I am using the Atmel SAM R21 Xplained Pro with model number A09-2127 Rev03.

Unfortunately, adding ```mutex_init(&ctx.mutex);``` didn't change anything. Also changing the assert is not changing anything either.

Enabling DEBUG in mutex.h gives the following output:

```
# Trying to read DHT11 sensor now..
# PID[2]: Mutex in use.
# PID[2]: mutex_wait early out.
# PID[2]: Mutex in use.
# PID[2]: Adding node to mutex queue: prio: 7
# debug, stack too small
# Cannot debug, stack too small
# PID[2]: Mutex in use.
# PID[2]: mutex_wait early out.
# PID[2]: Mutex in use.
# PID[2]: Adding node to mutex queue: prio: 7
# mutex_unlock(): queue.next: 0x20000ad0 pid: 2
# mutex_unlock: waking up waiting thread 2
```
And then it stops. Hope this helps.

MichelRottleuthner, 2018-07-24:
Thanks for providing more information. The problem with the assert is only visible when DEVELHELP is enabled.
With samr21-xpro I am able to reproduce your issue. It looks like interrupt handling on this platform is already to slow for this approach. I just tested if it might be possible to optimize it further but even if you just increment a counter in the ISR it is to slow to catch every flank. I haven't verified if this is due to hardware limits or interrupt handling in RIOT, though. I think [here in the data sheet](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM-R21_Datasheet.pdf#page=335) would be the place for further investigation.
But if we are hitting HW-limits here already we can't use interrupts because there are even slower platforms. Using only timeouts would be the better option then.

maribu, 2019-07-23:
Should be fixed by https://github.com/RIOT-OS/RIOT/pull/11876. @stegomon or @MichelRottleuthner: Can you check that the issue is indeed fixed there?

closed 2019-07-29 by MichelRottleuthner