RIOT-OS/RIOT #9559: semtech_loramac: Enable event callbacks on driver

bergzand, 2018-07-12:
Semtech-loramac does not properly enable the RX and TX callbacks on the radios

From what I see in `pkg/semtech-loramac/contrib/semtech_loramac.c`, it uses (parts of) the netdev API to communicate with the radio. It should in the initialization, after initializing the radio, enable the event callbacks on the radio. This is similar to #9467, but for the semtech-loramac case.

I'm not completely certain which events should be enabled here, so that's why this is an issue and not a PR :smiley:

dylad, 2018-07-12:
@bergzand Sorry for the dumb question (not a netdev user) What's the goal of initialize this event ? I mean this pkg is working (maybe we are lucky) so what's behind this initialization for netdev ? How/why netdev need it ?

bergzand, 2018-07-12:
@dylad I know that for the ieee802154 devices the reported events (such as TX transfer complete) from the netdev interface back to the network stack are only reported if the flag for that event is set.
If this is not the case for Loramac devices, as in all events are always reported back, then this issue is invalid.

jia200x, 2018-07-13:
hi @bergzand 

enabling or disabling these callbacks won't have effects in LoRaWAN. The reason is because the LoRa radio doesn't provide [this condition] nor NETOPT_{TX|RX]_END_IRQ netopt getter/setter. So, it will always call the event_callback

bergzand, 2018-07-13:
@jia200x Thanks for confirming this! this issue is invalid then, sorry for the noise

closed 2018-07-13 by bergzand

jia200x, 2018-07-13:
> @jia200x Thanks for confirming this! this issue is invalid then, sorry for the noise
@bergzand absolutely no problem ;). Thanks to you