RIOT-OS/RIOT #9582: periph_*: Allow extensions for external periph adapters

kYc0o, 2018-07-17:
Many times, internal peripherals in a MCU are enough to handle basic operations, like GPIOs, ADCs, DACs, etc. However, for complex applications, extensions for such peripherals are needed, and actually widely used in industry.

Since RIOT aim is to keep the periph interfaces as slim and portable as possible, making extensions to them is indeed a complicated task. The aim of this issue is to explore the possible solutions to this, and to work together towards a common definition on how to write APIs and driver extensions for this use-case.

I'll summarize here the interfaces that need changes, and the PRs if there's one.

## GPIO
  - [ ] #9190 (PR)

## ADC

## DAC

ZetaR60, 2018-08-02:
My proposal for implementing this. Most of this can been seen in #9190, but there are a few small differences.

1. Any periph interface that is addressed with some form of ID reserves the leading bit of that ID for use by the extension interfaces. This includes of course, GPIO, ADC, DAC, but also perhaps even I2C, SPI, UART, etc. Making this a general policy for all of the addressed interfaces would make it easy to add things like bridges for buses, for example. If the leading bit is a zero, the ID is intended to be used by the CPU, if it is a one then it is intended to be used by an extension interface.
2. If the identifier indicates that the resource being requested is an extension device _and_ extensions are being compiled for that interface, then the API call will be redirected to the relevant extension interface driver.
3. Redirections are implemented by splitting the periph API in two: one that is identical to the current list of functions to be called by users of the API, and one that has ```_ll``` (short for low-level) appended to it (for use by CPU implementations). The low-level functions are called by an inline function from the header that also contains the redirection code. When the extension interface for that periph API is not enabled, the redirection code is not included and so the inline should compile down to be exactly the same as if the extension interface did not exist.
4. The redirect code consists of three operations per API call: check to see if the leading bit is one, call a function ```foocall_redir```, and return (which may return the value from ```foocall_redir```, if the original API call uses a return value). When using CPU resources with the extension interface enabled, this only adds a single check to the overhead.
5. The redirect function ```foocall_redir``` is implemented as part of the extension interface, and figures out the proper callback and device descriptor based on identification information in the resource's address. For instance, a ```gpio_t``` would have the leading bit indicate that the GPIO is on an extension interface, then the next 7 bits would indicate which device it is on, and the rest of the ```gpio_t``` would be passed as the actual "pin" to the driver implementation.
6. Extension interface headers would live in ```drivers/include/extend```. These headers would be included for: drivers that implement callbacks for an extension interface, and files that are directly referencing a resource, e.g. using a facility similar to GPIO_PIN(x,y).
7. Drivers that provide callbacks for an extension interface do so in a manner similar to SAUL: a file ```foodriver_barinterface_extend.c``` defines a function for each of the API calls. Each of these functions have arguments and return values that are identical to the API calls _except_ that the first argument is for a void pointer to the device descriptor for the device used. The file also defines ```foointerface_ext_driver```, which contains callbacks for each of the API functions.
8. The list of devices providing extension services is defined in ```foointerface_ext_conf.h```. This file is provided by the contributor of the board port, and consists of an array of pointers to the device drivers and their respective device descriptors. Specific driver/descriptor pairs can be referenced as the nth item in this list. It also should also provide an enum that names each item in the list for convenience.

This ended up a bit long and complicated, but I guess it has to be for a new interface definition proposal. Please let me know where I have been unclear, left things out, or otherwise needs to be changed.

kYc0o, 2018-09-19:
I think this is going to the right direction. The PRs you proposed address one by one the issues, then we can reach consensus there. Thanks a lot @ZetaR60 !

gschorcht, 2018-11-19:
@ZetaR60 When I was writing the driver for the PCF857X I2C I/O Expander (PR #10430), I was wondering if it wouldn't be possible to completely hide that these GPIOs are being implemented by an extender to use them with standard peripheral GPIO functions. I was thinking about a concept of having GPIO providers and a registry that manages all GPIO providers and redirects function calls based on the pin number. Yes, it sounds like your concept with the only difference that I supposed that the MCU is also a GPIO provider.

@kYc0o pointed me to your PRs. Great work and good to hear that someone is already working on it.

I have some comments, questions and maybe suggestions. I went through all the commits and tried to read and understand all discussions and requests for change. Please forgive me if I ask or suggest something that has already been discussed or decided otherwise. I'm not doing it in the source code since it is more general for all files.

1. I'm not quite lucky with the naming scheme. From my point of view, it seems to be a bit inconsistent which makes the usage of the API more difficult. You have<br>
`gpio_<function>_ll`, 
`gpio_<function>_redir`
`gpio_exp_<function>_notsup`<br>
Why not call `_ll` functions what they are, namely` _mcu` functions, and why not call redirection functions `_ext` functions? From my point of view, extender GPIO functions are also low-level functions, and the redirection functions do nothing else than calling extension functions. If suffixes are used, maybe a suffix  should also be used for `gpio_ <function> _ext_notsup`:<br>
`gpio_<function>_mpu`, 
`gpio_<function>_ext`
`gpio_<function>_ext_notsup`<br>
Drivers use probably a prefix, e.g., `pcf857x_gpio_<function>`. Therefore, another naming scheme could be to use prefixes for all the functions like `<driver>_gpio_<function>` <br>
`mpu_gpio_<function>`, 
`ext_gpio_<function>`
`ext_gpio_<function>_notsup`<br>

2. In item 6 you propose "The list of devices providing extension services is defined in `foointerface_ext_conf.h`. If I understand it right, it  would be part of the board definition. So it would only work for extenders that are already on board. Even more critical would be that the extender device definitions would be fix for a certain board. But what if you have a standard board, lets say an Arduino Zero, and you want to add a GPIO extender. Or more worst, there is an extender on board and you want to add a further extender of same type to your application?<br>
Wouldn't be a registry better to which devices register them self when their driver module is loaded? I know, @kaspar30 didn't like to add a further registry, but a registry could solve this and other problems.

3. I have not seen a solution for the GPIO_NUMOF define. A number of applications use this definition to iterate over all GPIOs. This is even more critical for ADCs and DACs. Again, a registry could solve this problem by having a registry function`gpio_get_numof` which returns the sum of the number of GPIOs over all GPIO devices including the MCU. The GPIO_NUMOF define would be a call of this function.

That's it for the moment. I'm sure, that more questions will come up when I try to integrate an extender driver according to your approach. When I have time in next days, I could try it for the PCF857X driver.

ZetaR60, 2018-11-23:
@gschorcht Thank you for your interest! I appreciate your comprehensive look at this system.

About the naming of the functions: Here is my reasoning on why I chose the suffixes that I did:
1. I considered CPU functions to be "low-level" in that drivers should always be able to rely on them, even if they are extension drivers. It is perfectly legitimate with the way the extension system is written for a GPIO expander to use a GPIO on another device, so long as there are no circular dependencies. CPU GPIOs are then the lowest in the possible stack of dependencies amongst the GPIO expanders. Some GPIO devices will require the use of other GPIOs for controlling them. Also, while it might seem a bit strange at first to have GPIO expanders attached to other GPIO expanders (rather than the CPU), but this allows you to do interesting things like build field extensible buses using simple logic (like with 7400 series shift registers). Other periph drivers have similar considerations.
2. The redirection functions are so named because that is really the only thing they do: find with extension device is being requested and redirect the call and arguments as a call to that device. The redirection functions don't actually implement the extended function (that would be in a separate driver for each device type). See ```drivers/extend/gpio_redir.c``` from #9958 for an example.
3. The not-supported functions have a fairly self explanatory name.

About having a extension registry:
- Firstly, to answer your questions about the way ```foointerface_ext_conf.h``` currently works. Usually it would be present in the board definition, but it is also over-ridable by the application by including its own header for it. See ```drivers/extend/include/gpio_ext_conf.h``` from #9958 for an example. There is room here for some improvement, such as the possibility of extending the configuration file without completely overriding it. Most basic cases for extension devices are currently served however.
- Secondly, unfortunately there are a few problems here that would make a registry impractical to implement. Any access to an extension device must be done using an ID that is unique to that device, and is able to be assigned at compile time. That ID must be a very small integer, preferably 7 bits at most, so that it can be stuffed into existing device IDs. Also, the ID cannot be assigned on a per-driver basis, because a board may have multiple extension devices on it that use the same driver. Basically, by having a static array containing the configuration and an enum to resolve names into ID numbers, this whole potential mess is bypassed.

About ```GPIO_NUMOF```: Citation needed? ```grep -R GPIO_NUMOF .``` does not reveal much use of ```GPIO_NUMOF```. The closest thing is SAUL's GPIO configuration, which is compatible with the extension system. So far as I know, there is no way to enumerate GPIOs because their IDs are usually generated non-sequentially by ORing the CPU port number and pin number. The extension system basically does the same thing.

Also see #9054. The PR needs some updates, and I have been putting it off until the extension interface is merged. I also need to see how different your #10430 is from the devices supported by #9054.

gschorcht, 2018-11-25:
@ZetaR60 Thank you for your comprehensive answers. I really like consistent naming schemes, which are either suffixes or prefixes and no mixes. The `gpio_ext` functions seemed a bit inconsistent because the name extension is in the middle. Now that I have used the GPIO extension API in practice with my PCF857X driver, I can live with them the way they are because the user doesn't see them. 

> Firstly, to answer your questions about the way `foointerface_ext_conf.h` currently works. Usually it would be present in the board definition, but it is also over-ridable by the application by including its own header for it.

Yes, I know. I don't really like this kind of static definitions at compile time for the following reasons

- How to mix the definition of extension modules that are already on board and modules that are added by the application developer? The application developer has to copy `foointerface_ext_conf.h` to his application directory to extend it by his own entries.
- There is a copy of the static const array in each module which includes the header file. This is an overall design question in RIOT that is not related to the GPIO extension API.

I know, using static configuration arrays that are defined at compile time are very efficient during run-time and probably the problems above are the price we have to pay.

I can (probably I have to) live with it since there is no other choice for the moment.

> About GPIO_NUMOF: Citation needed? grep -R GPIO_NUMOF . does not reveal much use of GPIO_NUMOF. 

Although this does not seem to be a real problem for the GPIO extension API, it becomes a problem for other extension APIs. For example, the DAC and ADC test programs rely on the `DAC_NUMOF` and` ADC_NUMOF` definitions. I have DAC and ADC expander modules here, and I planned to begin developing drivers for them in the near future. I could use this development to think about an ADC / DAC extension API similar to the GPIO extension API.

gschorcht, 2018-11-25:
> Also see #9054.

If you want, I can review it when it is ready from your point of view.

ZetaR60, 2018-11-25:
> The ```gpio_ext``` functions seemed a bit inconsistent because the name extension is in the middle.

I think this is just a matter of perspective. The naming convention I was using forms a big-endian conceptual hierarchy. As an example: ```gpio_ext_read_redir```. Here ```gpio``` is the top of the hierarchy, naming which specific periph system it is a part of; ```ext``` specifies that it is part of the extension subsystem (empty if it is part of the root system); ```read``` specifies which API call it belongs to; and finally ```redir``` specifies that it is part of the call that redirects to an extension driver. Reordering the naming will change the conceptual hierarchy to be of mixed endinness.

> How to mix the definition of extension modules that are already on board and modules that are added by the application developer? The application developer has to copy foointerface_ext_conf.h to his application directory to extend it by his own entries.

What do you think of the addition of another config header (```gpio_ext_conf_app.h``` perhaps) that is included by ```gpio_ext_conf.h```? There can then be a macro that is put into ```gpio_ext_list``` that is defined in the second header.

> There is a copy of the static const array in each module which includes the header file. This is an overall design question in RIOT that is not related to the GPIO extension API.

Only ```drivers/extend/gpio_redir.c``` should include ```gpio_ext_conf.h```, for this exact reason (SAUL uses conf headers similarly). There needs to be an enum defined so that the extension devices can be referenced by name rather than position in the array, but this should not be done in ```gpio_ext_conf.h``` (I don't think I have specified a place for this yet, mainly because there are no boards using the extension interface yet).

> Although this does not seem to be a real problem for the GPIO extension API, it becomes a problem for other extension APIs. For example, the DAC and ADC test programs rely on the DAC_NUMOF and [...]

As far as updating ```FOO_NUMOF``` is concerned; I think we could have a ```FOO_NUMOF_LL``` that is part of the board definitions, and then if ```FOO_NUMOF``` is undefined it is set to equal ```FOO_NUMOF_LL```.

The bigger problem is that the device numbers are non-sequential, but this seems to already be partially taken care of with the ```FOO_DEV(x)``` macro. I think we can have a ```FOO_DEV_LL(x)``` macro similar to the above, and that should solve the automatic access problem.

gschorcht, 2018-11-26:
I tried to reaalize a DAC extension API according to @ZetaR60's GPIO extension API. It uses exactly the same principles and follows the items 1 to 8 of the proposal.

The problems I had:
- `DAC_LINE(x)` does not divide a particular DAC line into a port and a channel. That is, there is no easy  and efficient way to use the port identifier as device reference.
- Devices can have a different number of channels which makes a linear channel identifier scheme difficult if not impossible. So it will not be possible to iterate over the DAC lines using the `DAC_NUMOF` macro.

 What are the options?
- We define `DAC_EXT_LINE(x)` that has only one parameter as `DAC_LINE(x)` and leave the user alone with defining valid identifiers that refer the right device and right channel.
- We define `DAC_EXT_LINE(x, y)` where x refers the external device and y the channel. This option would be incompatible with the existing `DAC_LINE` definition but would support the user and is as efficient as the GPIO_EXT_PIN(x) macro.
- For each external DAC device, the number of channels is defined in its DAC extension list entry. `DAC_EXT_LINE(x)` would be nothing else than x and the `dac_ext_entry` function would sum up these number of channels starting with the `DAC_NUMOF` from low level interface to determine the device and the channel for line x. In that case
    * `0...DAC_NUMOF-1` would refer the channels of the internal DAC,
    * `DAC_NUMOF... DAC_NUMOF+3` would refer the channels of the first external 4-channel device, 
    * `DAC_NUMOF+3...DAC_NUMOF+4` would refer the channels of the second 2-channel device
    *  ...

  The iteration over DAC line would become possible but the problem would be the efficiency of the approach.

Any suggestion?

ZetaR60, 2018-11-27:
> - DAC_LINE(x) does not divide a particular DAC line into a port and a channel. That is, there is no easy and efficient way to use the port identifier as device reference.
> - Devices can have a [...]

The documentation is not very clear on this, but I assume ```FOO_LINE(x)``` is supposed to map the (n-1)th line to its address. This would allow you to iterate over each line up to line ```FOO_NUMOF - 1```. This macro should definitely reduce to some compile-time resolvable logic, so that it is possible to use in definitions and such that are outside of functions. The question is simply: what is the best way of doing this so that it is still pretty simple and clear what it is doing? I think you have some good starting points listed out.

Here is a suggestion for ```FOO_LINE(x)```:
```
#ifndef FOO_LINE_LL
#define FOO_LINE_LL(x)    (x)
#endif

#define FOO_LINE(x)    (x < FOO_NUMOF_LL \
                        ? FOO_LINE_LL(x) \
                        : FOO_EXT_LINE(x - FOO_NUMOF_LL) )
```

Where ```FOO_LINE_LL(x)``` can be defined in ```cpu/*/include/periph_cpu.h```. ```FOO_EXT_LINE(x)``` is then defined with the board files, in the same place as the enum for resolution of device names. Since on-board FOO devices are known in the board definition files, it can just be a static list of conditionals. Also needed is a macro that resolves device + line tuples into the combined address (basically the same as ```GPIO_EXT_PIN(x, y)```), which I will call ```FOO_EXT_LINE_REV(x, y)``` below (there is probably a better name).

```
#define FOO_DEVLIST_1    FOO_BAR1DEV_NUM
#define FOO_DEVLIST_2    (FOO_DEVLIST_1 + FOO_BAR2DEV_NUM)
#define FOO_DEVLIST_3    (FOO_DEVLIST_2 + FOO_BAR3DEV_NUM)

#ifndef FOO_EXT_LINE_APP
#define FOO_EXT_LINE_APP(x)    FOO_UNDEF
#endif

#define FOO_EXT_LINE(x)    (x < FOO_DEVLIST_1 \
                            ? FOO_EXT_LINE_REV(1, FOO_DEVLIST_1 - x) \
                            : (x < FOO_DEVLIST_2 \
                            ? FOO_EXT_LINE_REV(2, FOO_DEVLIST_2 - x) \
                            : (x < FOO_DEVLIST_3 \
                            ? FOO_EXT_LINE_REV(3, FOO_DEVLIST_3 - x) \
                            : FOO_EXT_LINE_APP(x) \
                            )))
```

A bit more complex than we usually want in a macro, but it can be a starting point at least. All of that logic should be taken care of at compile time, and it should be safe to use anywhere where you would use the existing macros.

Thoughts? If you prefer, I can write the DAC and ADC extension interfaces.

gschorcht, 2018-12-17:
@ZetaR60 Many thanks for your the very precise summary of what we are looking for and your analysis what the problems are. I completely agree with that. Let me comment some items.

> 1. A way for each API of mapping identifying information into a tuple that allows device access, e.g. like GPIO_PIN(x,y).

When I was reading this, I was also wondering why we don't have a macro like `ADC_LINE(i)` or `DAC_LINE(i)` for GPIOs. You called it inverse function in #10504. The concept behind seems to be at least very similar if not the same. As you said, the purpose of these macros is to map an ordinal to the device/channel tupel. Why don't we have something like that for GPIOs which maps an ordinal to the port/pin tuble? I know, a number of boards label their GPIOs like the port/pin tubles of the MCU. But, there are also a number of board that just label their GPIOs with an ordinal.

When I was reading this, I was wondering why we don't have a macro like `ADC_LINE(i)` or `DAC_LINE(i)` for GPIOs. The concept behind seems to be at least very similar if not the same. As you said, the purpose of these macros is to map an ordinal to the device/channel tupel. Why don't we have such a macro for GPIOs which maps an ordinal to the port/pin tuble? You called it inverse function in #10504. I know, a number of boards label their GPIOs like the port/pin tuples of the MCU. But, there are also a number of board that just label their GPIOs with an ordinal.

So, I'm quite sure that we should also have a macro that maps an ordinal to port/pin tupel (the inverse function). This would help users for boards that use ordinals as labels and would also help to solve item 3 for GPIOs. But, I would call it rather simply `GPIO` instead of `GPIO_LINE` or `GPIO_PIN_INV`?

> 2. A way to determine the number of pins / lines / etc, similar to FOO_NUMOF.

Yes. Although the total number of expansion pins/lines for the first version of the extension APIs does not seem to be necessary, I strongly recommend to have at least the per device number of extension pins/line in the `foo_ext_list` already in the first version. This is required, for example, for parameter checking and error handling.

> 3. A way to map an ordinal to the tuple that allows device access, which may be what the FOO_LINE(x) macros are intended to do (but the documentation is unclear). This allows looping over each pin / line / etc.

Yes, the question is whether it is really needed in real projects to iterate over all pins/lines of all devices. It is useful for testing to iterate over all ADC, DAC, or PWM channels to see whether they work properly. But how reasonable is that for real applications? The only real applications I could imagine are to initialize all GPIOs as inputs, for example, or to set all PWM channels to same duty cycle because all of them are driving the LEDs of the same lamp.

> For 2 and 3, there seem to be two problems:
>
> - Various things rely on these things being macros, because they are used where function calls would be illegal. I don't think we should try to change this behavior, because this would require extensive and invasive changes.
> - While 2 and 3 can be defined as macros, it is awkward and complex to do so.

Exactly.

> So therefore, I propose:
>
> That macro implementations of 2 and 3 are kept for the CPU only, e.g. FOO_NUMOF would define the number of CPU pins/lines.

Agreed.

> - That an entry for the number of pins/lines on a given extension device is added to foo_ext_list[] in foo_ext_conf.h.

This is urgently required, also for the first version of the extension APIs.

> - That an inline function for calculating the number of total pins/lines from FOO_NUMOF and foo_ext_list[] is added as part of the periph APIs. If extension devices are not enabled, this will be just return FOO_NUMOF, else it will return the total on all devices.

Even if we give up on the previous approach that everything is known at compile time, it is absolutely fine for me to have a function to calculate the total number of pins/lines.

> - That a function is added to the periph APIs for mapping an ordinal to the tuple for the nth pin/line, for the CPU and all extension devices. This should be fairly easy to calculate for everything except periph/gpio, due to its use of ports.

Yes. I will think about it for `periph/gpio`.

gschorcht, 2018-12-17:
@ZetaR60 I'm really sorry, but I have to come back to the naming. After writing the extension APIs for ADC, DAC, and PWM and realizing true extension drivers for all of the extension APIs, I am still not satisfied with their naming scheme. It still seems inconsistent to me.

On one hand we have the macros:

`FOO_LINE_LL` for features provided by the MCU
`FOO_EXT_LINE` for features provided by extensions 

For example, this naming scheme, which does not consistently use suffixes, leads to:
```C
#define ADC_LINE(x)         (x < ADC_NUMOF \
                             ? ADC_LINE_LL(x) \
                             : ADC_EXT_LINE(x - ADC_NUMOF))
```
In addition, in my view, the use of "MCU" as a suffix would make it clearer which component provides the feature. The same macro would then looks much more consistent:
```C
#define ADC_LINE(x)         (x < ADC_NUMOF \
                             ? ADC_LINE_MCU(x) \
                             : ADC_LINE_EXT(x - ADC_NUMOF))
```

On the other hand, we have the functions

`foo_function_ll` for functions provided by the MCU
`foo_function_redir` for functions provided by extensions
`foo_ext_function_t` for function prototypes of extensions

Although the function `foo_function_redir` redirects the call, from the user's point of view the extension function is called. The consistent use of suffixes and the suffix MCU for functions provided by the MCU seems to be clearer from the point of view of the user.

`foo_function_mcu` for functions provided by the MCU
`foo_function_ext` for functions provided by extensions
`foo_function_ext_t` for function prototypes of extensions

gebart, 2018-12-22:
The thing with ADC_LINE and DAC_LINE is that it may not be useful to create DAC_PIN(port, pin) macros for them since the MCUs may only have a handful of possible pins where to map the analog functions to, unlike gpio which usually is mappable to almost any pin. The mapping from package pin to adc module and internal adc channel number may not have a simple function but may require a look up table, and it may not even be 1:1. Some CPUs can direct the analog function of one package pin to more than one ADC module internally. How would ADC_PIN(PORT_B, 3) decide in that CPU if it should use for example ADC0 channel 2 or ADC1 channel 8? It is only the application designer that can know if it is better to use ADC0 or ADC1 in that situation.

gschorcht, 2018-12-22:
@ZetaR60 

>> When I was reading this, I was also wondering why we don't have a macro like `ADC_LINE(i)` or `DAC_LINE(i)` for GPIOs.

> I think we should. I suspect we don't have one simply because it would be more complex to write for the GPIO encodings. I think it is a good case for a new PR.

Agreed.

> I feel that `GPIO` should be avoided, because the macro name is too simple and broad.

Yes, that's clear.

> Another option might be something like GPIO_PIN_LIST(i), which would be much clearer but would be different than the naming convention with ADC/DAC.

We should keep that in mind for further PR. We could open an new issue not to forget it or we could add a ToDo list to this issue.

>> For example, this naming scheme, which does not consistently use suffixes

> I am a bit confused as to why you feel it is inconsistent after my explanation (rather than simply different than how you would have chosen to organize the names). I feel that putting EXT at the end rather in the middle would be like having a mixed endianness number.

Ok, I read your explaination again and I tried to understand your intention of the hierarchical concept. And yes, it is probably a question of different perspectives. If I understand you correctly, you consider MCU GPIOs as part of the root and extension GPIOs as a subsystem. I guess because extensions are connected to MCU GPIOs. Following this hierarchy, extensions would have to be called `gpio_ext_` and MCU GPIOs would have to be called `gpio_` but without any suffix.

I would argue differently. If we consider it as hierarchy, we have the peripheral system and inside of it all the peripheral subsystems GPIOs, ADCs, ... In each peripheral subsystems, we can have further subsystems, the subsystem provided by the MCU and the subsystem provided by extensions. That is, I would consider the MCU not as root but also as a subsystem. The naming hierarchiy would then be `periph/gpio_mcu_*` and `periph/gpio_ext_*` or `GPIO_MCU_PIN` and `GPIO_EXT_PIN`.

Since the middle name seems to be impractical, I would use suffixes subsystem identifiers. Finally
```c
#define FOO_BAR(x)         (x < FOO_NUMOF_MCU \
                            ? FOO_BAR_MCU(x) \
                            : FOO_BAR_EXT(x - FOO_NUMOF_MCU))
```
looks more cool than
```c
#define FOO_BAR(x)         (x < FOO_NUMOF_MCU \
                            ? FOO_BAR_MCU(x) \
                            : FOO_EXT_BAR(x - FOO_NUMOF_MCU))
```
:wink:

> It might at first make more sense to give `foo_function_redir` the name `foo_ext_function_redir`. However, the function prototype is in `periph/foo` rather than `extend/foo`, so it should not have `_ext`.

Ok, that's clear. Do we have the same problem with macro definitions which also have `_EXT` in their names?

As I already said before, whatever you decide, I can live with it and you made a very good job :smile: I just wanted to suggest to think about the different variants, because it's a definition of a basic interface that will last for a long time. Once it is committed, it will be hard to change it again. I have proven with real drivers that the interface you have defined works.

gschorcht, 2018-12-22:
@gebart 
> It is only the application designer that can know if it is better to use ADC0 or ADC1 in that situation.

In reality, the mapping is defined by the board developer in static const arrays. The only chance for the application developer to change this mapping is to overwrite the complete `periph_conf.h` of the board definition by an application-specific version of `periph_conf.h`.

To have the possibility to address a channel of a certain ADC device, I proposed to change the `ADC_LINE` and `DAC_LINE` macros in PR #10504 to `ADC_LINE(dev,channel)` and `ADC_LINE(dev,channel)`. I had to learn that it was not a good idea to change this line concept since it is used by the board developer to map a line to the tuple (device, channel).

The other way round, it might be useful to access a GPIO by an ordinal in ascending order, whatever the name of such a macro would be.

closed 2020-02-26 by stale[bot]