RIOT-OS/RIOT #9589: application/Makefile: environment settings after inclusion of Makefile.include

miri64, 2018-07-17:
#### Description
Environment variables that are set after the inclusion of `Makefile.include` in an application's Makefile seem to be ignored. This results in e.g. the default channel not to be configurable via environment variables in `gnrc_networking`

#### Steps to reproduce the issue
Try to compile the `gnrc_networking` example with `DEFAULT_CHANNEL` set to a different value than the actual default.

#### Expected results
The channel should be changed

#### Actual results
The channel is the same

#### Versions
a0a31120b of RIOT.

cladmi, 2018-07-17:
I think that it is just that it needs to be cleaned between builds. Which is, I agree, a problem.

miri64, 2018-07-17:
Yepp, a clean between builds "fixes" the problem ;-).

cladmi, 2018-07-17:
It is true for all `CFLAGS` modified after including `Makefile.include`. Replacing by `CFLAGS_WITH_MACROS` would help there. I will do something about it.

jcarrano, 2018-07-25:
@cladmi Why does it matter if the definition is made before or after including `Makefile.include`?

cladmi, 2018-07-26:
It is used but does not trigger rebuild through the `$(RIOTBUILD_CONFIG_HEADER_C)`

jcarrano, 2018-07-27:
@cladmi Any reason for the current behaviour of:

cladmi, 2018-07-28:
From searching history with `git log -G CFLAGS_WITH_MACROS` it was introduced by `77a15e7886c978161fb1bf442cc5a82226571cd2` which comes from https://github.com/RIOT-OS/RIOT/pull/5097 PR. It was both to fix the rebuild issue and clean the output https://github.com/RIOT-OS/RIOT/issues/5092.

To keep both introduced behaviors, without fixing any other non rebuilding issue for the moment, I would go for **option C**, `-D` are already stored in `$(EXTDEFINES)`, they could be not added to `CFLAGS` and have `CFLAGS_WITH_MACROS = $(CFLAGS) $(EXTDEFINES)`.

With this minimal change, the only consequence is that now only the `-DMODULE_` will not be in the command line but others `-D` defined in the Makefile (`BOARD_`, `CPU_ARCH`, others) will be kept in the CFLAGS and so appear on the command line.

By renaming `CFLAGS_WITH_MACROS` to something more explicit `CFLAGS_DEFINES_NON_VISIBLE_ON_COMMAND_LINE` there could be a way to add defines that would not be shown on the cli and passed to C by the `RIOTBIULD_CONFIG_HEADER_C` if it is really problematic.

I got it working with this minimal change, some renaming could help.

``` diff
diff --git a/Makefile.include b/Makefile.include
index a809d2291..2d4403b53 100644
--- a/Makefile.include
+++ b/Makefile.include
@@ -638,16 +638,14 @@ $(RIOTBUILD_CONFIG_HEADER_C):
        @mkdir -p '$(dir $@)'
        $(Q)'$(RIOTTOOLS)/genconfigheader/genconfigheader.sh' '$@' $(CFLAGS_WITH_MACROS)

-CFLAGS_WITH_MACROS := $(CFLAGS)
+CFLAGS_WITH_MACROS = $(CFLAGS) $(EXTDEFINES)

 ifneq (,$(RIOT_VERSION_OVERRIDE))
-export CFLAGS_WITH_MACROS += -DRIOT_VERSION=\"$(RIOT_VERSION_OVERRIDE)\"
+export CFLAGS += -DRIOT_VERSION=\"$(RIOT_VERSION_OVERRIDE)\"
 else
-export CFLAGS_WITH_MACROS += -DRIOT_VERSION=\"$(RIOT_VERSION)\"
+export CFLAGS += -DRIOT_VERSION=\"$(RIOT_VERSION)\"
 endif

-CFLAGS := $(patsubst -D%,,$(CFLAGS))
-CFLAGS := $(patsubst -U%,,$(CFLAGS))
 CFLAGS += -include '$(RIOTBUILD_CONFIG_HEADER_C)'

 # include mcuboot support
diff --git a/makefiles/modules.inc.mk b/makefiles/modules.inc.mk
index 7dd0575b3..8bf65e065 100644
--- a/makefiles/modules.inc.mk
+++ b/makefiles/modules.inc.mk
@@ -1,4 +1,5 @@
 ED = $(addprefix MODULE_,$(sort $(USEMODULE) $(USEPKG)))
+# EXTDEFINES will be put in CFLAGS_WITH_MACROS
 EXTDEFINES = $(addprefix -D,$(shell echo '$(ED)' | tr 'a-z-' 'A-Z_'))

 # filter "pseudomodules" from "real modules", but not "no_pseudomodules"
@@ -6,6 +7,4 @@ NO_PSEUDOMODULES := $(filter $(NO_PSEUDOMODULES), $(sort $(USEMODULE) $(USEPKG))
 REALMODULES = $(filter-out $(PSEUDOMODULES), $(sort $(USEMODULE) $(USEPKG))) $(NO_PSEUDOMODULES)
 export BASELIBS += $(REALMODULES:%=$(BINDIR)/%.a)

-CFLAGS += $(EXTDEFINES)
-
 export USEMODULE
```

It would solve this issue the following would correctly rebuild

    make BOARD=iotlab-m3 QUIET=0 DEFAULT_CHANNEL=11
    make BOARD=iotlab-m3 QUIET=0 DEFAULT_CHANNEL=26

and also fix https://github.com/RIOT-OS/RIOT/issues/5776

cladmi, 2018-07-28:
I can provide a PR with this but may like some feedback on naming and such before.
@jcarrano @gebart

closed 2019-09-28 by aabadie