RIOT-OS/RIOT #9594: Q: Is it OK to unlock a mutex on a different thread from which it was locked?

jcarrano, 2018-07-18:
This question was triggered by the use of a mutex as a signalling device in #9588.

In many OSes (POSIX, Windows), mutexes cannot be released by a thread that did not lock them. RIOT seems to allow that, and thus RIOT's mutex seems to be a more general synchronization mechanism.

If this usage is supported, it should be documented and guaranteed to work. If it is not, there should be a warning to avoid that kind of usage.

kaspar030, 2018-07-18:
> Is it OK to unlock a mutex on a different thread from which it was locked?

Yes, a mutex can be unlocked from any thread or ISR.

kaspar030, 2018-07-18:
Should we close again?

OlegHahm, 2018-09-12:
Just a small question:
> In many OSes (POSIX, Windows), mutexes cannot be released by a thread that did not lock them. 
That's new for me. As far as I know the POSIX specification does not require you to call `pthread_mutex_unlock()` from the same thread as the mutex was locked. Whether or not that is a good idea, is a different question. ;)

jcarrano, 2018-09-14:
@OlegHahm http://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_lock.html 

Sometimes it is an error and sometimes an undefined behavior.
I'm of the opinion that if I had doubts when reading the docs it is probable that other people too and if other systems document these kind of behaviors it is a good idea t do it here too.
When I have some time I will complete #9600 (unless someone wants to beats me to it).

OlegHahm, 2018-09-14:
I'm all in favor of documenting the behavior. And thanks for the pointer I was not aware of this restriction in POSIX.

gebart, 2018-09-16:
A RIOT mutex provides what is usually known as a binary semaphore, so it is a little more general than what most people mean by mutex.

kaspar030, 2018-09-16:
> A RIOT mutex provides what is usually known as a binary semaphore, so it is a little more general than what most people mean by mutex.

Yup. It is also sometimes just called a "lock" (e.g., in Python).

PeterKietzmann, 2018-09-25:
@jcarrano question answered - can we close the issue again?

miri64, 2018-09-27:
> @jcarrano question answered - can we close the issue again?

There was the request for a documentation update also buried in the question. Is that addressed?

miri64, 2018-09-27:
It is fixed in https://github.com/RIOT-OS/RIOT/pull/9600 but that is not merged yet.

closed 2019-09-10 by stale[bot]