RIOT-OS/RIOT #9602: SX1276 UART driver?

Citrullin, 2018-07-19:
Hey,
I can only find the SPI driver. Is there also a way to use a UART module? Like the E45-TTL-100?
Or do I need to add an own driver for it?

dylad, 2018-07-19:
Hello @Citrullin,
As I know, we currently support SX127x SPI IC but also rn2xx3 (UART) from Microchip, there is also this PR  #9289 for mkrwan.
@aabadie or @jia200x can correct me if I forget one.
If you plan to use another chip, you may have to create your own driver (but you can based your work from rn2xx3 I guess ?)

Citrullin, 2018-07-19:
I soldered a SPI SX1276 module this night. Still struggling with the configuration for my bluepill. But that's another topic. Thanks for your answer, I take later a look into it. Maybe I can add support for it to Riot.

closed 2018-07-19 by Citrullin