RIOT-OS/RIOT #9619: ATmega platform issues

maribu, 2018-07-20:
Hi,

it seems something is broken with ATmega platform, but it is hard to tell what.

# Symptoms
## Using `BUILD_IN_DOCKER` or Ubuntu's (latest LTS) toolchain
Basically, nothing works, even `example/hello-world` never produces any output on the terminal

## Using Alpine Linux or the Arch Linux toolchain
- `example/hello-world` works on both Arudino UNO and Arduino Mega2560
- `example/default` works only on Arduino Mega2560, on Arduino UNO the main thread is never started (checked via debugWIRE and a breakpoint). The `idle` thread seems to be running correctly
- If I connect the Ethernet Shield to the Arduino Mega2560, it the `exmaples/default` app stops working
- Before the update of `avr-gcc` from version 6.3 to 6.4 on Alpine Linux, the toolchain had the same issues as the one of Ubuntu or the one used in the `riot/riotbuild` docker image

## Tracing the problem on the Mega2560
First, I added those lines to the `Makefile` in `examples/default` in order to use the Ethernet Shield:
``` Makefile
BOARD_PROVIDES_NETIF += arduino-mega2560
USEMODULE += w5100
CFLAGS += -DW5100_PARAM_CS=ARDUINO_PIN_10
CFLAGS += -DW5100_PARAM_EVT=ARDUINO_PIN_2
```

The last function reached when the Mega2560 is booting is `void gnrc_pktbuf_init(void)` in `sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c`. More precisely: If I add an breakpoint before the `mutex_lock(&_mutex);`, this breakpoint gets triggered. If I put it after the line, it is no longer reached. If I interrupt the program the `idle` thread is always running. It seems that `mutex_lock` calls the scheduler, which switches to the idle thread and never comes back. This would make sense if the mutex was locked: As no other thread is running (afak) during `auto_init()` except for the `idle` thread, `idle` would be the only runnable thread and the main thread never becomes runnable again, as the mutex remains locked. But the mutex cannot be locked at that point...

# Hints for debugging the boards (skip this section if you don't want to reproduce my findings)
The Arduino Mega2560 can be debugged with JTag, but it speaks a dialect of JTag that is only understood by propriatary and expensive boards. Luckily, there is an open source software (AVaRICE) to interface with those boards. Sadly, it is not really maintained any more and not in a really good shape.

The Arduino UNO uses debugWIRE for debugging. The good thing is that debugging is possible using an actively maintained open source software [dwire-debug](https://github.com/dcwbrown/dwire-debug) and cheap hardware (see the software for details). (Use the `gdbserver` command to allow attaching with GDB.) AVaRICE and one of the propriatary board like the AVR Dragon also work, but that solution seems to be less reliable compared to dwire-debug. But: The debugWIRE interface is very slow and limited compared to JTAG and only a single hardware break point can be set. AVaRICE tries to insert break instructions into the machine code, but apparently fails to do so. dwire-debug uses the single hardware breakpoint. Beware: Use the command `bc` to clear the break point from within `dwire-debug` before starting the `gdbserver`, if it has been previously set. The break point seems will not be unset by detaching GDB or dwire-debug. Also beware: Stepping does not work correctly, neither with AVaRICE nor with dwire-debug. (Stepping with AVaRICE and debugWIRE simply fails, stepping with dwire-debug seems to increment the program counter by 2 each step and ignores any conditional jump.)

Summing up: On-chip debugging of ATmega boards sucks.

# Conclusion
It could be possible that the toolchain of the `riot/riotbuild` docker image is broken. Also, it could be possible that there is some strange bug in RIOT, which is triggered more reliable when build with `riot/riotbuild`, and less reliable with the current Alpine Linux and Arch Linux AVR toolchains. Honestly, I have no idea what is going on

gebart, 2018-07-21:
Can you inspect memory with the limited debugger? If so, check what the memory contents of the mutex is before the call to mutex lock. Also, check that the data and bss initialization is OK

dylad, 2018-07-21:
Could you also try with another toolchain version ? We already encountered issues with the arm-gcc from lastest ubuntu. Maybe this is the same for AVR ?

maribu, 2018-07-21:
@gebart: At least on the Mega2560 (JTag) this should be possible. I believe with the UNO (debugWIRE) it is also possible. Best would be a break point at kernel_init and then dump the memory, right?

@dylad: Sure. Can you suggest any known to work toolchain? Ideally in a docker container, so that I don't need to set up a VM for it?

@dylad: Sure

maribu, 2018-07-23:
@gebart: The mutex seems to not be initialized correctly:

```
(gdb) p _mutex
$1 = {queue = {next = 0x805f5f}}
(gdb) compare-sections
Section .data, range 0x6e0c -- 0x874a: matched.
Section .text, range 0x0 -- 0x6e0c: matched.
```

I guess RIOT depends on the c library to initialize the .bss section. So this is a toolchain bug.

But honestly, debugging with AVaRICE is so fragile, I'm not really sure if I can trust the output at all. Sometimes AVaRICE tells me that the program was interruped, while I interact with RIOT's shell (and the shell is responsive...).

dylad, 2018-07-23:
@maribu I am not a Docker user so I cannot help you
but google pointed to :
https://github.com/KaDock/avr-toolchain
https://hub.docker.com/r/mkdryden/avr-toolchain/
Hope it will help you.

maribu, 2018-07-23:
@dylad: Thank's for the links. It seems most docker AVR toolchains use the old avr-libc 1.8.0, which is compatible only with legacy GCC compilers. However, the legacy GCC versions don't provide c11 atomics, which are required to compile RIOT. So bad luck on that front.

maribu, 2018-07-23:
Can someone with an Arduino UNO or Arduino Mega2560 try to confirm these issues? @kYc0o: Do you have an UNO and get the default application running on the UNO? If so, does it also work for you with the docker container riot/riotbuild?

I just want to rule out that I have tough luck and my boards are broken.

kYc0o, 2018-07-23:
The UNO has not enough memory to run the default example, it can only run hello-world.

You might also check if the Mega has enough memory to do what do you want to do, shell apps are very memory consuming. Unfortunately, the toolchain won't prevent you to flash overflowed RAM/ROM binaries, so check carefully that you're on the boundaries.

kYc0o, 2018-07-23:
I don't have the hardware right now but the compilation goes smoothly:

```console
snake:RIOT facosta$ BUILD_IN_DOCKER=1 make BOARD=arduino-uno -C examples/hello-world/ 
make: Entering directory '/Users/facosta/git/RIOT-OS/RIOT/examples/hello-world'
Warning: no PORT set!
Launching build container using image "riot/riotbuild:latest".
docker run --rm -t -u "$(id -u)" \
    -v '/Users/facosta/git/RIOT-OS/RIOT:/data/riotbuild/riotbase' \
    -v '/Users/facosta/git/RIOT-OS/RIOT/cpu:/data/riotbuild/riotcpu' \
    -v '/Users/facosta/git/RIOT-OS/RIOT/boards:/data/riotbuild/riotboard' \
    -v '/Users/facosta/git/RIOT-OS/RIOT/makefiles:/data/riotbuild/riotmake' \
    -v '/Users/facosta/git/RIOT-OS/RIOT:/data/riotbuild/riotproject' \
    -v /etc/localtime:/etc/localtime:ro \
    -e 'RIOTBASE=/data/riotbuild/riotbase' \
    -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' \
    -e 'RIOTCPU=/data/riotbuild/riotcpu' \
    -e 'RIOTBOARD=/data/riotbuild/riotboard' \
    -e 'RIOTMAKE=/data/riotbuild/riotmake' \
    -e 'RIOTPROJECT=/data/riotbuild/riotproject' \
    -e 'BOARD=arduino-uno' \
    -w '/data/riotbuild/riotproject/examples/hello-world/' \
    'riot/riotbuild:latest' make  'BOARD=arduino-uno'
Building application "hello-world" for "arduino-uno" with MCU "atmega328p".

"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/auto_init
"make" -C /data/riotbuild/riotbase/sys/isrpipe
"make" -C /data/riotbuild/riotbase/sys/tsrb
"make" -C /data/riotbuild/riotbase/sys/uart_stdio
"make" -C /data/riotbuild/riotboard/arduino-uno
"make" -C /data/riotbuild/riotboard/common/arduino-atmega
"make" -C /data/riotbuild/riotboard/common/atmega
"make" -C /data/riotbuild/riotcpu/atmega328p
"make" -C /data/riotbuild/riotcpu/atmega_common
"make" -C /data/riotbuild/riotcpu/atmega_common/periph
   text	   data	    bss	    dec	    hex	filename
   5244	    314	    707	   6265	   1879	/data/riotbuild/riotproject/examples/hello-world/bin/arduino-uno/hello-world.elf
make: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/examples/hello-world'
```

maribu, 2018-07-23:
@kYc0o: Thank's for the info. I can confirm that the `default` is too big to fit the UNO's RAM and the `default` app with the `w5100` driver enabled overflows both falsh and RAM of the Mega2560. I wonder why avrdude does not complain about flashing an image bigger than the available flash.

Maybe the target `print-size` can call e.g. a little python script that takes the available flash and RAM size as optional parameters? If they are provided, in addition to the output of `avr-size` a summery line like `80.9% of RAM and 45.3% of Flash used` could be added and the script could fail on overflow.

Regarding the hello-world example on the UNO: I was able to build it with the docker toolchain, but I don't see any output on the console. With GCC 6.3.0 on Alpine it was the same, but with GCC 6.4.0 the hello world works. So there might still be an issue with the toolchain?

maribu, 2018-07-24:
@kYc0o: [This PR](https://github.com/RIOT-OS/RIOT/pull/9628) adds support for detection of RAM / flash overflow. Sadly some manual work is required: Adding the FLASH / RAM size for each CPU model, if the new feature should be provided

maribu, 2019-01-28:
During linking the RIOT build system does check for RAM/ROM overflow since https://github.com/RIOT-OS/RIOT/pull/9781 is merged, so this issue can be closed

closed 2019-01-28 by maribu