RIOT-OS/RIOT #9631: Makefiles: Remove abuses of FORCE and .PHONY

jcarrano, 2018-07-25:
# Description

## What

FORCE and .PHONY targets are being abused in RIOT's makefile. For example:

https://github.com/RIOT-OS/RIOT/blob/d6b2a9ed0152a7ddd38d380be9e291d1ed3701a2/Makefile.include#L383

https://github.com/RIOT-OS/RIOT/blob/d6b2a9ed0152a7ddd38d380be9e291d1ed3701a2/Makefile.include#L404

https://github.com/RIOT-OS/RIOT/blob/d6b2a9ed0152a7ddd38d380be9e291d1ed3701a2/Makefile.include#L443

## Why

This is incorrect. It triggers unnecessary rebuilds and may hide problems with the makefiles.

## How to solve it.

Targets should depend on their prerequisites only. To solve this:

1. Replace .PHONY with FORCE where it makes sense. This is a first step on making the code less wrong. (see #9623)
2. If the force is there to work around configuration variable changes, these should  be made into a target.
3. Remove the FORCE and PHONY.

cladmi, 2018-07-25:
I would still say, more important than not rebuilding for nothing, which "only" wastes time, we are currently not rebuilding when necessary https://github.com/RIOT-OS/RIOT/issues/9589 for example which makes the build invalid if you do not do `make clean all`.

But of course both are somehow linked together.

jcarrano, 2018-07-25:
@cladmi Of course they are linked. One can say that building too much is the lesser evil, but we should resist the temptation to work around our "too little built" issues by being overly conservative and forcing things to be build.

jcarrano, 2018-07-25:
Recursive use of make only complicates matters, see [Recursive Make Considered Harmful](http://aegis.sourceforge.net/auug97.pdf).
The rule that runs the recursive make *has* to be forced, because there's no way for the outer make to know things have not been modified.

closed 2020-10-18 by stale[bot]