RIOT-OS/RIOT #9645: Different build behavior between `murdock` and `riot/riotbuild:latest` image

cladmi, 2018-07-27:
#### Description

Compilation behavior is different between `murdock` the `docker` image with `BUILD_IN_DOCKER=1` even with `RIOT_CI_BUILD=1`.

It makes things hard to debug and understand the root cause of problems.

https://github.com/RIOT-OS/RIOT/pull/9362
https://github.com/RIOT-OS/RIOT/pull/9605
https://github.com/RIOT-OS/RIOT/pull/9507

#### Steps to reproduce the issue

To take one as an example:

Building `tests/pkg_fatfs_vfs` for `BOARD=wsn430-v1_4` works in murdock as it is not complaining but this does not

    make -C tests/pkg_fatfs_vfs/  BOARD=wsn430-v1_4 clean all RIOT_CI_BUILD=1 BUILD_IN_DOCKER=1 DOCKER="sudo docker"

#### Expected results

It has the same behavior in docker as in murdock, so either both fail or both succeed.

#### Actual results

Murdock is ok, but in docker it fails with:

```
/data/riotbuild/riotproject/tests/pkg_fatfs_vfs/bin/wsn430-v1_4/fatfs.a(ff.o):(.debug_loc+0x1656): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
collect2: ld returned 1 exit status
/data/riotbuild/riotbase/Makefile.include:411: recipe for target '/data/riotbuild/riotproject/tests/pkg_fatfs_vfs/bin/wsn430-v1_4/pkg_fatfs_vfs.elf' failed
make: *** [/data/riotbuild/riotproject/tests/pkg_fatfs_vfs/bin/wsn430-v1_4/pkg_fatfs_vfs.elf] Error 1
```

#### Versions

Building with docker in master.

cladmi, 2018-07-27:
@kaspar030 @smlng I think you are the person who would know the more about it

cladmi, 2018-07-27:
Other issue,  `RIOT_FILE_RELATIVE` is not really relative in `docker`:

```
-DRIOT_FILE_RELATIVE=\"/data/riotbuild/riotproject/tests/libfixmath_unittests/bin/pkg/stm32f0discovery/libfixmath/libfixmath/fix16_str.c\"
```

Outside I get:

```
-DRIOT_FILE_RELATIVE=\"tests/libfixmath_unittests/bin/pkg/stm32f0discovery/libfixmath/libfixmath/fix16_str.c\"
```

Which can easily explain build size difference when `RIOT_FILE_RELATIVE` is used.

cladmi, 2018-08-01:
Also if it's possible to have an archive of the currently  used image. I could diff with my local docker image and try to find the reason.

miri64, 2018-08-03:
At least for the LLVM thing @kaspar030 found a reason: https://github.com/RIOT-OS/RIOT/pull/9398#issuecomment-410248593

miri64, 2018-09-03:
It appears `ccache` was also the reason for the erroneous behavior in https://github.com/RIOT-OS/RIOT/pull/9362. Can't say anything specific about https://github.com/RIOT-OS/riotdocker/pull/42 though.

cladmi, 2018-09-03:
@ZetaR60 yes we can remove the blocking dependency.

There I have this PR here that handles some of the case for relative directories but it is also currently breaking working things: https://github.com/RIOT-OS/RIOT/pull/9646 I will try to find time to address them this week.

cladmi, 2018-09-06:
RIOT_CI_BUILD was not exported to docker which looks like the real reason for `wsn430` not building some examples in docker.

miri64, 2021-11-02:
The difference between the `riot/riotbuild:latest` and `murdock` worker image is now documented in https://github.com/RIOT-OS/riotdocker/tree/master/murdock-worker. Is this sufficient to close this issue?

kaspar030, 2021-11-03:
The board doesn't exist anymore, "RIOT_CI_BUILD was not exported to docker which looks like the real reason for wsn430 not building some examples in docker.", all tasks marked done.

Also, the murdock-worker container was always based on riotdocker. If there are differences how murdock builds wrt a local build, that should be independent of Docker. If Docker builds differently than a local build, that should be independent of Murdock.

So let's close this, feel free to re-open if an issue with Murdock vs. Docker remains.

closed 2021-11-03 by kaspar030