RIOT-OS/RIOT #9656: gnrc/netif: various problems after resetting interface a second time

PeterKietzmann, 2018-07-31:
#### Description
After resetting the network interface more than once, a node is not able to receive packets anymore.

#### Steps to reproduce the issue
Flash *examples/gnrc_networking* on two nodes A and B. Ping from node A to B via 
`ping6 <node B link-local IPv6 address>` .
Reset the network interface on node A via 
`ifconfig 7 set state reset`. 
Ping should still work. Reset the network interface of node A again. Ping fails.

Please also follow observations in #9616.

#### Expected results
Ping should always work no matter how often we reset the network interface.

#### Versions
Operating system: Ubuntu 18.04
Build environment: Docker

smlng, 2018-08-01:
copied from (my) duplicate issues:

#### Description

When running `examples/gnrc_networking` on (likely any board) eg. `samr21-xpro` or `pba-d-01-kw2x` the node does not respond to unicast (to its link local address) pings after 2 soft-resets, i.e. calling `ifconfig 7 set state reset`. However, the node does respond to multicast pings e.g. on `ff02::2`.

#### Steps to reproduce the issue

- flash and enter shell for `examples/gnrc_networking`
- do the same on another node and try to ping the first node: works
- run `ifconfig 7 set state reset` on the first node
- run `ifconfig 7 set state reset`, again
- try to ping the first node from the second: does not work

#### Expected results

ping (and any other network communication) should work after any number of resets, as described above. 

#### Actual results

network communication fails after 2 soft resets, however multicast pings still work (eg. on `ff02::1`).

#### Versions

- boards: samr21-xpro, pba-d-01-kw2x
- app: examples/gnrc_networking
- `main(): This is RIOT! (Version: 2018.10-devel-106-g52848-mobi25.inet.haw-hamburg.de)

toolchains:

```
Installed compiler toolchains 
-----------------------------
             native gcc: Apple LLVM version 9.1.0 (clang-902.0.39.2)
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
                avr-gcc: avr-gcc (GCC) 8.1.0
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: Apple LLVM version 9.1.0 (clang-902.0.39.2)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.11.4
               cppcheck: Cppcheck 1.84
                doxygen: 1.8.14
                 flake8: error
                    git: git version 2.18.0
             coccinelle: spatch version 1.0.6 compiled with OCaml version 4.05.0
```

smlng, 2018-08-01:
just to be clear: after more than 1 reset the node does not receive unicast packets but does receive and handle multicast packets.

cladmi, 2018-08-01:
I am not mistaken @bergzand talked about this issue

miri64, 2018-08-01:
I'll have a look.

miri64, 2018-08-01:
Mhhhh... I'm so sure this is a GNRC netif problem. Yes, the problem occurs with multiple radios, but when I put the following patch in

```diff
diff --git a/sys/net/gnrc/netif/gnrc_netif.c b/sys/net/gnrc/netif/gnrc_netif.c
index 8110af5cc..9ac725190 100644
--- a/sys/net/gnrc/netif/gnrc_netif.c
+++ b/sys/net/gnrc/netif/gnrc_netif.c
@@ -1360,6 +1360,7 @@ static void _event_cb(netdev_t *dev, netdev_event_t event)
     gnrc_netif_t *netif = (gnrc_netif_t *) dev->context;
 
     if (event == NETDEV_EVENT_ISR) {
+        puts("ISR event!");
         msg_t msg = { .type = NETDEV_MSG_TYPE_EVENT,
                       .content = { .ptr = netif } };
 
@@ -1368,7 +1369,7 @@ static void _event_cb(netdev_t *dev, netdev_event_t event)
         }
     }
     else {
-        DEBUG("gnrc_netif: event triggered -> %i\n", event);
+        printf("gnrc_netif: event triggered -> %i\n", event);
         switch (event) {
             case NETDEV_EVENT_RX_COMPLETE: {
                     gnrc_pktsnip_t *pkt = netif->ops->recv(netif);
```

I get the following output:

```
m3-1533132615.948118;m3-11;ISR event!
1533132615.949186;m3-11;gnrc_netif: event triggered -> 2
1533132615.950306;m3-10;ISR event!
1533132615.950933;m3-10;gnrc_netif: event triggered -> 4
1533132616.838752;m3-11;ISR event!
1533132616.839668;m3-11;gnrc_netif: event triggered -> 4
1533132616.840386;m3-10;ISR event!
1533132616.842998;m3-10;gnrc_netif: event triggered -> 2

ifconfig
1533132623.540039;m3-11;ifconfig
1533132623.541815;m3-11;Iface  7  HWaddr: 2d:56  Channel: 26  Page: 0  NID: 0x23
1533132623.542537;m3-10;ifconfig
1533132623.542879;m3-10;Iface  7  HWaddr: 3d:52  Channel: 26  Page: 0  NID: 0x23
1533132623.543166;m3-11;          Long HWaddr: 15:11:6b:10:65:fc:ad:56 
1533132623.543988;m3-10;          Long HWaddr: 15:11:6b:10:65:fc:bd:52 
1533132623.544821;m3-11;           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
1533132623.545588;m3-10;           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
1533132623.546797;m3-11;          AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
1533132623.547142;m3-11;          IPHC  
1533132623.547885;m3-11;          Source address length: 8
1533132623.548144;m3-11;          Link type: wireless
1533132623.548818;m3-10;          AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
1533132623.549069;m3-10;          IPHC  
1533132623.549383;m3-10;          Source address length: 8
1533132623.549909;m3-11;          inet6 addr: fe80::1711:6b10:65fc:ad56  scope: local  VAL
1533132623.550185;m3-10;          Link type: wireless
1533132623.550796;m3-11;          inet6 group: ff02::2
1533132623.551132;m3-11;          inet6 group: ff02::1
1533132623.551544;m3-10;          inet6 addr: fe80::1711:6b10:65fc:bd52  scope: local  VAL
1533132623.552000;m3-11;          inet6 group: ff02::1:fffc:ad56
1533132623.552542;m3-10;          inet6 group: ff02::2
1533132623.552861;m3-10;          inet6 group: ff02::1
1533132623.553152;m3-10;          inet6 group: ff02::1:fffc:bd52
1533132623.553702;m3-11;          inet6 group: ff02::1a
1533132623.553936;m3-11;          
1533132623.554331;m3-11;          Statistics for Layer 2
1533132623.554711;m3-10;          inet6 group: ff02::1a
1533132623.554944;m3-10;          
1533132623.555462;m3-11;            RX packets 7  bytes 287
1533132623.555833;m3-10;          Statistics for Layer 2
[... 86 lines cut ...]
1533132666.388973;m3-11;gnrc_netif: event triggered -> 2
1533132666.389178;m3-10;ISR event!
1533132666.392272;m3-11;ISR event!
1533132666.393209;m3-11;gnrc_netif: event triggered -> 4
1533132666.393517;m3-10;gnrc_netif: event triggered -> 4
1533132666.393738;m3-10;ISR event!
1533132666.393944;m3-10;gnrc_netif: event triggered -> 2
1533132666.396186;m3-10;12 bytes from fe80::1711:6b10:65fc:ad56: id=85 seq=2 hop limit=64 time = 9.286 ms
1533132667.398569;m3-11;ISR event!
1533132667.398955;m3-10;ISR event!
1533132667.399662;m3-11;gnrc_netif: event triggered -> 2
1533132667.400837;m3-10;gnrc_netif: event triggered -> 4
1533132667.403540;m3-10;ISR event!
1533132667.404465;m3-11;ISR event!
1533132667.405581;m3-10;gnrc_netif: event triggered -> 2
1533132667.406306;m3-11;gnrc_netif: event triggered -> 4
1533132667.406873;m3-10;12 bytes from fe80::1711:6b10:65fc:ad56: id=85 seq=3 hop limit=64 time = 9.924 ms
1533132667.407630;m3-10;--- ff02::1 ping statistics ---
1533132667.408523;m3-10;3 packets transmitted, 3 received, 0% packet loss, time 2.0633781 s
1533132667.409500;m3-10;rtt min/avg/max = 8.964/9.391/9.924 ms
1533132680.346434;Stopping

```

So after the double reset the device doesn't trigger the RX_COMPLETE event (meaning the `irq_handler` isn't even fired).

miri64, 2018-08-01:
For multicast however they are :-/.

miri64, 2018-08-01:
So maybe something happens to the address configuration to most of those devices on reset.

miri64, 2018-08-01:
Yeah... have a look at that:

```
m3-11;ifconfig
1533133070.667637;m3-11;> ifconfig
1533133070.669312;m3-11;Iface  7  HWaddr: 2d:56  Channel: 26  Page: 0  NID: 0x23
1533133070.669978;m3-11;          Long HWaddr: 15:11:6b:10:65:fc:ad:56 
1533133070.671896;m3-11;           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
1533133070.673002;m3-11;          AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
1533133070.673993;m3-11;          IPHC  
1533133070.674887;m3-11;          Source address length: 8
1533133070.675380;m3-11;          Link type: wireless
1533133070.676902;m3-11;          inet6 addr: fe80::1711:6b10:65fc:ad56  scope: local  VAL
1533133070.677384;m3-11;          inet6 group: ff02::2
1533133070.677999;m3-11;          inet6 group: ff02::1
1533133070.678888;m3-11;          inet6 group: ff02::1:fffc:ad56
1533133070.679899;m3-11;          inet6 group: ff02::1a
1533133070.680344;m3-11;          
1533133070.680941;m3-11;          Statistics for Layer 2
1533133070.681885;m3-11;            RX packets 3  bytes 129
1533133070.682923;m3-11;            TX packets 6 (Multicast: 6)  bytes 387
1533133070.683401;m3-11;            TX succeeded 6 errors 0
1533133070.683984;m3-11;          Statistics for IPv6
1533133070.684919;m3-11;            RX packets 3  bytes 192
1533133070.685906;m3-11;            TX packets 6 (Multicast: 6)  bytes 370
1533133070.686908;m3-11;            TX succeeded 6 errors 0
1533133070.687369;m3-11;
m3-11;ifconfig 7 set state reset
1533133073.154579;m3-11;> ifconfig 7 set state reset
1533133073.157447;m3-11;success: set state of interface 7 to RESET
m3-11;ifconfig
1533133075.002510;m3-11;> ifconfig
1533133075.004327;m3-11;Iface  7  HWaddr: 2d:56  Channel: 26  Page: 0  NID: 0x23
1533133075.005889;m3-11;          Long HWaddr: 15:11:6b:10:65:fc:ad:56 
1533133075.007546;m3-11;           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
1533133075.008866;m3-11;          AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
1533133075.009063;m3-11;          IPHC  
1533133075.010289;m3-11;          Source address length: 2
1533133075.011001;m3-11;          Link type: wireless
1533133075.011863;m3-11;          inet6 addr: fe80::1711:6b10:65fc:ad56  scope: local  VAL
1533133075.012787;m3-11;          inet6 group: ff02::2
1533133075.014167;m3-11;          inet6 group: ff02::1
1533133075.014672;m3-11;          inet6 group: ff02::1:fffc:ad56
1533133075.015186;m3-11;          inet6 group: ff02::1a
1533133075.015872;m3-11;          
1533133075.016345;m3-11;          Statistics for Layer 2
1533133075.017023;m3-11;            RX packets 3  bytes 129
1533133075.018300;m3-11;            TX packets 6 (Multicast: 6)  bytes 387
1533133075.018787;m3-11;            TX succeeded 6 errors 0
1533133075.019697;m3-11;          Statistics for IPv6
1533133075.021052;m3-11;            RX packets 3  bytes 192
1533133075.022188;m3-11;            TX packets 6 (Multicast: 6)  bytes 370
1533133075.022671;m3-11;            TX succeeded 6 errors 0
1533133075.023116;m3-11;
m3-11;ifconfig 7 set state reset
1533133077.827727;m3-11;> ifconfig 7 set state reset
1533133077.829318;m3-11;success: set state of interface 7 to RESET
m3-11;ifconfig
1533133079.225740;m3-11;> ifconfig
1533133079.228587;m3-11;Iface  7  HWaddr: 2d:52  Channel: 26  Page: 0  NID: 0x23
1533133079.229994;m3-11;          Long HWaddr: 15:11:6b:10:65:fc:ad:52 
1533133079.231526;m3-11;           TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
1533133079.233082;m3-11;          AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
1533133079.233279;m3-11;          IPHC  
1533133079.234379;m3-11;          Source address length: 2
1533133079.235817;m3-11;          Link type: wireless
1533133079.237068;m3-11;          inet6 addr: fe80::1711:6b10:65fc:ad56  scope: local  VAL
1533133079.237623;m3-11;          inet6 group: ff02::2
1533133079.238102;m3-11;          inet6 group: ff02::1
1533133079.238766;m3-11;          inet6 group: ff02::1:fffc:ad56
1533133079.239759;m3-11;          inet6 group: ff02::1a
1533133079.240270;m3-11;          
1533133079.240767;m3-11;          Statistics for Layer 2
1533133079.241385;m3-11;            RX packets 3  bytes 129
1533133079.242356;m3-11;            TX packets 6 (Multicast: 6)  bytes 387
1533133079.243861;m3-11;            TX succeeded 6 errors 0
1533133079.244332;m3-11;          Statistics for IPv6
1533133079.244808;m3-11;            RX packets 3  bytes 192
1533133079.245425;m3-11;            TX packets 6 (Multicast: 6)  bytes 370
1533133079.246402;m3-11;            TX succeeded 6 errors 0
1533133079.246869;m3-11;
```

Before the second reset the short address is `2d:56` and the EUI-64 is `15:11:6b:10:65:fc:ad:56`, after the second reset the short address is `2d:52` and the EUI-64 is `15:11:6b:10:65:fc:ad:52`, so the last byte changed for both addresses for some reason. Both [`at86rf2xx`][1] and [`kw2xrf`][2] use the `luid_get()` function to generate the address, which has the side-effect of incrementing a `lastused` counter that is [incorporated into the result][3].

[1]: https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L67
[2]: https://github.com/RIOT-OS/RIOT/blob/master/drivers/kw2xrf/kw2xrf.c#L48
[3]: https://github.com/RIOT-OS/RIOT/blob/master/sys/luid/luid.c#L31-L36

miri64, 2018-08-01:
@PeterKietzmann The title says "various problems", but the issue description only goes into the reception problem (so does @smlng). What are the other problems? If this is the only problem, mind that we rename it `drivers/*: get_luid() not idempotent` or something, to reflect the cause I demonstrated?

cladmi, 2018-08-01:
The PR was chosen to replace https://github.com/RIOT-OS/RIOT/issues/8118 so just check if renaming it does not hides known problems.

PeterKietzmann, 2018-08-01:
@miri64 I just chose this title after @bergzand stated

>I suspect that the issue fixed here is only one of the issues that causes problems after a reset

and @smlng asked me to open an issue that is more generic than a single driver implementation. Feel free to change to whatever fits better.

cladmi, 2018-08-01:
@miri64 from the documentation it's normal that `luid_get` returns different values, so it should be more an issue that the driver don't save the value.

It could be replaced by a new call `luid_set_t luid_get_seed(..., luid_set_t *seed)` that returns the `lastused` value and accepts taking it as input to generate the corresponding key. It would allow saving only the `lastused` part in the driver.

miri64, 2018-08-01:
@cladmi:

> The PR was chosen to replace #8118 so just check if renaming it does not hides known problems.

What PR are you talking about?

> It could be replaced by a new call luid_set_t luid_get_seed(..., luid_set_t *seed) that returns the lastused value and accepts taking it as input to generate the corresponding key. It would allow saving only the lastused part in the driver.

That might be a good solution. AFAIK the `lastused` global context was introduced to allow for multiple interfaces to have different link-layer addresses (interesting though, that the first reset doesn't have any effect...), so the intention is valid, but the execution wasn't thought through.

@PeterKietzmann 

> @miri64 I just chose this title after @bergzand stated
>
> >I suspect that the issue fixed here is only one of the issues that causes problems after a reset
>
> and @smlng asked me to open an issue that is more generic than a single driver implementation. Feel free to change to whatever fits better.

Ok, let's wait for a solving PR to or some experimental results that proof the `luid_get()` problem, before we pin-point the issue to just that.

cladmi, 2018-08-01:
@miri64 I meant "this issue", too much used to talk about PRs.

> That might be a good solution. AFAIK the lastused global context was introduced to allow for multiple interfaces to have different link-layer addresses (interesting though, that the first reset doesn't have any effect...), so the intention is valid, but the execution wasn't thought through.

Ok. The drivers could also save the address somewhere as the default link layer address. A new call is only to, maybe, save some RAM by saving just the seed instead of the link layer address. At least it could be compared.

First thing could be a simple test, where the address is printed each time after a reset and should be the same each time. This could simplify requesting maintainers to test on their board.

miri64, 2018-08-01:
> Ok. The drivers could also save the address somewhere as the default link layer address. A new call is only to, maybe, save some RAM by saving just the seed instead of the link layer address. At least it could be compared.

Well it does that already in the `netdev_ieee802154_t` struct, but @bergzand is already working on getting those layers nicely separated I would like to avoid the device to access those members directly.

bergzand, 2018-08-01:
My current idea for netdev is to remove the link layer address members from the netdev structs. This because IMHO there is no need for netdev to remember the link layer address if the higher layer (gnrc, lwip etc) already have the link layer address.

The getters and setters are only necessary to get the ll address if the interface has one by default (e.g. netdev_tap) and to set the address for ll filtering on the hardware device.

On a `NETOPT_STATE_RESET`, the upper layer only has to resupply the configured ll address to the interface instead of having the network interface generate (a new) one.

miri64, 2018-08-01:
> On a `NETOPT_STATE_RESET`, the upper layer only has to resupply the configured ll address to the interface instead of having the network interface generate (a new) one.

This I find unacceptable. The network device should still be able to be idempotent on its own.

bergzand, 2018-08-01:
> This I find unacceptable. The network device should still be able to be idempotent on its own.

I agree with you on this.

My idea was to have the upper layer generate the link address on init and call the setter on the netdev device to configure it for the LL filter on the hardware. In a certain way the LL address would be handled similar to the way the `NETDEV_FLAGS_` are handled now, they have to be set when the netdev instance is initialized or reset.

cladmi, 2018-08-11:
As mentioned in https://github.com/RIOT-OS/RIOT/pull/9616 I would find it great to have a test for these issues. Single node test if possible or in the https://github.com/RIOT-OS/Release-Specs if they require two nodes for testing.

miri64, 2018-10-30:
How about the following kind of fix:

```diff
diff --git a/drivers/at86rf2xx/at86rf2xx.c b/drivers/at86rf2xx/at86rf2xx.c
index 8006168..d7728d4 100644
--- a/drivers/at86rf2xx/at86rf2xx.c
+++ b/drivers/at86rf2xx/at86rf2xx.c
@@ -65,7 +65,8 @@ void at86rf2xx_reset(at86rf2xx_t *dev)
     }
 
     /* get an 8-byte unique ID to use as hardware address */
-    luid_get(addr_long.uint8, IEEE802154_LONG_ADDRESS_LEN);
+    luid_custom(addr_long.uint8, IEEE802154_LONG_ADDRESS_LEN,
+                (intptr_t)dev);
     /* make sure we mark the address as non-multicast and not globally unique */
     addr_long.uint8[0] &= ~(0x01);
     addr_long.uint8[0] |=  (0x02);
```

This way we don't need some extra state, but the address is always the same after a reset.

gebart, 2018-11-12:
@miri64 the drawback of using the pointer is that we will get different link layer addresses for each build. E.g. Imagine devices updating using OTA updates and then coming back online with a completely new address.

miri64, 2018-11-12:
> @miri64 the drawback of using the pointer is that we will get different link layer addresses for each build. E.g. Imagine devices updating using OTA updates and then coming back online with a completely new address.

That's true :-/

closed 2020-08-24 by benpicco