RIOT-OS/RIOT #9692: BOSSA/arduino-due based board possible issue

dylad, 2018-08-03:
#### Description

I am suspecting a bug in the BOSSA use for arduino-due based board. I was playing with arduino-zero like board and I tried to flash it using BOSSA, I borrowed the Makefile.include from common/arduino-due folder.
I was able to flash but app doesn't run after reset.
I found out I need to add the dedicated linker script (behaviour taken from sodaq-explorer Makefile.include).
After that, my board was working as expected but now I'm wondering if arduino-due based are working using BOSSA.

Would be awesome if someone owning one of this board could confirm or not the issue !

PeterKietzmann, 2018-08-17:
@leandrolanzieri maybe look at this one?

kYc0o, 2018-08-17:
Can you paste the output of `arm-none-eabi-readelf -S` of the generated ELFs with and without the adapted linker script?

leandrolanzieri, 2018-08-17:
I just tested a few examples on the arduino-due board and it's working. :thumbsup:

dylad, 2018-09-03:
Well, if it works, we're all good :) 
Closing this one.

closed 2018-09-03 by dylad