RIOT-OS/RIOT #9695: SPI Mode 2 & 3 problems in sam0_common cpu

leandrolanzieri, 2018-08-03:
#### Description
SPI modes 2 & 3 have incorrect clock signals on samr21-xpro board or anything with the sam0_common cpu.

#### Steps to reproduce the issue
On the samr21-xpro run tests/periph_spi looking at PB22 and PB23
```
init 1 2 0 0 0
2018-08-03 13:40:14,532 - INFO # SPI_DEV(1) initialized: mode: 2, clk: 0, cs_port: 0, cs_pin: 0
> send UUUU
2018-08-03 13:40:18,405 - INFO #  send UUUU
2018-08-03 13:40:18,406 - INFO # Sent bytes
2018-08-03 13:40:18,408 - INFO #    0    1    2    3 
2018-08-03 13:40:18,410 - INFO #   0x55 0x55 0x55 0x55
2018-08-03 13:40:18,412 - INFO #     U    U    U    U 
2018-08-03 13:40:18,413 - INFO # 
2018-08-03 13:40:18,414 - INFO # Received bytes
2018-08-03 13:40:18,416 - INFO #    0    1    2    3 
2018-08-03 13:40:18,418 - INFO #   0x00 0x00 0x00 0x00
2018-08-03 13:40:18,420 - INFO #    ??   ??   ??   ?? 
```

If scoped you should see the above waveforms on the MOSI/SCK

#### Expected results
**> Correct waveforms**
This are the spected waveforms for each SPI mode (0 to 3), by changing the **polarity** and the **phase**.

| Mode | CPOL | CPHA |
|------|------|------|
|  0   |  0   |  0   |
|  1   |  0   |  1   |
|  2   |  1   |  0   |
|  3   |  1   |  1   |

[image]

#### Actual results
**> Mode 2**
When SPI was configured in Mode 2 it was observed that SCK should be HIGH when the CS pin goes LOW (enabled) but it is LOW.

In this mode the falling edge should occur at the middle of the data pulse, but its happening at the beginning of it.
[image]

**> Mode 3**
In this case data sample should occur with the first rising edge of SCK and that should happen right in the middle of the data pulse. Instead, the rising edge of SCK occurs at the beginning of the data and at the middle there is a falling edge on SCK.
[image]

#### Versions
```
Operating system: Linux
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "2.5.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: missing
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.1
             coccinelle: missing
```

leandrolanzieri, 2018-09-04:
@dylad nice one! 
This seems to fix both mode 2 and 3 issues.

## Measurements

### Mode 2
**Before**:
* SCK was LOW when CS went LOW (enabled)
* SCK falling edge at the beginning of the data

[image]

**Now**:
* SCK is HIGH when CS goes LOW
* SCK has it's falling edge in the middle of the data
[image]

### Mode 3
**Before**:
* SCK rising edge at the beginning of the data
* SCK was LOW when CS went LOW (enabled)
[image]

**Now**:
* SCK rising edge in the middle of the data
* SCK is HIGH when CS goes LOW
[image]

Would you open a PR for this fix?

dylad, 2018-09-04:
I'll open a PR today then :)

PeterKietzmann, 2018-09-10:
#9885 has been merged

closed 2018-09-10 by PeterKietzmann