RIOT-OS/RIOT #9708: Error when compiling native

spirit900, 2018-08-04:
#### Description

Cannot build hello-world application for native  board. 

#### Steps to reproduce the issue

Hello! I use Ubuntu 14.04, gcc 4.9. I have compiled Hello-world for bluepill without any problem. But then tried the same for native, it's return this:

"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/boards/native
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/boards/native/drivers
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/core
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/cpu/native
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/cpu/native/periph
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/cpu/native/vfs
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/drivers
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/drivers/periph_common
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/sys
"make" -C /home/pavel/Desktop/STM32_RIOT/RIOT-master/sys/auto_init
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.9/libgcc.a when searching for -lgcc
/usr/bin/ld: cannot find -lgcc
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.9/libgcc_s.so when searching for -lgcc_s
/usr/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status

Then i tried to install multilib-gcc 
sudo apt-get install multilib-gcc
 it answered can't find multilib-gcc
In  /etc/apt/sources.list.d, i have file: ia32-l (without extension)
It contains this line: 
deb http://old-releases.ubuntu.com/ubuntu/ raring main restricted universe multiverse
How yo install multilib-gcc? And will it help? 
Thank you.

#### Expected results

#### Actual results

#### Versions

miri64, 2018-08-04:
Hi, I think (but can not confirm) in older ubuntu versions the package was called `ia32-libs`. Hope that helps.

spirit900, 2018-08-18:
This helped: 
sudo apt-get install gcc-4.9-multilib.

smlng, 2018-08-27:
can this be closed, then?

miri64, 2018-09-30:
I think so.

closed 2018-09-30 by miri64