RIOT-OS/RIOT #9709: examples: failed assertion in dtls-echo example

miri64, 2018-08-04:
#### Description
The `dtls-echo` example fails an assertion when trying to send between 6Lo nodes with global addresses.

#### Steps to reproduce the issue
Compile `dtls-echo` with `USEMODULE += gnrc_rpl auto_init_gnrc_rpl` flash it to two nodes

Flash a third RIOT not with `gnrc_networking` and configure it as a root:

```
ifconfig 6 add 2001:db8::<blafoo>
rpl root 0 2001:db8::<blafoo>
```

Start a DTLS server on one of the `dtls-echo` nodes:

```
dtlss start
```

Send a DTLS message from the other `dtls-echo` node:

```
dtlsc 2001:db8::<foobar> "Hello world"
```

Originally reported in #9706.

#### Expected results
DTLS server should receive message, DTLS client should receive an echo.

#### Actual results
One of the two nodes aborts with a failed assertion.

#### Versions
Tested on IoT-Lab (both `lille` and `saclay`

rfuentess, 2018-08-04:
I got a similar issue with the gcoap example, but I was unable to determine if there was a faulty commit affecting both cases. I tried with this specific example, due it was the base for some parts of dtls-echo.

miri64, 2018-08-07:
Tried on IoT-LAB also and can not reproduce anymore :-/

rfuentess, 2018-08-07:
>Tried on IoT-LAB also and can not reproduce anymore :-/

I was wondering the chances that IoT-Lab was the source of the issues instead of the code. I was getting similar behavior even with previous commits that were well tested. But sadly, without physical nodes, I was unable to confirm my suspects.

miri64, 2018-08-07:
Ok, then let's keep this open as a known issue for this release (@cladmi ?) in case it pops up again.

cladmi, 2018-08-07:
check

miri64, 2018-08-08:
@rfuentess I retested on testbed in `lille` and `saclay` now (with static routes though) and I still wasn't able to reproduce :-/

miri64, 2020-07-02:
@pokgak can you please look into this?

pokgak, 2020-07-03:
I managed to reproduce this by manually setting global addresses for the node using 2 unmodified iotlab-m3 on saclay.
```
m3-11;ifconfig 5 add 2001:db8::1
1593733446.359799;m3-11;> ifconfig 5 add 2001:db8::1
1593733446.360544;m3-11;success: added 2001:db8::1/64 to interface 5
m3-12;ifconfig 5 add 2001:db8::2
1593733454.709320;m3-12;> ifconfig 5 add 2001:db8::2
1593733454.710112;m3-12;success: added 2001:db8::2/64 to interface 5
m3-11;dtlss start
1593733463.630917;m3-11;> dtlss start
m3-12;dtlsc 2001:db8::1 "Hello World"
1593733490.901670;m3-12;> dtlsc 2001:db8::1 "Hello World"
1593733500.904304;m3-12;0x80035a7
1593733500.905185;m3-12;*** RIOT kernel panic:
1593733500.905590;m3-12;FAILED ASSERTION.
1593733500.905950;m3-12;
1593733500.906297;m3-12;*** halted.
1593733500.906638;m3-12;

```

pokgak, 2020-07-03:
I cannot reproduce this on `native` using the same steps as above

pokgak, 2020-07-03:
By adding `CFLAGS += -DDEBUG_ASSERT_VERBOSE` to the Makefile I found the failing assert:
```
1593736614.081889;m3-11;sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c:335 => 0x8000e21
```
which points to this section of code: https://github.com/RIOT-OS/RIOT/blob/2ed76e2d45e8eb8f6ad895622aca14a3e6ad0eea/sys/net/gnrc/network_layer/ipv6/nib/_nib-internal.c#L333-L338

After that, I added `CFLAGS += -DCONFIG_GNRC_IPV6_NIB_ARSM=1` to the Makefile, and with that, I got no failed assertions any more. I don't think this is the solution, but at least we know what is causing this issue now :)

pokgak, 2020-07-03:
I managed to reproduce the FAILED_ASSERTION using only one `samr21-xpro` board flashed with the dtls-echo firmware. It should be easier to test this locally now:

```
> ifconfig 6 add 2001:db8::2  # add global ip for client
> dtlsc 2001:db8::1 "HELLO WORLD"    # try to send to non-existent server
```

miri64, 2020-07-03:
That's weird though. Hope I find some time to investigate.

miri64, 2020-09-01:
Found it! See #14920.

closed 2020-09-01 by miri64