RIOT-OS/RIOT #9737: Coding convention: Dependency injection

Citrullin, 2018-08-08:
Hi,
I am working on creating a PR for the shiftIn topic: https://github.com/RIOT-OS/RIOT/issues/9712
Since I need gpio and it is a module I asked myself how to get it into bitarithm. One nice solution could be dependency injection with a pointer function. So, I just want to know if this the way to go or if the RIOT contributors prefer something different.

gebart, 2018-08-08:
I suggest you create a new module, preferably under drivers, since the shiftIn function is basically one half of a bit banging SPI implementation. Don't put it in core/bitarithm, core is used only for important core kernel functionality, such as task switching and IPC .

kaspar030, 2018-08-08:
> preferably under drivers

if it is plain logic code, wouldn't sys/ make more sense?

gebart, 2018-08-08:
>if it is plain logic code, wouldn't sys/ make more sense?

But it is not, at least not according to https://www.arduino.cc/reference/en/language/functions/advanced-io/shiftin/

closed 2018-08-17 by Citrullin