RIOT-OS/RIOT #9739: lpc2387: non-ARM statement in naked function

miri64, 2018-08-08:
During my efforts to better our LLVM/clang support (see #9398, #9730) I noticed that for our `lpc2387` boards I get the following error when compiling with `TOOLCHAIN=llvm`:

```
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/msba2
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/common/msba2
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/lpc2387
/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/lpc2387/cpu.c:90:5: error: non-ASM statement in naked function is not supported
    irq_disable();
    ^
/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/lpc2387/cpu.c:88:16: note: attribute is here
__attribute__((naked,noreturn)) void arm_reset(void)
               ^
1 error generated.
/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:83: recipe for target '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/msba2/cpu/cpu.o' failed
make[2]: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/msba2/cpu/cpu.o] Error 1
/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:20: recipe for target 'ALL--/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/lpc2387' failed
make[1]: *** [ALL--/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/lpc2387] Error 2
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/../../Makefile.include:417: recipe for target '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/msba2/application_hello-world.a' failed
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/msba2/application_hello-world.a] Error 2
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world'
```

I've done some research and found conflicting meanings for the `naked` attribute. The [GCC manual](https://gcc.gnu.org/onlinedocs/gcc-3.1/gcc/Function-Attributes.html) only states

> Use this attribute on the ARM or AVR ports to indicate that the specified function do not need prologue/epilogue sequences generated by the compiler. It is up to the programmer to provide these sequences. 

[This source](http://www.keil.com/support/man/docs/armclang_ref/armclang_ref_jhg1476893564298.htm) (a documentation on ARM clang support I assume) however states a `naked` function must only contain ARM instructions.

The question is: is it safe to just remove the `naked` attribute, since `arm_reset` doesn't contain any assembly at all? For Cortex-M platforms there was a similar discussion 4 years ago: https://github.com/RIOT-OS/RIOT/issues/1654

miri64, 2018-08-08:
@maribu maybe you want to have a look as well.

gebart, 2018-08-08:
Not sure. Is arm_reset the system entry point? It might be bad to start the system by pushing some nonsense onto the stack.

jcarrano, 2018-08-08:
@gebart The entry point in ARM is `Reset_Handler`.

@miri64 The `naked` attribute should be removed. It is wrong write anything but `__asm` statements inside a naked function (which `arm_reset` does). The attribute not necessary because it is not called at startup.

It seems  `arm_reset` (beatifully undocumented, btw)  is only called from within `pm.c`. While I have not read the datasheet, I assume it's setting the WDT to reset the cpu and the going into an infinite loop, waiting to be killed (weird way to reset).

With regards to the other attribute, `noreturn`, it's kind of pointless because the prototype is not defined as such: https://github.com/RIOT-OS/RIOT/blob/a5bdf0a831868a5e965cef664ad3ca540104e9cd/cpu/arm7_common/include/arm_cpu.h#L26

The purpose of `noreturn` is to inform the _caller_ that the call will never return, so it needs to be in the header to be effective.

maribu, 2018-08-14:
I can confirm that @jcarrano is correct. See https://github.com/RIOT-OS/RIOT/pull/9773. I tested and the `reboot` shell command, which in turn calls `pm_reboot`, which in turn calls `arm_reset`, still works.

miri64, 2018-10-03:
Was fixed when #9773 was merged.

closed 2018-10-03 by miri64