RIOT-OS/RIOT #9742: `buildtest` uses wrong build directory

cladmi, 2018-08-08:
#### Description

When building using `buildtest` the build is not using the BOARD `bin` directory but the default BOARD defined in the application Makefile one `BOARD ?= native`.

#### Steps to reproduce the issue

I am using https://github.com/RIOT-OS/RIOT/pull/9741 to make it more visible during compilation but can be done without.

When building for `iotlab-m3` the output is saved in `bin/native`.

```
$ git clean -xdff examples/hello-world

$ BOARDS=iotlab-m3 make -C examples/hello-world/ buildtest 'BUILDTEST_MAKE_REDIRECT='
...
Building application "hello-world" for "iotlab-m3" with MCU "stm32f1".

   text    data     bss     dec     hex filename
   8480     140    2740   11360    2c60 /home/harter/work/git/RIOT/examples/hello-world/bin/native/hello-world.elf

$ find examples/hello-world/bin/
examples/hello-world/bin/
examples/hello-world/bin/native
examples/hello-world/bin/native/hello-world.hex
examples/hello-world/bin/native/hello-world.elf
examples/hello-world/bin/native/hello-world.map
```

When running some compilations with `buildtest` I also sometime had strange issues with packages with `native` but was correct when build alone.
I think it is related to an issue with the package `clean` which was shown with this path collision.

#### Expected results

Building in `bin/iotlab-m3`

#### Actual results

Build in `bin/native`.

#### Versions

Ubuntu 16.04

cladmi, 2018-10-24:
More information on this one, the issue is only present when providing variable configuration from the command line.

cladmi, 2018-11-19:
What I should address is that directories are overridden only if necessary in this line:

https://github.com/RIOT-OS/RIOT/blob/c7894d2bfc0c437da2ca6472ac5c2c439b358330/Makefile.include#L69
 
To not have different behavior when setting a variable on the command line.

cladmi, 2018-11-19:
@MrKevinWeiss this is the tracking issue of https://github.com/RIOT-OS/RIOT/pull/9831

crasbe, 2025-04-03:
The culprit of this issue is that the build does not start "clean":
https://github.com/RIOT-OS/RIOT/blob/a84955c7cafc24d1d7d2c4636e05e19f3df1c791/makefiles/buildtests.inc.mk#L5-L23

Adding an `env; \` between line 10 and 11 reveals the issue:
```
cbuec@W11nMate:~/RIOTstuff/riot-vanilla/RIOT$ BOARDS=feather-nrf52840 make -C examples/basic/hello-world/ buildtest
make: Entering directory '/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world'
using BOARD="native64" as "native" on a 64-bit system
Building for feather-nrf52840 ... LESSOPEN=| /usr/bin/lesspipe %s
DOCKER_MAKECMDGOALS_POSSIBLE=all buildtest-indocker scan-build scan-build-analyze tests-%
INSIDE_DOCKER=0
USER=cbuec
WT_PROFILE_ID={4ff56d04-d9cf-57ea-bae2-ad396374e7e3}
SHLVL=1
DOCKER_ENV_VARS=APPDIR AR AS ASFLAGS BINDIR BINDIRBASE BOARD BOARDS BUILDRELPATH BUILDTEST_MCU_GROUP BUILDTEST_VERBOSE CC CC_NOCOLOR CFLAGS CONTINUE_ON_EXPECTED_ERRORS CPPMIX CXX CXXEXFLAGS CXXUWFLAGS  ELFFILE FLASHFILE HEXFILE IOTLAB_NODE LINK LINKFLAGPREFIX LINKFLAGS LTO OBJCOPY OFLAGS PARTICLE_MONOFIRMWARE PICOLIBC PREFIX PROGRAMMER QUIET RIOT_CI_BUILD RIOT_VERSION RIOT_VERSION_CODE RUSTFLAGS SCANBUILD_ARGS SCANBUILD_OUTPUTDIR SIZE TOOLCHAIN UNDEF WERROR
OLDPWD=/home/cbuec/RIOTstuff/riot-vanilla
HOME=/home/cbuec
MAKEFLAGS=rw -- RIOTBASE=/home/cbuec/RIOTstuff/riot-vanilla/RIOT RIOTCPU=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/cpu RIOTMAKE=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/makefiles RIOTPKG=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/pkg RIOTTOOLS=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/dist/tools RIOTPROJECT=/home/cbuec/RIOTstuff/riot-vanilla/RIOT APPDIR=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world BUILD_DIR=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/build BINDIRBASE=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world/bin BINDIR=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world/bin/native64 PKGDIRBASE=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/build/pkg DLCACHE_DIR=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/.dlcache
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
WSL_DISTRO_NAME=Ubuntu-22.04
MAKE_TERMERR=/dev/pts/0
WAYLAND_DISPLAY=wayland-0
LOGNAME=cbuec
PULSE_SERVER=unix:/mnt/wslg/PulseServer
NAME=W11nMate
WSL_INTEROP=/run/WSL/429_interop
_=/usr/bin/make
BOARDS=feather-nrf52840
TERM=xterm-256color
TESTRUNNER_CONNECT_DELAY=2
DOCKER_ENV_VARS_ALWAYS=DISABLE_MODULE DEFAULT_MODULE FEATURES_REQUIRED FEATURES_BLACKLIST FEATURES_OPTIONAL USEMODULE USEPKG
PATH=...
WT_SESSION=d75ddbc5-9743-4e59-87ac-f7dd02175291
XDG_RUNTIME_DIR=/run/user/1000/
MAKELEVEL=1
DISPLAY=:0
LANG=C.UTF-8
LS_COLORS=...
SHELL=/bin/bash
DOCKER_IMAGE=docker.io/riot/riotbuild@sha256:0993a39a90e0b573637ea488d51fb33ba1cbeacbfbae4a27cca2091d0873d383
MAKE_TERMOUT=/dev/pts/0
LESSCLOSE=/usr/bin/lesspipe %s %s
DOCKER_BUILD_ROOT=/data/riotbuild
PWD=/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world
XDG_DATA_DIRS=/usr/local/share:/usr/share:/var/lib/snapd/desktop
MFLAGS=-rw
WSL2_GUI_APPS_ENABLED=1
HOSTTYPE=x86_64
WSLENV=WT_SESSION:WT_PROFILE_ID:
DOCKER_MAKECMDGOALS=

success.
make: Leaving directory '/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world'
cbuec@W11nMate:~/RIOTstuff/riot-vanilla/RIOT$
```

There is the seemingly harmless variable named `MAKEFLAGS` that contains the variable `BINDIR`, which is not overridden by a new make call and therefore everything is built in the `native64` folder (it was `native` when this was written, but time goes on :D ).

This is somewhat similar to #21019, but now I have an idea how to fix it: Add `env --unset=MAKEFLAGS` in front of the two `$(MAKE)` calls.

<hr/>

```
cbuec@W11nMate:~/RIOTstuff/riot-vanilla/RIOT$ BOARDS=feather-nrf52840 make -C examples/basic/hello-world/ buildtest
make: Entering directory '/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world'
using BOARD="native64" as "native" on a 64-bit system
Building for feather-nrf52840 ... success.
make: Leaving directory '/home/cbuec/RIOTstuff/riot-vanilla/RIOT/examples/basic/hello-world'
cbuec@W11nMate:~/RIOTstuff/riot-vanilla/RIOT$ ls examples/basic/hello-world/bin
feather-nrf52840
```

As you can see, the correct folder got created. There is a spurious message coming from the newly introduced `makefiles/board-alias.inc.mk` in #21242 because the `BOARD` defaults to `native` from the `examples/basic/hello-world/Makefile`.
This needs another check in the `board-alias.inc.mk` file, but that's no big deal.

closed 2025-04-06 by crasbe