RIOT-OS/RIOT #9753: periph/i2c: Possibility to set I2C bus speed is missing in new I2C

gschorcht, 2018-08-09:
With new I2C the ```speed``` parameter was dropped in the ```i2c_init``` function.

The reason was explained in PR #6576: "... As I2C is not like SPI, where each device on the bus can be interfaces with a different clock speed, the bus clock for I2C buses needs to be assigned globally for all devices on the bus, so we might as well define it as part of the I2C config in the boards periph_conf.h. ..."

Although it is true that bus speed needs to be globally defined for all I2C devices, I'm not really happy with the approach to define the I2C bus speed in the board configuration. I2C is a peripheral interface and the board can't know what devices are connected to it. The only instance that knows what devices are connected is the application. Therefore, the application should at least have the possibility to set the bus speed through an API function.

For example, the I2C bus speed for Arduino Zero is defined as ```I2C_SPEED_FAST``` in ```periph_conf.h```. If a sensor device is connected that only supports ```I2C_SPEED_NORMAL```, the bus configuration would have to be changed. As I understand the concept of board configuration, it should be fixed for a certain board, e.g., Arduino Zero, and shouldn't be changed for a certain application.

From my point of view, the bus speed should be set in ```i2c_init``` to ```I2C_SPEED_NORMAL``` (or maybe a default bus speed defined in board configuration), but the application should be able to set it to a higher bus speed using an API function like ```i2c_set_speed``` if all devices support this speed.

dylad, 2018-08-09:
>  I2C is a peripheral interface and the board can't know what devices are connected to it

I disagree with you on this point, I2C is not designed to be a plug’n’play system like USB. What I mean here is that sensors are already present on a board, so we know them and there is no need to enumerate or scan throught available services/capabilities (like Bluetooth). In addition, add the speed parameter to the device description could work, but if you have several sensors on the same bus, we would have to add a protection to ensure all sensors want the same speed which takes space  as we tried to save as much RAM/ROM as we can. This was the main approach when designing the new interface.

Yes you can use a board and plug a daughter board with a new sensor on it so you will have to append the periph_conf.h file, re-compile and re-flash your board. I agree this is boring some times when you’re playing with several boards and sensors but again this is for ROM/RAM space sake.

Of course, we could expand the API if there is a consensus on it. Personally, I have no strong opinion on it but I cannot think of an application which need to change dynamically I2C speed so I’m a bit puzzled.

> From my point of view, the bus speed should be set in i2c_init to I2C_SPEED_NORMAL

Yes, I2C_SPEED_NORMAL could be used as default for all periph_conf.h

>  Arduino Zero is defined as I2C_SPEED_FAST

Hmm, this could be change to I2C_SPEED_NORMAL as default in its periph_conf.h, I am not opposed to that.

gschorcht, 2018-08-10:
> What I mean here is that sensors are already present on a board, so we know them

This is probably the case for special boards like the IoT-LAB M3 open node, but not for generic boards like Arduino*. From the board definition's point of view, the sensors connected at I2C bus are unknown for such generic boards.

> Yes you can use a board and plug a daughter board with a new sensor on it so you will 
> have to append the periph_conf.h file, re-compile and re-flash your board. I agree this is
> boring some times when you’re playing with several boards and sensors but again this 
> is for ROM/RAM space sake.

Should it really be RIOT's concept of board definitions that the application developer changes the board definition for a generic board according to his requirements? The difference between the configuration done by the application and the board is that the application configuration is not part of the RIOT repository and under control of the application developer. What if he has different applications with different requirements for the same generic board?

> .. but I cannot think of an application which need to change dynamically I2C speed
> so I’m a bit puzzled.

The speed should not be changed dynamically. The application should only have the chance to set it during the initialization in the main thread. The application developer knows the sensors connected to a generic board and has to add according drivers using USEMODULE variable in its makefile. So, he also could call the function ```i2c_set_speed``` in the main thread. If not, the default bus speed is used. For special boards with all sensors on board, this aproach would also work.

dylad, 2018-08-11:
> The difference between the configuration done by the application and the board is that the application configuration is not part of the RIOT repository and under control of the application developer

is this an issue ? This is also true for PWM, ADC and so on. If you are using an Arduino board with different daughter boards you're likely to change so ADC or PWN channels because these modules doesn't use the same pins, or you need to add a second UART definitions because your GSM module is a fancy model with GPS. 
Modify the periph_conf.h file as an app developer must be done very often especially if you're using a generic board with daughter boards.
Yes, this is boring when you need to handle several configuration for the same board and it can be 'solve' by either #ifdef some parameters in the periph_conf.h and allow users to append these variable in their Makefile but this will uglify the periph_conf.h file a lot and I'm afraid some maintainers will be against it. Or, you can add new API functions which will allow more versatility/flexibility but at RAM/ROM costs.

It would be great to have the point of views of others maintainers on this topic.

MrKevinWeiss, 2018-08-13:
Since I2C speed is dependent on the pullup resistor value and the pullups are a board feature and not a sensor/daughter board feature (you don't want multiple resistors in parallel) I think it should stay the way it is.  This also happens to be less work since it is already done...  Does anyone have an example of a sensor that cannot receive and I2C clock less than 400kHz?

kaspar030, 2018-08-14:
IMO, ```i2c_set_speed()``` should be present. All periph/i2c implementations need the corresponding code anyways, so this is just a matter of factoring it in its own function.

This would also be the entry point to hot-pluggable i2c devices, or at least pluggable without re-compilation. Work is being done in that direction (by e.g., the rapstore guys).

While for compile-time compilation that is not necessary, it is also cumbersome to manually find the highest supported I2C-speed supported by all peripherals connected to a bus. Usually the maximum supported speed is desired, and device drivers know best what their device can do. One could think of ```i2c_max_speed(i2cdev, speed),i2c_max_speed_release(i2cdev, speed)``` that would block/unblock speed settings as done for pm levels.

gschorcht, 2018-08-14:
@MrKevinWeiss 
> I think it should stay the way it is. This also happens to be less work since it is already done

Introducing such a function should not require to much work following the ```#define HAVE_*``` approach. The function would have to be declared in ```driver/include/periph/i2c.h```.  In ```driver/periph_common/i2c.c``` a function dummy could be defined which simply returns ```-EOPNOTSUPP```. This dummy is only compiled if ```HAVE_I2C_SET_SPEED``` is not defined. The implementation of a CPU could then define the real function and ```HAVE_I2C_SET_SPEED```. This could be introduced, if necessary, step by step, without changing anything immediatly..

dylad, 2018-09-18:
> That would be great, the board defines what bus speed it can support and device drivers define what bus speed they can use at maximum. The common maximum is then used.

Again this is at RAM/ROM cost.

>  I wish RIOT had an application-specific configuration approach that could override default board definitions.

I think this is good idea in general but out of the scope of this issue.

> All periph/i2c implementations need the corresponding code anyways, so this is just a matter of factoring it in its own function.

Yes adding such a function is trivial. If this is the direction everyone want, I'll not be opposed.

gschorcht, 2019-09-10:
I'm still having it on my ToDo-List.

MrKevinWeiss, 2019-09-10:
I also would like to expose the clock param since it would make the automated tests easier...

MichelRottleuthner, 2020-02-06:
Just stumbled upon this and I also agree that some more dynamic way of configuration is needed.
The practical use case for me was simply testing a RIOT sensor driver on multiple boards - for every single one I needed to fiddle around with another periph_conf file - adding a custom periph_conf to the test application obviously won't cut it in this case.
Most use cases would already be covered by compile-time configuration like it is already possible on the ESPs. I.e. overwriting the configuration from an application Makefile: `CFLAGS+=-DI2C0_SPEED=I2C_SPEED_NORMAL`.

smlng, 2020-02-18:
I'm not too much into all the hardware stuff (like required (or not) pull-ups and how the I2C depends on or is affected by them). However, I recently discussed this topic IRL with @MichelRottleuthner and we came to the conclusion (please correct me if I'm wrong 😉) the I2C speed shouldn't be compile-time configuration (as of now) but rather run-time configurable. 
This is also required for the general idea of being able to link an application against existing objects, without re-compiling them (see @kaspar030 comment above). 

API wise to me `speed` should ideally be part of the init function, because it is not as dynamic as in SPI (other pointed this out here too) so a separate `i2c_set_speed` would be counter-intuitive. An application has to know the common speed of all devices connected and initialise the bus accordingly. In general we might also want to have a proper de-init/power-off function to disable the I2C bus and being able to re-init later again. This would also allow for corner-cases where (kind-of) hot plugging I2C slaves is required with different speeds.

MichelRottleuthner, 2020-02-18:
> the I2C speed shouldn't be compile-time configuration (as of now) but rather run-time configurable. 

Just a minor addition: run-time configurable is IMO the best way and should be the goal, but proper compile-time configuration would still be preferable over what we have now and would be possible without API change.