RIOT-OS/RIOT #9766: Initial UART data corruption with libfixmath test after flash

MrKevinWeiss, 2018-08-13:
#### Description
It appears that sending lots of data after flashing has corrupted data when using `make flash term BOARD=...`.    I tried this on nucleo boards and samr21 with the same result.  I tried adding extra timeout after resetting.  If a reset button is pressed after the it seems to be able to handle the data.  If a sleep or delay is added before spitting out data in the main function if fixes it.  I tried with the nucelo connection (stlink), a separate uart to usb converter and the samr debug port.  Maybe the tests should always have something that starts running instead of just dumping data like a [START] command or something?

#### Steps to reproduce the issue
`make BOARD=samr21-xpro flash term -C tests/libfixmath`

#### Expected results
`Unary` in the first line after the connection is complete

#### Actual results
Varying and corrupt data, usually missing the first 20 or so characters.

cladmi, 2018-08-13:
More precision, it is when doing `make flash test` and so in practice having `make term` open and then `make reset`. Which should after reset, have the first node output in the terminal.

cladmi, 2018-10-21:
After debugging this before doing the issue and also other issues with `jlink.sh` found while debugging with @kYc0o I start having ideas on where it may come from.
`pexpect` only reads from the child when we an `expect` command is run.

With `jlink.sh`, `make test` could not be run because, we found/thought that, the `rtt_server` was not fully started as it was stuck doing a `printf` on a full `pipe`.
The fix in that case was to dismiss the output so not be stuck

https://github.com/RIOT-OS/RIOT/pull/9916/commits/dfd349a05b0fb5ae3a0c8f65c25cafa6dc333d6c#diff-6aa5730f69738bacea66a7d92c64489bR248

And as we found that having `make term` running while resetting does not discard characters, I start thinking it could also be that the `TERMPROG` is stuck trying to write an output, so does not process reading the node `dev` anymore and so characters are dropped.

I did not have time to investigate too much more though. But if it is the case, it would make the `MAKE_TERM_STARTED_DELAY` even more problematic in these cases were the node floods the output.

One solution I would see for this, would be to have a thread running in `testrunner` that reads everything continuously from the `child` and then `expect` would be reading from the buffered lines.
But it requires real debugging before going for this.

MrKevinWeiss, 2018-10-22:
Is it crazy to think that we just stay this is a constraint of term?  Any tests or application that want to flood the buffer and depend on term will have to wait a bit or better yet coordinate (which I think all tests should do)

cladmi, 2018-10-23:
To be precise, from my analysis, it is not a constraint of `term` but of the program running the tests.

I agree with your suggestion. It is true that currently, an app that flood the uart has to wait a bit or coordinate. It is a fact. So the easy fix is to fix the applications now with a sync mechanism (sleep delay).

However, my tooling side says that I could still, later, fix the tool to allow this behavior.
But providing a hack fixing the test now as you did is better than waiting 6 month for a global fix.

PR #10201 (2018-10-25): tests/libfixmath: Add delay at start of test

### Contribution description

Add a delay to the start of the test since many boards start outputting before a connection is formed and the input buffer gets flooded.

### Testing procedure

`BOARD=nucleo-f103rb make flash term -C tests/libfixmath`
check the first output, it should be `Unary.`
compare against master.

_I have tested on nucleo but it should be tested on other boards especially with different default clock speeds since it uses a dumb delay_ 
### Issues/PRs references

fixes #9766
closed 2018-10-25 by jia200x