RIOT-OS/RIOT #9775: hello-world example crashes on BOARD=nucleo-f446re

benVolatiles, 2018-08-14:
#### Description

Current RIOT master fails to run on the [Nucleo-F446RE](https://www.st.com/en/evaluation-tools/nucleo-f446re.html) eval board.

#### Steps to reproduce the issue
I checked out the latest RIOT master (2018.10-devel-463-g0c14681a3) and ran `make BOARD=nucleo-f446re flash` in `/examples/hello-world`.

#### Expected results
```
Hello World!
You are running RIOT on a(n) nucleo-f446re board.
This board features a(n) stm32f446re MCU.
```

#### Actual results
No output could be observed on `/dev/ttyACM0`

`make BOARD=nucleo-f446re debug` reveals the program ended up in the `hard_fault_handler` at `/cpu/cortexm_common/vectors_cortexm.c:272`

This will not happen when all calls to `puts` / `printf` / `LOG_INFO` are removed.

Running the binary in Renode results in

```
13:53:36.3828 [ERROR] cpu: PC does not lay in memory or PC and SP are equal to zero. CPU was halted.
13:53:36.3834 [INFO] cpu: Setting initial values: PC = 0x30303030, SP = 0x3032303A.
13:53:36.3838 [WARNING] cpu: Patching PC 0x30303030 for Thumb mode.
```

and no output on USART2 either.

#### Versions
Operating system: Ubuntu 18.04

Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
                avr-gcc: avr-gcc (GCC) 5.4.0
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
      riscv-none-embed-gcc: missing
                  clang: clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

Installed compiler libs
-----------------------
      arm-none-eabi-newlib: "2.4.0"
      mips-mti-elf-newlib: missing
      riscv-none-embed-newlib: missing
      avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: Cppcheck 1.82
                doxygen: missing
                 flake8: missing
                    git: git version 2.17.1
             coccinelle: missing

PeterKietzmann, 2018-08-17:
Thanks for reporting this issue. Could you `git bisect`? I think people just used that board recently.

benVolatiles, 2018-08-21:
Hm, that's the thing - I can't find a `good` revision. Going back to 2016.10 and `BOARD=nucleo-f446` (and all tags in between) shows the same issue.

So it might be a problem with my toolchain, but then again I use the same toolchain to compile our (non-RIOT) firmware for this board (sans newlib). And this is just the standard toolchain straight from the Ubuntu repository.

PeterKietzmann, 2018-08-21:
Could you try to build with [Docker](https://github.com/RIOT-OS/RIOT/wiki/Use-Docker-to-build-RIOT)?

benVolatiles, 2018-08-21:
Yes it works with docker! Should I close the issue?

PeterKietzmann, 2018-08-22:
I would leave it up to you to close the issue. Could it be related to [this one](https://github.com/ArduPilot/ardupilot/issues/8663)? If I'm being honest, I just had similar problems after the update to Ubuntu 18.04 and re-instelled `gcc-arm-embedded`. I'm just a bit surprised that the same toolchain works smoothly for you in an other project.

forellendev, 2021-08-12:
Hi, this problem still persists when run on Ubuntu 20.04 64 bit with gcc version 10.3.1:

`/usr/bin/objcopy: Unable to recognise the format of the input file `/home/user/Documents/test/network/project01/arm/RIOT/examples/hello-world/bin/nucleo-f446re/hello-world.elf'`