RIOT-OS/RIOT #9784: enc26j60 stuck in a loop

maribu, 2018-08-15:
# Description
The enc28j60 driver occasionally stucks in the following loop:

https://github.com/RIOT-OS/RIOT/blob/90c4ddcfb135ea7c7ac01528938c5038a37abaab/drivers/enc28j60/enc28j60.c#L469-L472

# Steps to reproduce the issue
Let the enc28j60 receive a lot of traffic and wait for some time. In my hostile network environment I have to wait for something between 2 seconds to 60 seconds for the driver to get stuck in the loop.

# Further details
After adding this debug code
``` patch
diff --git a/drivers/enc28j60/enc28j60.c b/drivers/enc28j60/enc28j60.c
index 1ba6240d0..1eb15b976 100644
--- a/drivers/enc28j60/enc28j60.c
+++ b/drivers/enc28j60/enc28j60.c
@@ -31,7 +31,7 @@
 #include "enc28j60.h"
 #include "enc28j60_regs.h"
 
-#define ENABLE_DEBUG    (0)
+#define ENABLE_DEBUG    (1)
 #include "debug.h"
 
 /**
@@ -466,10 +466,13 @@ static void nd_isr(netdev_t *netdev)
             }
         }
         if (eir & EIR_PKTIF) {
+            uint8_t pkt_count;
             do {
                 DEBUG("[enc28j60] isr: packet received\n");
                 netdev->event_callback(netdev, NETDEV_EVENT_RX_COMPLETE);
-            } while (cmd_rcr(dev, REG_B1_EPKTCNT, 1) > 0);
+                pkt_count = cmd_rcr(dev, REG_B1_EPKTCNT, 1);
+                DEBUG("[enc28j60]: %i packets remaining\n", (int)pkt_count);
+            } while (pkt_count > 0);
         }
         if (eir & EIR_RXERIF) {
             DEBUG("[enc28j60] isr: incoming packet dropped - RX buffer full\n");
```

I could record this serial output on an nucleo-f446re connected to an ENC28J80 breakout adapter:

```
/home/maribu/Repos/software/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"
2018-08-15 17:22:10,803 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
2018-08-15 17:22:11,806 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:11,807 - INFO # [enc28j60] recv: unable to get packet - buffer too small
2018-08-15 17:22:11,808 - INFO # [enc28j60]: 1 packets remaining
2018-08-15 17:22:11,808 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:11,809 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:11,810 - INFO # [enc28j60] isr: packet transmitted
2018-08-15 17:22:11,810 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:11,811 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:11,811 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:11,812 - INFO # [enc28j60] recv: unable to get packet - buffer too small
2018-08-15 17:22:11,813 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:11,813 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:11,814 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:11,815 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:11,815 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:13,240 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:13,245 - INFO # [enc28j60] recv: unable to get packet - buffer too small
2018-08-15 17:22:13,248 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:13,794 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:13,798 - INFO # [enc28j60]: 0 packets remaining
2018-08-15 17:22:14,842 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:14,846 - INFO # [enc28j60]: 0 packets remaining
[... 184 lines cut ...]
2018-08-15 17:22:35,611 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:35,614 - INFO # [enc28j60]: 2 packets remaining
2018-08-15 17:22:35,617 - INFO # [enc28j60] isr: packet received
2018-08-15 17:22:35,619 - INFO # [enc28j60]: 2 packets remaining
...
```

Adding this DEBUG statement completely solves the problem:

```
diff --git a/drivers/enc28j60/enc28j60.c b/drivers/enc28j60/enc28j60.c
index 1eb15b976..901e98b2a 100644
--- a/drivers/enc28j60/enc28j60.c
+++ b/drivers/enc28j60/enc28j60.c
@@ -330,6 +330,7 @@ static int nd_recv(netdev_t *netdev, void *buf, size_t max_len, void *info)
     next = (uint16_t)((head[1] << 8) | head[0]);
     size = (uint16_t)((head[3] << 8) | head[2]) - 4;  /* discard CRC */
 
+    DEBUG("[enc28j60] size: %i, next: %i\n", (int)size, (int)next);
     if (buf != NULL) {
 #ifdef MODULE_NETSTATS_L2
         netdev->stats.rx_count++;
```

So maybe timing is relevant for this issue to appear?

maribu, 2018-08-15:
Btw: This seems to occur only the next packet received after receiving a packet bigger than the buffer (`[enc28j60] recv: unable to get packet - buffer too small`). But it doesn't occur every time an incoming packet was dropped.

PeterKietzmann, 2018-08-16:
@haukepetersen, @gschorcht, @smlng might be interested

gschorcht, 2018-08-20:
I can take a look on it. When I was investigating the TX lock, I had a crash since ```size``` determined from ```head[]``` had a negative value. Maybe, it is the same issue.

gschorcht, 2018-08-20:
@maribu I can reproduce the behavior immediately when flooding the enc28j60 node with ICMP echo request with maximum size
```
sudo -f ping6 fe80::b48a:eeff:fe23:2323 -I eth0 -s 65507
```
Give me some time to investigate the problem.

gschorcht, 2018-08-21:
I have added some additonal DEBUG instructions. Following shows the normal case with only some traffic. ```XXX``` indicates the entry to ```nd_isr```. Mostly, only one packet is processed in the ```while``` loop. From this output we can see that the RX ring buffer is working correctly.
```
XXX
[enc28j60] isr: packet received 72714787
[enc28j60] size: 63, next: 4892
[enc28j60] size: 63, next: 4892
[enc28j60]: 1 packets remaining
[enc28j60] isr: packet received 72727083
[enc28j60] size: 63, next: 4966
[enc28j60] size: 63, next: 4966
[enc28j60]: 0 packets remaining
XXX
[enc28j60] isr: packet received 75478991
[enc28j60] size: 397, next: 5374
[enc28j60] size: 397, next: 5374
[enc28j60]: 0 packets remaining
XXX
[enc28j60] isr: packet received 75601200
[enc28j60] size: 413, next: 5798
[enc28j60] size: 413, next: 5798
[enc28j60]: 0 packets remaining
XXX
[enc28j60] isr: packet received 75721889
[enc28j60] size: 415, next: 80
[enc28j60] size: 415, next: 80
[enc28j60]: 0 packets remaining
XXX
[enc28j60] isr: packet received 75789713
[enc28j60] size: 171, next: 262
[enc28j60] size: 171, next: 262
[enc28j60]: 0 packets remaining
XXX
[enc28j60] isr: packet received 75842568
[enc28j60] size: 403, next: 676
[enc28j60] size: 403, next: 676
[enc28j60]: 0 packets remaining
```
When we flood the interface with traffic following output can be observed.
```XXX
[enc28j60] isr: packet received 423407449
[enc28j60] size: 86, next: 754
[enc28j60] size: 86, next: 754
[enc28j60]: 0 packets remaining
XXX
[enc28j60] isr: packet transmitted 423420822
[enc28j60] isr: packet received 423424848
[enc28j60] size: 562, next: 1326
[enc28j60] size: 562, next: 1326
[enc28j60]: 8 packets remaining
[enc28j60] isr: packet received 423437854
[enc28j60] size: 562, next: 1898
[enc28j60] size: 562, next: 1898
[enc28j60]: 8 packets remaining
[enc28j60] isr: packet received 423450857
[enc28j60] size: 562, next: 2470
[enc28j60] size: 562, next: 2470
[enc28j60]: 8 packets remaining
[enc28j60] isr: packet received 423463860
[enc28j60] size: 562, next: 3042
[enc28j60] size: 562, next: 3042
[enc28j60]: 8 packets remaining
[enc28j60] isr: packet received 423476864
[enc28j60] size: 562, next: 3614
[enc28j60] size: 562, next: 3614
[enc28j60]: 8 packets remaining
[enc28j60] isr: packet received 423489867
[enc28j60] size: 562, next: 4186
[enc28j60] size: 562, next: 4186
[enc28j60]: 9 packets remaining
[enc28j60] isr: packet received 423502870
[... 96 lines cut ...]
[enc28j60] size: 562, next: 4136
[enc28j60] size: 562, next: 4136
[enc28j60]: 10 packets remaining
[enc28j60] isr: packet received 423825728
[enc28j60] size: 562, next: 4136
[enc28j60] size: 562, next: 4136
[enc28j60]: 10 packets remaining
[enc28j60] isr: packet received 423838306
[enc28j60] size: 562, next: 4136
[enc28j60] size: 562, next: 4136
[enc28j60]: 10 packets remaining
[enc28j60] isr: packet received 423850885
[enc28j60] size: 562, next: 4136
[enc28j60] size: 562, next: 4136
[enc28j60]: 10 packets remaining
[enc28j60] isr: packet received 423863464
[enc28j60] size: 562, next: 4136
[enc28j60] size: 562, next: 4136
[enc28j60]: 10 packets remaining
[enc28j60] isr: packet received 423876042
[enc28j60] size: 562, next: 4136
[enc28j60] size: 562, next: 4136
[enc28j60]: 10 packets remaining
```
For some reason the ring buffer seem to be hanging.

gschorcht, 2018-08-21:
I'm not really sure anymore whether it is problem of the enc28j60 driver. I have added some additional ```printfs``` to ```nd_recv```, but this time inside 
```
if (buf != NULL) {
    DEBUG("YYYY %d %d\n", rx_rd_ptr, ERXRDPT_TO_NEXT(rx_rd_ptr));
    ...
    DEBUG("ZZZZ %d %d\n", next, NEXT_TO_ERXRDPT(next));
}
```
These additional outputs only appear when function ```nd_recv``` is called with ```buf != NULL``` by upper layers.

When the endless loop problem occurs, the output looks now:
```
[enc28j60] isr: packet received 28820266
[enc28j60] size: 562, next: 1114
[enc28j60] size: 562, next: 1114
YYYY 541 542
ZZZZ 1114 1113
[enc28j60]: 3 packets remaining
[enc28j60] isr: packet received 28835670
[enc28j60] size: 562, next: 1686
[enc28j60] size: 562, next: 1686
YYYY 1113 1114
ZZZZ 1686 1685
[enc28j60]: 3 packets remaining
[enc28j60] isr: packet received 28851247
[enc28j60] size: 562, next: 2258
[enc28j60] size: 562, next: 2258
YYYY 1685 1686
ZZZZ 2258 2257
[enc28j60]: 4 packets remaining
[enc28j60] isr: packet received 28866826
[enc28j60] size: 562, next: 2830
[enc28j60] size: 562, next: 2830
[enc28j60]: 5 packets remaining
[enc28j60] isr: packet received 28879231
[enc28j60] size: 562, next: 2830
[enc28j60] size: 562, next: 2830
[enc28j60]: 6 packets remaining
[enc28j60] isr: packet received 28891637
[enc28j60] size: 562, next: 2830
[enc28j60] size: 562, next: 2830
[enc28j60]: 7 packets remaining
[enc28j60] isr: packet received 28904042
[enc28j60] size: 562, next: 2830
[enc28j60] size: 562, next: 2830
```
```size``` and ```next``` appear two times since the first call of ```nd_recv``` uses ```buf = NULL``` just to ask for next packet size.

In normal operation the additional outputs appear in the second call of ```nd_recv```. When endless loop happens, ```size``` and ```next``` are also printed out two times, but the function seems to be called with ```buf = NULL``` all the time. Thus the ring buffer is not really read and seem to hang in an endless loop.

gschorcht, 2018-08-21:
Hm, as supposed, when the problem happens, function ```nd_recv``` is called with ```buf = NULL``` all the time.

[enc28j60] isr: packet received 7135403
[enc28j60] size: 562, next: 5752 buf=0
[enc28j60] size: 562, next: 5752 buf=0x3ffefaf0
YYYY 5179 5180
ZZZZ 5752 5751
[enc28j60]: 5 packets remaining
[enc28j60] isr: packet received 7152721
[enc28j60] size: 562, next: 180 buf=0
[enc28j60] size: 562, next: 180 buf=0x3ffefaf0
YYYY 5751 5752
ZZZZ 180 179
[enc28j60]: 5 packets remaining
[enc28j60] isr: packet received 7169681
[enc28j60] size: 562, next: 752 buf=0
[enc28j60] size: 562, next: 752 buf=0
[enc28j60]: 6 packets remaining
[enc28j60] isr: packet received 7182867
[enc28j60] size: 562, next: 752 buf=0
[enc28j60] size: 562, next: 752 buf=0

gschorcht, 2018-08-21:
Enabling debugging in ```sys/net/gnrc/netif/gnrc_netif_ethernet.c``` gives more information about possible reason, at least what I see on my platform.

Successful operation looks like following:
```
[enc28j60] isr: packet received 11441632
[enc28j60] size: 60, next: 3642 buf=0
[enc28j60] size: 60, next: 3642 buf=0x3ffefce8
YYYY 3571 3572
ZZZZ 3642 3641
gnrc_netif_ethernet: received packet from 00:15:99:9e:cb:0f of length 60
[enc28j60]: 13 packets remaining
```
Later, I get the following:
```
[enc28j60] isr: packet received 11464968
[enc28j60] size: 562, next: 4214 buf=0
[enc28j60] size: 562, next: 4214 buf=0x3ffefce8
YYYY 3641 3642
ZZZZ 4214 4213
gnrc_netif_ethernet: no space left in packet buffer
[enc28j60]: 13 packets remaining
```
When the endless loop happens, I get the following:
```
[enc28j60] isr: packet received 11908213
[enc28j60] size: 562, next: 1384 buf=0
gnrc_netif_ethernet: cannot allocate pktsnip.
[enc28j60] size: 562, next: 1384 buf=0
```
That seem to be the reason why ```nd_recv``` is always called with ```buf = NULL```. So it seems to be a resource problem when packets are arriving faster than they can be processed.

What I have also observed is that function ```nd_isr``` is never left when the endless loop happens.

gebart, 2018-08-21:
http://api.riot-os.org/structnetdev__driver.html#ae2c8cad80067e3b1f9979931ddb3cc8b
>If buf == NULL and len == 0, returns the packet size without dropping it. If buf == NULL and len > 0, drops the packet and returns the packet size.

@gschorcht check whether the enc driver actually implements dropping frames when buf = NULL and len > 0. If it doesn't, then that is a bug.

gschorcht, 2018-08-21:
@gebart Thanks for the hint. That seems to be the problem.

maribu, 2018-08-21:
@gschorcht, @gebart: Thanks for investigating the problem. I opened a new Issue to start a discussion on the API of `netdev_driver::recv`, as I consider this a textbook example of a misleading API

closed 2018-08-21 by maribu