RIOT-OS/RIOT #9785: AT parser driver improvements

keestux, 2018-08-15:
Hi,

With this "issue" I want to be able to give some feedback about the AT parser driver, and also start a discussion about its usage.

Besides the normal code documentation there is no description how the driver is supposed to be used. Also for the URC part there is no real description.

keestux, 2018-08-15:
To start of with a first simple question. Let's look at some code duplication. At least, it's almost duplication, which deserves an explanation.

    int at_send_cmd(at_dev_t *dev, const char *command, uint32_t timeout)
    {
        size_t cmdlen = strlen(command);
    
        uart_write(dev->uart, (const uint8_t *)command, cmdlen);
        uart_write(dev->uart, (const uint8_t *)AT_SEND_EOL, AT_SEND_EOL_LEN);
    
        if (AT_SEND_ECHO) {
            if (at_expect_bytes(dev, command, timeout)) {
                return -1;
            }
    
            if (at_expect_bytes(dev, AT_SEND_EOL AT_RECV_EOL_1 AT_RECV_EOL_2, timeout)) {
                return -2;
            }
        }
    
        return 0;
    }

and

    int at_send_cmd_wait_prompt(at_dev_t *dev, const char *command, uint32_t timeout)
    {
        unsigned cmdlen = strlen(command);
    
        at_drain(dev);
    
        uart_write(dev->uart, (const uint8_t *)command, cmdlen);
        uart_write(dev->uart, (const uint8_t *)AT_SEND_EOL, AT_SEND_EOL_LEN);
    
        if (at_expect_bytes(dev, command, timeout)) {
            return -1;
        }
    
        if (at_expect_bytes(dev, AT_SEND_EOL AT_RECV_EOL_2, timeout)) {
            return -2;
        }
    
        if (at_expect_bytes(dev, ">", timeout)) {
            return -3;
        }
    
        return 0;
    }

The two functions are nearly identical, except for

- waiting for the prompt (of course)
- drain
- with or without `AT_RECV_EOL_1`

Are these last two differences deliberate? If so, why?
Why can't we just use `at_send_cmd` in `at_send_cmd_wait_prompt`?

kaspar030, 2018-08-16:
Thanks for working on this!

> Are these last two differences deliberate? If so, why?
> Why can't we just use at_send_cmd in at_send_cmd_wait_prompt?

Same answer as last time: two authors, no time. No reason not to refactor.

vincent-d, 2018-08-16:
> No reason not to refactor.

True, but there is one reason though. In `at_send_cmd_wait_prompt` the EOL is different on purpose.

Normal case, we send:  `AT+CMD\r` we receive `AT+CMD\r\r\nOK`.
In case of a prompt: we send `AT+CMD\r`, we receive `AT+CMD\r\n>`.
That's why there are 2 different functions. But I think it could be improved by making the prompt character configurable as it seems some device use different prompt characters.

keestux, 2018-08-27:
@vincent-d Which modem device(s) do you usually work with?

I have a Ublox SARA-U270 and a SARA-G350. In the Ublox manual (UBX-13002752) I can't find anything about a different reply in case of a prompt.

closed 2019-08-12 by keestux