RIOT-OS/RIOT #9789: Repeating definitions in the web api documentation

HendrikVE, 2018-08-16:
#### Description
In the API documentation on the web are several tables with repeating items.

#### Steps to reproduce the issue
Open the [api](https://riot-os.org/api/cc2538_2include_2periph__cpu_8h.html#a491a2cbfb4e94f2afcc0d5bdef2dc454) in a browser.

#### Expected results
Only one definition per item

#### Actual results
GPIO_OD_PU is defined 3 times there, always with another definition. Its not the only case where definitions are failing.

gschorcht, 2018-11-30:
Everytime, a platform specific declaration of a type is done in `cpu/*/include/periph_cpu.h` to override the default declaration defined in `drivers/include/perihp/*.h`, it contributes its own declaration to doxygen. Not only that there are a lot of duplicates, also their documentation are completely different and inconsistent. More strange is that sometimes cpu cc2538 is shown as the source.

jcarrano, 2018-12-03:
I also stumbled on this when trying to insert doxygen docs into Sphinx - it causes duplicate references/anchors to be generated, which is a pain.

I'll try to investigate this too. In these cases I find it useful to enable XML generation in doxygen and then to inspect the XML documents (after pretty-formatting, for my own sanity.)

jcarrano, 2018-12-03:
I did some research. This seems to be a Doxygen issue.

I made a minimal example here: https://gist.github.com/jcarrano/c26c2b2590a85daf878feb2ec0558e43

- Two .h files.
- Two enums of the same name in the same group.
- Doxygen produces docs for the two enums but:
  - The enums are merged: both share the same id (`group__plane_1gac0c8f50204f6f11b72a3c00548a813ea`)
  - For the value `Chicken`, it also *merges* the two values. You can see in the XML there are repeated definitions **with the same id** (`group__plane_1ggac0c8f50204f6f11b72a3c00548a813eaae6b50df914ce624e18cf9ab683d2baec`) but different content.

This seems to be a known issue:

- https://github.com/doxygen/doxygen/issues/2344
- https://github.com/doxygen/doxygen/issues/6617
- http://doxygen.10944.n7.nabble.com/duplicate-enumerator-symbols-tt730.html#none

I tried with the latest Doxygen master and the issue is still there.

gschorcht, 2018-12-19:
@jcarrano Thank you for investigating the problem in more detail. I'm not sure, but it might be that

- https://github.com/doxygen/doxygen/issues/2344
- https://github.com/doxygen/doxygen/issues/6617

are describing a problem that is slightly different. They have different enumerations with same member names. In our case, we have enumerations with same name and same as well as additional member names.

For example, for `gpio_mode_t` the default definition adds members
https://github.com/RIOT-OS/RIOT/blob/3ce4b7454f9213eaf3ba1a190ed234f5c2456b26/drivers/include/periph/gpio.h#L96-L105 to the enumeration type  of `gpio_mode_t`.

When doxygen is parsing the file, I suppose that it is looking for an existing element. If there is no one, it creates a new one. After that, it adds all members.

When doxygen is parsing another file where `gpio_mode_t` is defined, e.g., https://github.com/RIOT-OS/RIOT/blob/3ce4b7454f9213eaf3ba1a190ed234f5c2456b26/cpu/esp32/include/periph_cpu.h#L135-L146 it is doing the same.

Since there is already an element for `gpio_mode_t`, it is reused and all the members are added, which leads to duplicates and additional members.

However, in difference to your test example both definitions of `gpio_mode_t` are in different groups, which is the problem initially described in this issue. The first one is in group `drivers_periph_gpio` and the other should be in group `cpu_esp32`. Since group names are used in identifiers,  `gpio_mode_t` in `drivers_periph_gpio` shouldn't be the same element as `gpio_mode_t` in `cpu_esp32`. And of course, they shouldn't be mixed.

When you change your example in `a.h` as follows
```diff
-  * @defgroup	plane	Airplane related stuff
+  * @defgroup	plane_a	Airplane A related stuff
```

and in `b.h` as follows
```diff
-  * @ingroup	plane
+  * @defgroup	plane_b Airplane B related stuff
```
you can observe in the XML  the enums get different ids id of something like `group__plane__a_1gabf...` and `roup__plane__b_1gac0`, but both of them have the same 4 members with ids `group__plane__b_1ggac...`. That is obviously the problem.

At the moment, the only way to avoid this is to embed all overwritings in `#idndef DOXYGEN ... #endif`. But in this case, the platform specific overwritings are also not documented in the platform specific part of the documentation.

The question is how we proceed with our online documentation. The options are:

1. We accept for the moment that the online documentation is inconsistent and enumerations contain duplicate members as well as additional members that are not valid for all platforms.
2. We embed all redefinitions/overwritings in `#idndef DOXYGEN ... #endif` and accept that platform specific overwriting are not documented.

I don't know, both options have their disadvantages.

gschorcht, 2018-12-19:
Maybe we sould open a new issue with a short description of the problem and the options we have as RFC.

closed 2020-02-26 by stale[bot]