RIOT-OS/RIOT #9795: tests/lwip_sock_*: compilation errors

gschorcht, 2018-08-18:
Compilation of these tests throw the following error message on the ESP8266 platform:
```
.../tests/lwip_sock_ip/stack.c:355:36: error: 'msg.content.value' is used uninitialized in this function [-Werror=uninitialized]
     while (data_len != (msg.content.value - sizeof(ipv6_hdr_t)))
```
which seems to be indeed the case if stack variables are not initialized by default. If I understand the semantic of the source correctly, the following construct could solve the problem:
```
do {
    msg_receive(&msg);
} while (data_len != (msg.content.value - sizeof(ipv6_hdr_t)));

gschorcht, 2018-08-19:
Additional comment: According to ISO/IEC 9899:TC3 WG14/N1256 (C99 standard) section 6.7.8 clause 10 an object that has automatic storage duration is not initialized explicitly, its value is indeterminate.

I'm a bit wondering that these tests do now throw error messages on other platforms.

miri64, 2018-08-19:
> I'm a bit wondering that these tests do now throw error messages on other platforms.

That's surprising indeed. At least our static code analyzer should have picked that one up. While I agree that this should be fixed, I think

```C
    msg_t msg = { .content { .value = 0 } };

    while (data_len != (msg.content.value - sizeof(ipv6_hdr_t))) {
        msg_receive(&msg);
    }
```

offers a more readable solution than a foot-operated loop. Do you want to provide a fix for that or should I.

gschorcht, 2018-08-19:
As you prefer. I could provide a fix, but I suppose  that merging into the master is done immediately by yourself if you fix it. I need the fix to satisfy murdock.

miri64, 2018-08-19:
> As you prefer. I could provide a fix, but I suppose that merging into the master is done immediately by yourself if you fix it. I need the fix to satisfy murdock.

I can't ACK my own PRs either ;-)

miri64, 2018-08-19:
(as such it would indeed go faster with the combination of you being the contributor, me being the reviewer, instead of waiting for a third party)

gschorcht, 2018-08-19:
Ok, should I fix it?

gschorcht, 2018-08-19:
Ok

miri64, 2018-08-19:
Yes please.

PR #9796 (2018-08-19): tests/lwip_sock_*: fix compilation error
Fixes #9795
closed 2018-08-19 by miri64