RIOT-OS/RIOT #9798: SPI: MOSI/MISO on one PIN?

Citrullin, 2018-08-19:
Hi,

I started writing a driver for a SPI epaper display. So I got one issue: It doesn't have a dedicated MOSI and MISO pin. (See: http://www.good-display.com/download_detail/downloadsId=575.html)

Some people recommended the following solutions: MUX, Connected lanes via Schottky diode or resistor. I will try out all solutions.

But in case I just connect both pins to the data pin of the epaper: Does it make problems with the RIOT hard-SPI implementation? My flow would be the following:
1. init SPI bus
2. Aquire & read pin to LOW
3. transfer bytes
4. Release
5. Aquire & read pin to HIGH
6. transfer reg
7. Release

best regards
Philipp

Citrullin, 2018-08-20:
@PeterKietzmann The display only has one data PIN. The display receives data on this PIN and sends data on this. Therefore: It is MISO and MOSI. I've seen that Waveshark uses a similar pattern. Actually I think it's just a Good Display panel with a breakout board.

Sorry, there is no read pin. I just confused it with the data/command PIN.

leandrolanzieri, 2018-08-20:
@Citrullin Having a look at the datasheet (p. 28, section 7-3-2-2) it says that :

> Under serial mode, only write operations are allowed.

Are you sure there is something to read? It seems that it only sends data and commands.

Citrullin, 2018-08-20:
@leandrolanzieri Good catch. That is interesting, since in (10) data stop the following is written:
"To stop data transmission, this command must be issued to check the data_flag.
Data_flag:
Data flag of receiving user data.
0: Driver didn’t receive all the data.
1: Driver has already received all the one-frame data (DTM1 and DTM2)."

closed 2018-08-21 by Citrullin