RIOT-OS/RIOT #9803: Codacy: unused struct members errors

gschorcht, 2018-08-20:
PRs are now checked against Codacy. Codacy very often claims that a member of a struct is not used,  e.g.:
```
struct member 'XTensa_rtos_int_frame_s::exitPtr' is never used.
...
struct XTensa_rtos_int_frame_s {
     uint32_t exitPtr;
     uint32_t pc;
     uint32_t ps;
     ...
};
```
The structure in the example defines the layout of the interrupt stack frame as used in ```thread_arch.c```. It is a standard approach to define the memory layout by structures even though not all the members are then really used.

Codacy should not claim such structures as errors.

jia200x, 2018-08-21:
@gschorcht thanks for pointing this out. I just turned off that style check.

We might find some other issues during these days, but we are working hard to solve them :)

jia200x, 2018-08-21:
Solved. Closing

closed 2018-08-21 by jia200x