RIOT-OS/RIOT #9811: Support for self contained modules

cladmi, 2018-08-21:
#### Description

I am working toward self contained modules outside of RIOT support with the same capabilities as a module defined in RIOT.

Currently adding a module may require to modify one of the global `Makefile.include` or `Makefile.dep` files and I would like to change this.

### Steps

* [ ] make: process include and dep for external modules #8987 
* [x] make: Add 'BUILDDEPS' variable #9002 
* [ ] Allow declaring an AUTO_INIT function from within a module directory
    * [ ] core: introduce crossfile arrays (xfa), refactored #9105 
        * [ ] make: Use ld -r for intermediate linking of each module #8711 
        * [ ] cortexm: ldscript refactor #9104
* [ ] Correctly handle `EXTERNAL_MODULE_DIRS` with docker

In the same time with the same idea, I try to also make packages self contained

* [x] pseudomodules: allow defining them in pkg/PKG/Makefile.include #9003
* [ ] WIP RFC Replace usage of USEPKG by USEMODULE. #9307 
* [ ] Move packages tests to package directory TODO (ref [tests: split unittests?](#8941))

jia200x, 2018-08-22:
awesome feature

Are you considering to include different sources? (e.g fetching a module from a github repo, from the file system or from a URL should be transparent to the builder). This way makes possible to have alternative implementation of modules that might live in a different repo

cladmi, 2018-08-22:
That was proposed by kaspaar: https://github.com/RIOT-OS/RIOT/pull/8987#issuecomment-383043605

There are still things to solve before but could be yes.

maribu, 2022-09-19:
Support for external modules is mature these days and part of the regular CI checks.

closed 2022-09-19 by maribu