RIOT-OS/RIOT #9813: Unable to flash on CC1310-LaunchPad

Oberpas, 2018-08-21:
Hi
I'm currently writing my thesis and need to build a sensor network with the LaunchXL-CC1310.
While working through the tutorials, I was unfortunately unable to flash the tasks on the Launchpad.
The following error message came up:

> Flash program "UNIFLASH_PATH unconfigured"/uniflash.sh not found. Aborting.
> ..../Makefile.include:395: recipe for target 'flash' failed
> make: *** [flash] Error 1

I've already installed Uniflash, but I think Ubuntu does not recognize the board on the USB input.
I use the board setup from [#6715](https://github.com/RIOT-OS/RIOT/issues/6715) by @Lafka.
Under /boards I only can find the CC1350 Launchpad and have tried with it.
Would I have to change anything for the CC1310?

Unfortunately, I am an absolute beginner and just about to get used to RIOT OS.

Thank your for your help.
I would really appreciate it if someone is helps me.

miri64, 2018-08-21:
Hi! This doesn't seem to be a board that is supported in upstream (yet). Have you tried contacting @lafka about that?

Oberpas, 2018-08-21:
Hi, thx for your quick answer.
No I didn't contact Lafka.
I thought that CC1310 could be supported cause in #6715 somebody already asked for it and @Lafka only mentioned CC13xx.

PeterKietzmann, 2018-08-22:
@Oberpas if you need help in terms of first steps or porting guidance I recommend to browse through the[wiki](https://github.com/RIOT-OS/RIOT/wiki) first and ask questions on the [devel](https://lists.riot-os.org/mailman/listinfo/devel) or [users](https://lists.riot-os.org/mailman/listinfo/users) mailing list. Furthermore, I recommend to focus on an already supported platform if you just want to get familiar with RIOT.

closed 2018-08-22 by PeterKietzmann