RIOT-OS/RIOT #9846: tunslip/pyterm no board output

kb2ma, 2018-08-27:
#### Description
When I build gcoap for tunslip from master, using the instructions in the README-slip.md in the examples, I no longer get terminal output from a samr21-xpro board from the 'make term' command. pyterm appears to start OK, but I don't see any output from the board.

This problem happens with older branches as well as master, so I don't think it's RIOT. My only guess is that I recently upgraded to Ubuntu Bionic from 16.04. Has anyone else encountered this issue?

dylad, 2018-08-27:
@kb2ma Could you check if your issue is related to #9248 please ?

kb2ma, 2018-08-28:
Thanks @dylad, that fixed the problem! Added a comment to #9248 with details.

closed 2018-08-28 by kb2ma