RIOT-OS/RIOT #9856: TF: Configuration Task Force status

jia200x, 2018-08-29:
This issue contains the status and discussions about the recently opened RIOT [Configuration Task Force](https://github.com/RIOT-OS/RIOT/wiki/Configuration-Task-Force-%28CTF%29).

### Problems we want to solve
Here's a list of problems we might want to solve. Feel free to discuss

1. Application specific configuration (#9825)
2. Kernel/modules configurations (expose defaults, override configs, load from external sources, etc)
    1. All modules seeing all other modules configuration values. This breaks encapsulation and means that a rebuild for EVERYTHING must be triggered each time ANYTHING changes.
    2. Configurations dependencies (a module might require to set some configuration from another module, but that can be also defined elsewhere. E.g `CFLAGS +=  -DFF=default -DFF=new_value` will fail.
3. Configuration for ~external~ modules (specially if [#8987 (comment)](https://github.com/RIOT-OS/RIOT/pull/8987#issuecomment-383043605) gets implemented)
    1.  Configuration in makefiles can be evil. If the external pkg is not from a trusted source, it could eventually run arbitrary code injected via Makefiles. it is not even possible to get package metadata from a makefile without risking running arbitrary code
4. Metadata and build configuration:
    1. Improve board/periph configuration (define boards withouth touching the RIOT folder, switch between configurations, override, etc)
    2. Description of available periphs/devices and it configurations
    3. Inject modules (a.k.a closed source module living outside of the RIOT repo). (#9105?)

### Corollaries (related problems)
- It's required to touch the RIOT folder in order to add new packages, drivers, or even a different board configuration (WIP #9811)
- This:
```
- A: It is not easy to have more than one version of a PACKAGE and not possible to have more than one version of a MODULE. 
- B: All components must live in RIOT. 
- A+B = To update a component one has to update all of RIOT. Updating all of RIOT means getting new version of other components, whether that is necessary/wanted or not.
```

### Related Issues/PRs
#9825 so far

jcarrano, 2018-08-29:
> E.g CFLAGS += -DFF=default -DFF=new_value will fail.

I'd say that `CFLAGS += -DFF=some_value -DFF=conflicting_val` _will not fail_ (at compile time), instead the value will be _silently overwritten_ and that's the problem.

jia200x, 2018-08-29:
> I'd say that CFLAGS += -DFF=some_value -DFF=conflicting_val will not fail (at compile time), instead the value will be silently overwritten and that's the problem.

Addressed. Thanks!

jia200x, 2018-09-24:
Here's a summary of the RIOT summit breakout session:

## Minute

* The community identified the following problems:
    * Mixed build configuration and functionality (app) configuration
    * Overriding periph configuration: it's possible to copy&paste the board periph configuration in the application, but generates a lots of code duplication and doesn't allow overriding.
    * Everything gets recompiled when changing a configuration
    * App and hw configurations are not exposed, is hard to know what can be configured (specially for beginners). 
    * There is unnecesary periph configuration for some application (e.g hello world only requires 1 UART, some boards configure 3)
    * Boards don't have human-readable names, sometimes is hard to remember the actual name of a board when flashing.
    * No standard mechanism to pass app data (keys, etc)
    * No standard mechanism to configure third party packages (lwip, openthread, etc).
    * There are no API versions nor modules versions, it's not easy to know if an application will work with a given RIOT version (and it's not possible to know if it's safe to update to a new RIOT release).
    * It's possible to inject arbitrary code if configurations are handled in Makefiles.

* The following solutions/concepts were proposed to address one or more of these problems
    * Cross file arrays (#9105 )
    * High level parsers for detecting invalid configurations and modules
        * YAML (or similar) for describing dependencies and valid configuration params
    * Human Readable CLI make arguments
    * Linux tools: KConfig
    * Fetch other repositories (e.g core repo, utils repo, etc)
    * Declare configurations in header files
    * Configurations should be close to the module (so, it's easy to get rid)
    * Whatever we do, we still need to consider third party pkgs have their own configuration mechanism, which should be invoked from RIOT.

Check the original post-its based minute [here](https://gist.githubusercontent.com/jia200x/daac639c022d5b085e0019b63eadbdb0/raw/86948d9dfe50faff8d3b1d9dbeb760de0a6b9fe6/summit_bs_outcome.jpg). @leandrolanzieri did a nice transcription [here](https://gist.githubusercontent.com/jia200x/daac639c022d5b085e0019b63eadbdb0/raw/86948d9dfe50faff8d3b1d9dbeb760de0a6b9fe6/configuration_breakout_result.jpg)

You can watch the full session [via Youtube](https://www.youtube.com/watch?v=IBPHqAwLdI4&t=5013s)

## Outcome

We agreed on **start with per-module config declaration (macros) in header files**:
* It's non intrusive
* This approach could solve some of the issues exposed by the community (declare configurations, define default values, override configurations) E.g see off-line email from @kaspar030 

```
/**
 * @brief maximum number of cooks per soup
 *
 * fmt:int:min=1:max=10
 */
#ifndef CONFIG_KITCHEN_MAX_COOKS_PER_SOUP
#define CONFIG_KITCHEN_MAX_COOKS_PER_SOUP 1U
#endif
```

A tool could parse and expose these configurations to the user. 

We can always add later something on top for handling third party PKGs configurations (that have their own configuration scheme), friendly GUIs (ncurses, web based, etc) and handling dependencies and invalid cofigurations.

MrKevinWeiss, 2024-02-29:
We can close this I suppose...

closed 2024-02-29 by MrKevinWeiss