RIOT-OS/RIOT #9873: 1 Start bit in UART

Citrullin, 2018-08-31:
I am currently writing a driver for a RFID reader. So I got one issue. The reader uses 1 start bit and 1 stop bit. Since RIOT only supports 1 stop bit I cannot use the UART implementation of RIOT. So I wanted to know if there is a workaround or some configuration for changing this behavior.

gebart, 2018-09-01:
There is always a start bit. 1 start bit, 8 data bits, 1 stop bit, no parity is the default mode for most (all?) UART drivers in RIOT, this mode is usually referred to as 8-N-1 as a shorthand.
It sounds like you will be able to communicate with this RFID device with the default settings. If it doesn't work, then you might want to check the wiring and maybe use a scope to see that the electrical signals are actually getting through.

Citrullin, 2018-09-01:
@gebart I already debug it with another UART to USB controller and I can see the bytes there. Just need to get my head around it and need some experience in this field to now common mistakes. But maybe oscilloscope makes also sense. Thanks! :)

PeterKietzmann, 2018-09-05:
Do you have a logic analyzer at hand? You could send the captures

Citrullin, 2018-09-05:
@PeterKietzmann Had the same issue for my UART Lora module. So I debuged it with a logic analyzer. I get the RX response. 
The Logic Analyzer: https://imgur.com/a/xeImKXN
So my problem is the code. https://pastebin.com/6BbNd1fY

So, probably I need to open a thread and this thread shouldn't stop so the callback function is called when the bytes come in.