RIOT-OS/RIOT #9888: gcoap: cannot get resources with paths containing multiple digit numerics

pokgak, 2018-09-05:
#### Description
gcoap states that resources must be ordered alphabetically and checks for it with `strcmp` but that actually compares it ASCII-ordered (`/0, /1, /10, /11, /2, /3, ...`) and not alphabetically (`/0, /1, /2, ..., /10, /11`)

Note: this doesn't affect paths such as `/a/10` and `/b/2` because it stops comparing the paths at the alphabets and doesn't reach the numbers

#### Steps to reproduce the issue

1. declare coap resources which contains multidigit numeric (`/0, /1, ..., /10, /11`)
2. `coap get [address] [port] /10`

#### Expected results
Gets value for /10

#### Actual results
`gcoap: response Error, code 4.04, empty payload` because gcoap stops looking for /10 after comparing /10 and /2 because it is not 'alphabetically ordered' and returns GCOAP_RESOURCE_NO_PATH instead.

pokgak, 2018-09-05:
As possible solutions, 
1. change the alphabetically ordered statement in gcoap.h to ASCII-ordered OR
2. use something like the [alphanum algorithm](http://www.davekoelle.com/alphanum.html) and not `strcmp` to look for the path

kb2ma, 2018-09-06:
Thanks for the report, @pokgak! I agree that the documentation is wrong on the requirement for sorting the entries. The list is indeed ASCII-ordered rather than alphabetic . So, in addition to your examples, `/B` sorts before `/a`. However, the issue you describe is that numbers are not sorted numerically.

The reason for the sort requirement is just to make the lookup simple and fast. For this purpose in an embedded system, I would not go farther than that. So, I definitely support an update to the documentation to say that entries must be in ASCII order, and would include an example for clarity.

kaspar030, 2018-09-06:
+1 for fixing the documentation.

PR #9918 (2018-09-13): net/gcoap: fix doc for order of server resources
### Contribution description

Simple fix to gcoap module documentation for the required order of server resources. Specifies use of ASCII order rather than alphabetic. Thanks to @pokgak for pointing this out in #9888.

### Testing procedure

N/A, just compare the documentation

### Issues/PRs references

fixes #9888
closed 2018-09-13 by bergzand