RIOT-OS/RIOT #9889: ethos: desync between ethos::last_framesize and tsrb content possible

bergzand, 2018-09-06:
#### Description

At the end of frame reception, ethos updates the last_framesize value with the size of the last received frame. It is possible that the previously received frame has not yet been read by the network stack. This causes a situation where the n-1'th is read with the size of the n'th frame. This break network frame reception.

#### Steps to reproduce the issue

For me it is rather easy to trigger the issue with the patch below.
 1 apply patch
 2 set up gnrc_border router on a samr21-xpro
 3 set up ethos connectivity with `make term`
 4 flood the node with `sudo ping6 -f fe80::2%tap0`

For me the assert usually triggers between  5 - 20 seconds. Without the patch, the node ceases to respond instead of crashing on the assert. This can be verified when enabling `DEBUG` in `sys/net/gnrc/netif/gnrc_netif_ethernet.c`. 

Packet loss is usually quite high, but this is expected as the serial interface can't really keep up with the flood of packets. 

```patch
diff --git a/drivers/ethos/ethos.c b/drivers/ethos/ethos.c
index cecec4254..f21a9a637 100644
--- a/drivers/ethos/ethos.c
+++ b/drivers/ethos/ethos.c
@@ -112,6 +112,7 @@ static void _end_of_frame(ethos_t *dev)
     switch(dev->frametype) {
         case ETHOS_FRAME_TYPE_DATA:
             if (dev->framesize) {
+                assert(dev->last_framesize == 0);
                 dev->last_framesize = dev->framesize;
                 dev->netdev.event_callback((netdev_t*) dev, NETDEV_EVENT_ISR);
             }
```

#### Expected results

Network keeps functioning

#### Actual results

With `DEBUG` enabled in `sys/net/gnrc/netif/gnrc_netif_ethernet.c` and without the patch I'm able to trigger situation like:

```
gnrc_netif_ethernet: received packet from 00:00:00:00:00:00 of length 111
gnrc_netif_ethernet: received packet from 00:00:00:00:00:00 of length 111
gnrc_netif_ethernet: received packet from 00:00:00:00:00:00 of length 111
gnrc_netif_ethernet: received packet from 00:00:00:00:00:00 of length 86
gnrc_netif_ethernet: received packet from 00:39:3a:40:fe:80 of length 86
gnrc_netif_ethernet: received packet from 2a:2b:2c:2d:2e:2f of length 86
gnrc_netif_ethernet: received packet from 11:12:13:14:15:16 of length 86
gnrc_netif_ethernet: received packet from 80:00:b9:91:57:f8 of length 86
gnrc_netif_ethernet: received packet from 00:8c:42:03:ff:fe of length 86
uhcp_client(): no reply received
uhcp_client(): sending REQ...
gnrc_netif_ethernet: send to 33:33:00:00:ab:cd
gnrc_netif_ethernet: received packet from 3d:50:8e:42:03:e5 of length 86
gnrc_netif_ethernet: received packet from 1c:1d:1e:1f:20:21 of length 86
gnrc_netif_ethernet: received packet from 5b:00:00:00:00:40 of length 86
gnrc_netif_ethernet: received packet from 00:00:00:00:00:00 of length 86
uhcp_client(): no reply received
uhcp_client(): sending REQ...
gnrc_netif_ethernet: send to 33:33:00:00:ab:cd
gnrc_netif_ethernet: received packet from 04:fc:a4:00:39:3a of length 86
gnrc_netif_ethernet: received packet from 27:28:29:2a:2b:2c of length 86
gnrc_netif_ethernet: received packet from 00:00:10:11:12:13 of length 86
gnrc_netif_ethernet: received packet from 00:00:02:80:00:77 of length 86
gnrc_netif_ethernet: received packet from 00:00:00:00:8c:42 of length 86
gnrc_netif_ethernet: received packet from 58:6b:6a:3d:50:8e of length 86
gnrc_netif_ethernet: received packet from 19:1a:1b:1c:1d:1e of length 86
gnrc_netif_ethernet: received packet from fe:80:00:00:00:00 of length 86
gnrc_netif_ethernet: received packet from fe:80:00:00:00:00 of length 86
```

The MAC address of my tap interface is: 8e:42:03:e5:a6:cd

#### Related issues

Unrelated to #9869

kaspar030, 2018-09-06:
could we just prepend the packet in the ringbuffer with the packetsize?

https://github.com/RIOT-OS/RIOT/issues/9889

bergzand, 2018-09-06:
We don't know the frame size before the full frame is received right? And updating the variable after the full frame is received is probably not thread safe ;)

bergzand, 2018-09-06:
A possible fix is in #9890 by the way

PR #9890 (2018-11-14): ethos: Only accept frame if previous frame is read
### Contribution description

This PR changes ethos to only accept a new frame if the previous frame is read. It only changes whether the bytes are actually stored to prevent disrupting the ethos state machine.

I've also included the assert from #9889 to prevent a similar breakage in the future.

### Testing procedure

same as #9889 (flood a gnrc_border_router with ping), but now it should keep working. 

### Issues/PRs references

Fixes #9889
closed 2018-11-14 by kaspar030