RIOT-OS/RIOT #9898: https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi

123ebad, 2018-09-06:


123ebad, 2018-09-06:
https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi
I m currently working on an IOT project(6lowpab based patient monitoring system) and i m making my raspberry pi-2 as a border router by following this link,but when i fully got ready my sd-card for new kernel and put the sd-card in pi-2,it was not booting and only black screen occur on LCD ,kindly help me to resolve this issue and tell me where i made a mistake. I will be very thank full

dylad, 2018-09-10:
Did you try skipping the Kernel build and tried with a default img from Raspbian ? I think 6LowPAN is enabled by default on recent Image.

miri64, 2018-09-10:
That tutorial is pretty outdated anyway. @RIOT-OS/maintainers shall we remove it or add a deprecation note?

dylad, 2018-09-10:
I would prefer add a deprecation note,
It would be a shame to drop some useful informations even if some parts are deprecated.

miri64, 2018-09-10:
Just noticed: there is already a fat deprecation note on top ;-).

miri64, 2018-09-10:
With link to an external, more cared for version.

miri64, 2018-09-10:
Since the wiki page is already deprecated, please refer to the linked project in the deprecation note @123ebad.

closed 2018-09-10 by miri64