RIOT-OS/RIOT #9904: CMake Upgrade

npcode15, 2018-09-07:
Hello,

I was trying to follow the instructions from CCN-Lite-Relay example.
I encountered an error saying cmake needs to be upgraded.
I am not able to unzip/source from files after download.
Is there any other way to upgrade?

Thanks.

dylad, 2018-09-10:
Hello @npcode15,
cmake is not related to RIOT or CCN-Lite-Relay but rather with your local installation.
If you're using an Ubuntu based distro, you can run` sudo apt-get install cmake`
that will do the trick.

npcode15, 2018-09-10:
Hello @MesoticSAS, thanks for getting back. I am using Windows host and running RIOT OS via Vagrant. So I need to update Windows cmake?

dylad, 2018-09-10:
I have no clue, I am not using Windows nor Vagrant.
Maybe try https://cmake.org/download/

npcode15, 2018-09-12:
Hey @Citrullin, Already tried, I was able to find a way around unzip. But, when I try to upgrade, I receive an error saying C++ 11 not found.

Citrullin, 2018-09-13:
@npcode15 You probably should also upgrade your gcc then.

dylad, 2018-09-18:
@npcode15 If you managed to solve this issue, feels free to close it ;)

npcode15, 2018-09-19:
@dylad, I did not check it. I will look at CCN later if needed. 

But, I will close it for now. Thanks, @Citrullin @dylad

closed 2018-09-19 by npcode15