RIOT-OS/RIOT #9910: examples/gnrc_*: gnrc_networking and gnrc_minimal compatibility

PeterKietzmann, 2018-09-10:
#### Description
Cannot ping between nodes running *gnrc_networking* and *gnrc_minimal*. This problem also relates to [Tutorials/task-07](https://github.com/RIOT-OS/Tutorials/tree/master/task-07)

#### Steps to reproduce the issue
#####  Board A (pba-d-01-kw2x)
- flash *gnrc_minimal*
- remember the link local IPv6 address it prints during start

##### Board B (samr21-xpro)
- flash *gnrc_networking*
- unset the 6lo IPHC option (via `ifconfig <if> -iphc)
- `ping6` the IPv6 address of Board A

#### Expected results
As both nodes run standard IPv6 the ping should be answered.

#### Actual results
Board B sends out the ping request which is not answered by Board A. 

As it seems, the IPv6 address of Board A is not based on the interface identifier (MAC address). When pinging this address from Board B it utilizes the the expected MAC address from the IPv6 address which does not correspond to the actual MAC of Board A.

#### Versions
```
Installed compiler toolchains 
-----------------------------
             native gcc: gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
   riscv-none-embed-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.0.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: cmake version 3.10.2
               cppcheck: missing
                doxygen: 1.8.13
                 flake8: missing
                    git: git version 2.17.1
             coccinelle: missing
```

miri64, 2018-09-10:
Also works with two `samr21-xpro`.

miri64, 2018-09-10:
Also with two `iotlab-m3`s ;-). Any reason why you deactivated IPHC on the sending node?

miri64, 2018-09-10:
> Any reason why you deactivated IPHC on the sending node?

Ah, because it isn't compiled in.

miri64, 2018-09-10:
Do you also see a link-local address based on the short address? This [is wrong when using 6Lo-ND](https://tools.ietf.org/html/rfc6775#section-5.2):

> ```
>   A link-local address is formed based on
>   the EUI-64 identifier [EUI64] assigned to the interface as per
>   [RFC4944] or the appropriate IP-over-foo document for the link […]
> ```

PeterKietzmann, 2018-09-10:
I don't get it working with two samr21 either. The node with which runs *gnrc_minimal* gets the local IPv6 address: `fe80::ff:fe00:e1a` even though its MAC address is `79:76:64:5c:f4:7e:8e:1a`. So the IP addres is **not** based on the interface identifier. Now when I try to `ping6 `fe80::ff:fe00:e1a` from the other node which runs *gnrc_networking* it sends the ping request to the MAC address: `02:00:00:ff:fe:00:0e:1a` which no one owns.

As for IPHC see [README ](https://github.com/RIOT-OS/RIOT/blob/master/examples/gnrc_minimal/README.md).

Any ideas what the difference in our setups is? BTW: I'm on:
```
commit a7b83e92cee772f41c58ec95113e6ee51c31edc8 (HEAD -> master, upstream/master)
Merge: 25868d1e0 472bd0ae0
Author: Gaëtan Harter <hartergaetan@gmail.com>
Date:   Mon Sep 10 11:52:11 2018 +0200

    Merge pull request #9567 from kaspar030/unify_test_target
    
    make: unify test target
```

miri64, 2018-09-10:
Writting the commit for the fix while we write ;-). Just give me a minute to describe the problem space there ;-).

miri64, 2018-09-10:
See https://github.com/RIOT-OS/RIOT/pull/9912

PR #9912 (2018-09-10): gnrc_ipv6_nib: always configure 802.15.4 long address with IPv6 included
### Contribution description
Our `gnrc_minimal` example configures the link-local address from the IEEE 802.15.4 short address since it does not include 6Lo-ND.
This causes the application to be incompatible with our other GNRC application that do include 6Lo-ND, since it [assumes][1] the link-local address to be based on the EUI-64 for address resolution.

This enforces long addresses (aka EUI-64) for all IEEE 802.15.4 devices when IPv6 is compiled in so `gnrc_minimal` is compatible again to the rest.

[1]: https://tools [... cut]
commit 23566166 (2018-09-10): gnrc_ipv6_nib: always configure 802.15.4 long address

Our `gnrc_minimal` example configures the link-local address from the
IEEE 802.15.4 short address since it does not include 6Lo-ND.
This causes the application to be incompatible with our other GNRC
application that do include 6Lo-ND, since it [assumes][1] the link-local
address to be based on the EUI-64 for address resolution.

This enforces long addresses (aka EUI-64) for all IEEE 802.15.4 devices
when IPv6 is compiled in so `gnrc_minimal` is compatible again to the
rest.

Fixes #9910

[1]: https://tools.ietf.org/html/rfc6775#section-5.2
closed 2018-09-10 by PeterKietzmann