RIOT-OS/RIOT #9920: drivers/tsl2561: The i2c bus is not released

jcarrano, 2018-09-11:
#### Description

The driver for the TSL2561 acquires the bus and does not release it. This means that it is not possible to use other devices connected to the same bus.

See https://github.com/RIOT-OS/RIOT/blob/6265da2894ed5cb7c414a44cd2856be918c3a06b/drivers/tsl2561/tsl2561.c#L57

#### Expected results

- Every call to `i2c_acquire()` should be matched by a call to `i2c_release()`
- The bus should not be be held if it is not being used.

#### Actual results

- There is only one call to `i2c_acquire()` when the driver is initialized.
- There are no calls to  `i2c_release()`.

dylad, 2018-09-11:
Good catch ! This bug must be there for a long time as it has not been introduced by the I2C rework.
Anyway, @jcarrano I can provide a PR for fixing this if you have the hardware to test it.

jcarrano, 2018-09-14:
@dylad I don't have this hardware. I found the bug because I was implementing a "similar" sensor (TSL4531, #9911 ) which turned out not to be similar at all.

dylad, 2018-09-14:
@jcarrano Alright, I'll provide a fix soon then and we'll see if someone can test it on real hardware.

dylad, 2018-09-21:
@jcarrano I'm closing this, since #9937 got merged.

closed 2018-09-21 by dylad