RIOT-OS/RIOT #9925: on what platforms RIOT support OpenThread?

Kabariya, 2018-09-12:
I want to use Openthread with RIOT but I don't know on what platform ROIT support Openthread?

Citrullin, 2018-09-12:
@Kabariya I just looked into the package. Looks like RIOT is just cloning the repo and using RIOTs network abstraction layer. So every driver which is available in RIOT should also work in combination in Openthread. Just take a look into: http://riot-os.org/api/group__pkg__openthread__cli.html and http://riot-os.org/api/group__drivers__netdev__api.html#ga14012f723b7591ad2fa42ace34601ac4

aabadie, 2018-09-12:
> https://github.com/RIOT-OS/RIOT/issues/9925#issuecomment-420656323 | view it on
> GitHub ] , or [
> | mute the thread ] .

Hi, 

From what I see in the OpenThread adaption code in RIOT [1], only at86rf2xx based radios are supported. So it should work with samr21-xpro, iotlab-a8-m3, iotlab-m3, fox, at least. 
For others 802.15.4 radios (cc2538, xbee), it's not supported and it even doesn't build. 

Alex 

[1] https://github.com/RIOT-OS/RIOT/blob/master/pkg/openthread/contrib/openthread.c#L73 

-----

Kabariya, 2018-09-13:
thanks for helping me out. I am still new to this field so I don't know what kind of changes, I need to make it work, 

btw, @Citrullin  if you find out your questions answer please let us know... 

Shouldn't it also work with other drivers, since it is just using the network layer?  :)

PeterKietzmann, 2018-09-25:
@jia200x can you please answer the question?!

PeterKietzmann, 2018-09-25:
PS: If it is answered I guess we can close the issue

jia200x, 2018-09-25:
> Shouldn't it also work with other drivers, since it is just using the network layer? :)

It _should_. The reason why we can only use at86rf2xx radios is because [that's the only radio with FRAME_PENDING support](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx_netdev.c#L664). If we are able to add this to other radios (not that hard to do), we are able to use OpenThread in all radios.

bergzand, 2018-09-25:
> It _should_. The reason why we can only use at86rf2xx radios is because [that's the only radio with FRAME_PENDING [...]

Is there a tracking issue for this somewhere? I don't mind adding this to the mrf24j40, but then I need to know that it is missing and what/when this event should trigger :)

jia200x, 2018-09-25:
> Is there a tracking issue for this somewhere? I don't mind adding this to the mrf24j40, but then I need to know that it is missing and what/when this event should trigger :)

Cool! Thanks.
I will open an issue explaining the situation and how to proceed for other radios :)

PeterKietzmann, 2018-09-25:
Once done, please close this issue here

jia200x, 2018-09-26:
I just opened [a new issue](https://github.com/RIOT-OS/RIOT/issues/10045) for tracking OpenThread status in other radios. Then, I close this issue.

closed 2018-09-26 by jia200x