RIOT-OS/RIOT #9929: objcopy not found. Compiler arm-none-eabi-gcc is required but not found in PATH

dcjw, 2018-09-13:
#### Description
objcopy not found, and arm-none-eabi-gcc is required but not found in PATH

#### Steps to reproduce the issue
I am trying out the example on http://riot-os.org/api/index.html#the-quickest-start. As I enter "sudo make BOARD=openmote-cc2538 all", I see the error in the following "Actual results" section

However, if I use the native platform, I don't see these errors. 
In addition, when I do "which objcopy" or "which arm-none-eabi-gcc", both returns the correct path. Not sure why error occurs if I specify a different platform

I am running this on Ubuntu 16.04.
The tool chain I downloaded is gcc-arm-none-eabi-5_4-2016q3-20160926-linux.tar.bz2
I am using this version because it is the closest version to the environment at my work place. 

#### Actual results
makefiles/toolchain/gnu.inc.mk:14: objcopy not found. Hex file will not be created.
Compiler arm-none-eabi-gcc is required but not found in PATH.  Aborting.
Makefile.include:333: recipe for target '..compiler-check' failed
make: *** [..compiler-check] Error 1

#### Versions
native gcc: gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609
msp430-gcc: missing/error
avr-gcc: missing/error
  arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
   mips-mti-elf-gcc: missing/error
../RIOT/dist/tools/ci/print_toolchain_versions.sh: 92: ../RIOT/dist/tools/ci/print_toolchain_versions.sh: clang: not found
arm-none-eabi-newlib: "2.4.0"
mips-mti-elf-newlib: missing/error
           avr-libc: missing/error (missing/error)
           cppcheck: missing
         coccinelle: missing
                git: git version 2.7.4

dylad, 2018-09-13:
Hello @dcjw,
could you share your PATH variable ?
`echo $PATH`
If arm-gcc is not present in your PATH variable you can add it using `export PATH=/your/path/to/gcc:$PATH`

dcjw, 2018-09-13:
Hi dylad,
Thanks for the quick reply. Yes, I can see the correct path included for the compiler in PATH variable

$ echo $PATH
/home/dcw/bin:/home/dcw/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/dcw/Tools/gcc-arm-none-eabi-5_4-2016q3/bin

$ which arm-none-eabi-gcc
/home/dcw/Tools/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc

dylad, 2018-09-13:
@dcjw 
I just noticed this in your fist comment:
`sudo make BOARD=openmote-cc2538 all`
There is no need to run this command using sudo, user and root user share a different PATH variable so if you set your GCC for your default user but run make using sudo, it will not find the compiler.
Just remove sudo and it should works.

dcjw, 2018-09-13:
Hi dylad,
Previously, there was some permission issue when I ran the command, so I've added sudo to run since then.
I don't recall how I resolve the permission issue, but I can run it without sudo now, and I don't get the path error for the compiler and objcopy. 
Thanks a lot for the help!

dcjw, 2018-09-13:
Hi dylad,
Sorry to bother you again. I am trying to run terminal for my device. However, I got a permission error. If I do it with sudo, I do see the message about "objcopy not found. Hex file will not be created" again. However, terminal seems to be running fine. Is there a way I can do it without sudo?

### Without sudo
$ make BOARD=openmote-cc2538 PORT=/dev/ttyUSB0 term
/home/dcw/ANRG/RIOT/Tutorials/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"
No handlers could be found for logger "root"
Traceback (most recent call last):
  File "/home/dcw/ANRG/RIOT/Tutorials/RIOT/dist/tools/pyterm/pyterm", line 775, in <module>
    args.log_dir_name, args.newline, args.format, args.prompt)
  File "/home/dcw/ANRG/RIOT/Tutorials/RIOT/dist/tools/pyterm/pyterm", line 179, in __init__
    '.log', level=logging.DEBUG, format=self.fmt_str)
  File "/usr/lib/python2.7/logging/__init__.py", line 1547, in basicConfig
    hdlr = FileHandler(filename, mode)
  File "/usr/lib/python2.7/logging/__init__.py", line 913, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python2.7/logging/__init__.py", line 943, in _open
    stream = open(self.baseFilename, self.mode)
IOError: [Errno 13] Permission denied: '/home/dcw/.pyterm/HP-Pavilion-NB/default-run.log'
RIOT/Makefile.include:400: recipe for target 'term' failed
make: *** [term] Error 1

### With sudo
$ sudo make BOARD=openmote-cc2538 PORT=/dev/ttyUSB0 term
RIOT/Tutorials/RIOT/makefiles/toolchain/gnu.inc.mk:14: objcopy not found. Hex file will not be created.
RIOT/Tutorials/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyUSB0" -b "115200"
No handlers could be found for logger "root"
2018-09-13 13:45:59,126 - INFO # Connect to serial port /dev/ttyUSB0
Welcome to pyterm!
Type '/exit' to exit.
help
2018-09-13 13:46:30,846 - INFO # help
help
2018-09-13 13:46:41,411 - INFO #  help
2018-09-13 13:46:41,413 - INFO # Command              Description
2018-09-13 13:46:41,416 - INFO # ---------------------------------------
2018-09-13 13:46:41,419 - INFO # reboot               Reboot the node
2018-09-13 13:46:41,423 - INFO # ps                   Prints information about running threads.
\> /exit
2018-09-13 13:46:56,778 - INFO # Exiting Pyterm

dcjw, 2018-09-14:
Hi dylad,
I used `sudo usermod -a -G dialout $USER` previously, so I am already a member of dialout group. I also reboot after that, but I still got permission error.
I just check the permission of /home/dcw/.pyterm. It is 
drwxr-xr-x 3 root root 4096 Sep  9 18:03 /home/dcw/.pyterm

Should I change it to dialout group?

dylad, 2018-09-14:
Hmm, before that can you try to install a minicom or picocom and connect it to your board to check if you have the same issue ?

dcjw, 2018-09-14:
Hi dylad, gebart,
After deleting the directory, I can run it without sudo now. Thanks a lot for the help!

closed 2018-09-14 by dcjw