RIOT-OS/RIOT #9936: Unclear board name: `stm32mindev` (STM32 Minimum System Development Board)

silkeh, 2018-09-14:
The name *STM32 Minimum System Development Board* (`stm32mindev`) is ambiguous: a search on AliExpress or Google image search returns various boards, mainly [Blue pill][] boards, but also some [Black pill][] boards and [STM32F407][] clones.

Reading the description the board seems to refer to the [Black pill][] (having a different pinout and the LED on PB12, also see the [Black pill comparison with the Blue pill][Comparison]).

Proposal: rename the `stm32mindev` board to `blackpill`.

[Blue Pill]: http://wiki.stm32duino.com/index.php?title=Blue_Pill
[Black Pill]: http://wiki.stm32duino.com/index.php?title=Black_Pill
[Comparison]: http://wiki.stm32duino.com/index.php?title=Black_Pill#Comparison_with_the_Blue_Pill
[STM32F407]: https://wiki.stm32duino.com/index.php?title=STM32F407

maribu, 2018-09-19:
Hi. Thanks for the feedback. I added the support of this board and can confirm that it was sold as "STM32 Minimum System Development Board" and still is sold with this name. I suggest to keep the name therefore.

silkeh, 2018-09-19:
@maribu: the problem is that [several different boards][1] are sold with the name "STM32 Minimum System Development Board". The name is too generic to refer to a unique board (it seems to refer to cheap development boards for several STM32 mcu's). From the board files I think the one you have is probably nicknamed [*Black Pill*][2] (please correct me if you have a different one). I'm also not a big fan of having to refer to these boards by their nicknames rather than the names the boards are sold under, but this is one of the downsides of vendors on sites like AliExpress.

[1]: https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20180919014047&SearchText=STM32+Minimum+System+Development+Board
[2]: http://wiki.stm32duino.com/index.php?title=Black_Pill

maribu, 2018-09-19:
Phew, you're right. It got much more fragmented. I have a few different boards sold as STM32 Minimum System Development Board, but they all are "identical" from RIOT's point of few: The were all STM32F103C8T6 break-out boards with an on-board LED on pin B12. So even when the physical dimensions differed and even some broke out the GPIOs in different order, all had correctly labeled PINs and the difference was negligible from a software point of view. But now there are even STM32F030* and STM32F407* boards sold under that name. They are clearly not at all compatible with the `stm32mindev` board in RIOT.

You convinced me that renaming to black pill is very sensible. (But my opinion doesn't count, so be prepared to convince someone who's opinion does count ;-))

Do you mind to create a pull request to change it? It would also be very nice to have the board in the RIOT wiki. I wanted to do that myself, but I had too little time.

PR #9963 (2018-09-19): Rename stm32mindev to blackpill
### Contribution description

Rename the `stm32mindev` (STM32 Minimum Development Board) to `blackpill` to match the commonly known board name.

### Issues/PRs references

Fixes  #9936
closed 2018-09-19 by MrKevinWeiss